bigtreetech/BIGTREETECH-SKR-mini-E3 #236: SKR Mini E3 - TMC Drivers/Stepping mode change (Off) is not saved

RiMr71, 2020-04-28:
### Description

### Steps to reproduce

1. Change stepping mode of TMC drivers StealthChop Enabled on OFF (because motors are noisy)
2. Save settings to EEPROM
3. Check motors (Autohome) - motors are quite
4. Restart printer
5. Stepping mode of TMC drivers is again ON

**Expected behavior**
Saved changed value of Stepping mode

**Actual behavior**
Changed value of Stepping mode ate not saved

### Additional Information
Values like steps per mm or TMC drivers current are saved. Only Stepping mode is NOT saved.
Video with behavior:
https://youtu.be/uwCqnxzFnE4

paul-1, 2020-04-28:
I think if you have the spread jumper set for spread, then it will override and default on.    I have the jumper set to stealth, and I'm able to set
X, Y, Z to StealthChop and E to Spread.

RiMr71, 2020-04-28:
Thanks helps a lot. I have a few new Mini E3 with a different jumper positions. Now it is clear. Thanks.

closed 2020-04-28 by RiMr71

FJJWFP, 2020-04-30:
Hi, I am having the exact same problem, how exactly do you fix this? Thanks a lot

RiMr71, 2020-04-30:
I remove jumper and I can now select stepping mode on LCD and this options
is still saved after printer restart.
Richard.

FJJWFP, 2020-04-30:
Thank you. I have figured it out.