bigtreetech/BIGTREETECH-SKR-mini-E3 #237: [FR]firmware-bltouch-for-z-homing with -Z axis

eiscorp, 2020-05-01:
hi, i have an Ender 3.
Normaly, this board is make for ender3 ?? i would like to know for what we don't have the possibilities to adjust the Z axis under 0. We can only adjust the bed manualy, ithink this FW is not forwhat you say ! and all people need to compile the FW alone.

It's very difficult for noob !

Please make an FW with the possibilities to adjust the Z axis under 0 !

Thank you in advance

paul-1, 2020-05-01:
Its that way to keep you from running the z into the bed manually in normal operation.

Just do math.   Set you Z offset to -4.00 mm  (Something more negative than your normal offset)  Then make your measurement, you will get a positive number, as you will hit the bed before zero.   Then add that number to -4   ( i.e  -4.00 + 1.46 = -2.54)

eiscorp, 2020-05-01:
ok, i'm just asking to uncomment in FW this line : MIN_SOFTWARE_ENDSTOPS -z offset .

And just after you don't need math or whatever.....

paul-1, 2020-05-01:
Yes, but maybe I don't want to run the extruder into the bed. 

A Feature to one person is a Bug to another.   But that's the beauty of open source.....just build your own version with the features you want.

Patrity, 2020-05-05:
Use pronterface and send `M211 s1` to disable the software end stops. Then I would continue using pronterface to lower the bed to get and establish your probe z-offset.

eiscorp, 2020-05-05:
ok i don't use any software, i configure with cura and i put the file on sd. after i let the printer make the work. If bbt sell tft35 it's to do that ? no ?
If we can use it, don't sell it

Patrity, 2020-05-05:
> ok i don't use any software, i configure with cura and i put the file on sd. after i let the printer make the work. If bbt sell tft35 it's to do that ? no ?
> If we can't use it, don't sell it

I've got little to no clue what you're attempting to convey but, if you're not sure how to use PronterFace, it is a completely separate program than your slicer (Cura in your case).
It interfaces with your printer via USB. If you aren't sure how to use it, there are several helpful guides on youtube.
Good luck!