bigtreetech/BIGTREETECH-SKR-mini-E3 #238: SKR Mini E3 v1.2 fans always on 100%

galardao, 2020-05-02:
Hello, 
 Sorry about my bad English, I speak Portuguese. I recently purchased an SKR Mini E3 v1.2 card. 
 Everything was working fine, but yesterday the cooling fan did not comply with the chosen speed, it was 100% all the time regardless of any settings on the panel or in the gcode. Measured with multimeter on the fan0 connector and is always with 23.6 volts. I would like help in solving this problem.  Best Regards,

sregister, 2020-05-02:
I have purchased two boards in the last week and had both fail the same way. I think the mosfet next to the connector is failing. I have more mosfets on order but I wonder if there is something else causing issues. Are you using 12v fans with a buck converter?

galardao, 2020-05-02:
I using original fans....

my ender3 is all stock.

only skr mini e3 v1.2

galardao, 2020-05-05:
Yes, I have control, everything was ok, until a few days ago.

There is an output that is 24v full, and the other output is 24v variable depending on the temperature setting.
You may have connected something wrong with your board, the fan connectors of the cpu case and the part ventilation is reversed in the SKR mini E3 v1.2

galardao, 2020-05-09:
Video recorded as requested by the BTT support team

https://youtu.be/s7zhgiU1xNs

cht47, 2020-05-10:
It's probably broken. Had the same issue with my first board after some days. Cheap Mosfet, my new board seems to be more reliable.

galardao, 2020-05-10:
I am waiting for the return of BTT support.
I sent the video as requested by BTT support, still waiting for a reply regarding the card's warranty.

sregister, 2020-05-12:
I replaced the mosfet and regained fan control. I agree--this is likely hardware failure.

galardao, 2020-05-12:
What is the code of this mosfet so that those who have the same problem can buy and exchange.

Please inform the technical specifications of the component.

galardao, 2020-05-12:
In Brazil, if I don't have the mosfet specification, they don't sell.

> Troquei o mosfet e recuperei o controle do ventilador. Eu concordo - isso é provável falha de hardware.
Qual é o código desse mosfet para que aqueles que têm o mesmo problema possam comprar e trocar.

Por favor, informe as especificações técnicas do componente.

In Brazil, if I don't have the mosfet specification, they don't sell.

sregister, 2020-05-13:
> What is the code of this mosfet so that those who have the same problem can buy and exchange.
> 
> Please inform the technical specifications of the component.

I'm using a SI3404-TP. Datasheet details here: https://www.digikey.com/product-detail/en/micro-commercial-co/SI3404-TP/SI3404-TPMSCT-ND/7681699.

galardao, 2020-05-13:
> > Qual é o código desse mosfet para que aqueles que têm o mesmo problema possam comprar e trocar.
> > Por favor, informe as especificações técnicas do componente.
> 
> Estou usando um SI3404-TP. Detalhes da folha de dados aqui: https://www.digikey.com/product-detail/en/micro-commercial-co/SI3404-TP/SI3404-TPMSCT-ND/7681699 .

friend how is the operation using this SI3404-TP mosfet

drjeh, 2020-05-16:
I've tried two different boards with the same problem.  Fan won't respond to M107 and won't turn off. I can't seem to find any other answers.

galardao, 2020-05-19:
> Eu tentei duas placas diferentes com o mesmo problema. O ventilador não responde ao M107 e não desliga. Não consigo encontrar outras respostas.

You are connecting correctly, a parts fan connection is reversed in position in relation to the original creality plate

galardao, 2020-05-19:
> Eu tentei duas placas diferentes com o mesmo problema. O ventilador não responde ao M107 e não desliga. Não consigo encontrar outras respostas.

https://youtu.be/s7zhgiU1xNs

See the link above.

CRTified, 2020-05-23:
As @cht47 said, it's probably a broken MOSFET. 
I have an E3 DIP v1.0 and looked at the MOSFET `Q2` with an oscilloscope, which is driven by the non inverting buffer U8. 

[image]
[image]

While the `FAN_PWM` signal (which comes from the SMT32) is a totally fine PWM signal, both `FAN` **and** the drain of the MOSFET was at 3V when the PWM signal was off, which supplies 21V (Voltage potential between 24V and 3V) to the fan in the `FAN0` port (which is enough for most 24V-fans to spin at a relatively high speed).

Today, the replacement MOSFET came. My soldering job is pretty bad in this case (I touched headers with the soldering iron and the replacement MOSFET is not flat on the PCB - I'm normally better at soldering, I promise ;) ), but this is fine as the component does not handle that much load (it's just that one fan, and one additional LED).

[image]

For soldering, I used a TS100 with the I tip and leaded solder (easier to solder with, but not RoHS-compliant and basically slightly toxic because of the lead). I removed the broken `PL4009` by removing the solder (desoldering braid) and lifting it up with tweezers. Then, you just solder the new MOSFET in place. Look at `SKR-mini-E3-TOP.pdf` or `SKR-mini-E3-PIN.pdf` to locate the MOSFET. For the v1.0, it is `Q2` next to the `FAN0`-port - **I did not check whether it's `Q2` for the other versions, do that yourself**.

As a replacement, you can use other MOSFETs than a [`PL4009`](http://www.szparkson.net/file-download-694-left.html). I used an `IRLML6344`. 
The most important things that I'd look for when searching a replacement:

 - N-Channel enhancement mode MOSFET
 - `SOT-23` or `TO-236` form factor (with the pin assignment as shown below, from the Datasheet of the `PL4009`)
 - V_DS of at least 30V
 - R_ds(on) of at most ~33 mΩ (@V_GS=4.5)
 - V_GS(th) of at most 1.1V
 - Total Gate Charge Q_g of < 9.7 nC (So that we do not inhibit PWM capabilities by switching too slow)

[image]

I hope this post helps other people fixing that issue.

dmitrybabeshko, 2020-05-30:
The same story on SKR 1.4 Turbo. FAN0 Q4 PL4009 mosfet is a total junk. Passed out in one week. I've replaced with AO3400A http://www.aosmd.com/pdfs/datasheet/AO3400A.pdf

galardao, 2020-05-31:
> A mesma história no SKR 1.4 Turbo. O mosfet FAN0 Q4 PL4009 é um lixo total. Desmaiou em uma semana. Substituí pelo AO3400A http://www.aosmd.com/pdfs/datasheet/AO3400A.pdf

Hi,
Very good friend, I will try to find this model in Brazil, because at the time of the corona virus, depending on freight from China is difficult.

galardao, 2020-05-31:
In Brazil I found only this model on a sales website:

http://www.aosmd.com/pdfs/datasheet/ao3400.pdf

Will it work well?

> A mesma história no SKR 1.4 Turbo. O mosfet FAN0 Q4 PL4009 é um lixo total. Desmaiou em uma semana. Substituí pelo AO3400A http://www.aosmd.com/pdfs/datasheet/AO3400A.pdf

CRTified, 2020-05-31:
@galardao Yes, the AO3400A is _nearly_ identical to the AO3400 and will probably work, too. The most important parameters are identical or match the description that I've given.

galardao, 2020-06-02:
@CRTified tanks...

best regards

lcricco, 2020-06-26:
The same here! I bought 2 SKR V1.4 Turbo with problems and returned them. Then I bough s SKR mini e3 v1.2 -> FAN0 (cooling parts fan not working at all, even when switching it. The fan is ok. Finally, I bought another SKR min e3 v1.2 and now the FAN0 green led is always on. I can turn on and off through Octprint, and the LCS shows me the fan icon on or off, but still always ON. I've been spending days trying to figure out. @galardao, boa sorte!

CreeperShift, 2020-07-14:
Wow thank you, I just received my SKR 1.4 and literally the same issue. Part cooling fan doesn't work despite being in the correct pin, fan works 100% in the non PWM pins. Multimeter spits out 3.2V when at 100% and when off.... I'm gonna order some external mosfets so I can hook up a buck converter to the regular 24v and then pwm it using one of the pins I don't need because I wanted to replace the stock fan with 12v ones anyways... so yeah not a big issue for me but it was driving me crazy.

marchaos, 2020-07-28:
Same here - which mosfet is for FAN0 on the SKR 1.4? Can someone include a picture?

sregister, 2020-07-29:
> Same here - which mosfet is for FAN0 on the SKR 1.4? Can someone include a picture?

Scroll up on this page--there is an image with blue circle showing the mosfet

CRTified, 2020-07-29:
@marchaos This is the repository for the SKR mini E3, not the SKR 1.4. But according to the schematics, it is the MOSFET `Q4`, located on the top side between the `FAN0` header, the `X-CLS` header and `Q1`. It is the same `PL4009` FET that broke on my board, so the replacements that are already stated here /should/ work.

But I can not guarantee that this is all correct, I do not have that board at hand. This is just based on the schematics and board PDF in [this folder](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4/Hardware).

tablatronix, 2020-09-06:
I plugged my board in and within first 5 minutes fried these also, I must have shorted a fan while rearranging fan ports or something, Q2 exploded, Q4 dead, Q1 E1 dead

replaced fans Q2 and Q4 with whatever I had on hand, A03414 (20v...) they worked
replaced Q1 with whatever I had, it didn't work, I think the 74hct125 output might be dead.

Bought a new board, ugh, gotta find replacements and try to fix this one I guess or $50 down the drain, these fets need better protection.

DaveJS46, 2020-09-28:
When I first set this card up after waiting for a TFT35 screen to be delivered, I connected the dual fans that I intended to use on my 3D printer to the SKR1.4 Turbo board FAN0 connector. I then used the touch screen to run the fans from 0% to 100% and it was while doing this that the FAN0 control mosfet (Q4) failed. Consulting this page I found that this was a common occurrence on SKR1.4 and 1.4 Turbo boards. I ordered some replacement mosfets and change Q4 but no joy. So I checked the signal to the Gate of the mosfet from the CPU via U6 and found in both case there was no signal and I assumed that when the mosfet failed it allowed 24V onto this signal line and wiping it out.
I have proved this to BTT product support, by doing checks that they require, but they are just now being silly asking me to do one unnecessary thing after another, but totally ignoring the scope checks that I did - in fact anything rather than send me a replacement board and admit they got the design wrong.

[image]

Like many others I gave in and bought another board but before I did I connected my fans to a 24V bench PSU and did scope checks which revealed that my fans produced 9V spikes of 60uS duration about every 2 to 3 mS. As the PL4009 spec is 30V Vds max (and with its very small physical form) it is probably the 9V spike that blows it. I connected a 10uF 20V electrolytic cap between the FAN0 connector control pin and 0V (effectively across the mosfet) which basically shorts any spikes to 0V and protects the mosfet.
So far all has been well on the new board when running the fans up to a 100%!

tablatronix, 2020-09-28:
What kind of fans are you using, brushed or brushless?

evilpoly, 2020-10-01:
DaveJS46 got a link to the parts you used and how u had modded the board . im new to the print scene and just had my board for 3 weeks . Sad to say my board is not working like it used too . thank you in advance .

tablatronix, 2020-10-01:
You might be able to put a backfeed diode across the fan, if it is indeed kickback, I can see brushed fans or bucks doing that

DaveJS46, 2020-10-01:
I have been replying to queries by answering emails generated by this site but it doesn't seem that they get here like that so here are the two replys that I emailed earlier -
1) Brushless - Fan parameters Anet® 24V DC 5015 50x50x15mm 2-pin Exhaust Blower Cooling Fan. Two wired in parallel. (Tablatronix)
2) Well, I'm still arguing with BTT over them supplying me a replacement board - let's just say that their Customer Services Dept. are not very customer focused and love the photo (even if it shows them nothing) rather tan the written word - and in the meantime I've purchased a new board and all I did was wire a 10uF 50V capacitor (eg from Aliexpress) under the board, cap +ve leg to FAN0 -ve terminal and cap -ve leg to FAN 1 -ve terminal (0V). I've not had any problems with Q4 - yet!! (evilpoly)

evilpoly, 2020-10-10:
CRTified i replaced this mosfet aswell . didnt solve the problem. any other ideas?

tablatronix, 2020-10-10:
The buffer ic is probably bad for that output, probably blew it. I think thats what happened to mine

evilpoly, 2020-10-10:
My fan doesnt turn off anymore . Well i guess on the possitive side ill just add  a manual switch to it instead . Or just unplug it .

DaveJS46, 2020-10-11:
@evilpoly and @tablatronix: Try putting it to BTT Customer Services as I did. I think that I finally convinced them that I had a valid case for my money back (they asked for my order number and what did I want a replacement or money back - as I'd bought a new board I asked for my money back) but I've seen no sign of it - and probably never will. But if more who complain it will maybe cause a design update - good for the future.

evilpoly, 2020-10-11:
I think i got mine off amazon . Knowing it was china made i figured its no biggy its prolly a burnt resister or something . Easy fix . The board works fine . Just the fan doesnt turn off . As many have done i bought a new board. I figure it was human error . I added 2 blower fans to it and overloaded the board and broke something . It was working fine for weeks .

DaveJS46, 2020-10-11:
@evilpoly The mosfet Q4 (the one that fails) is rated at more than 5amps - I doubt that your 2 cooling fans had a current draw of more than a couple of hundred milliamps, i.e. well in spec. The max voltage between Drain and Source of the mosfet (the pins connected to the fans and 0V respectively), however, is 30V and this can be easily exceeded (on 24V systems) if, as is likely, the fans produce noise large spikes. A physically larger mosfet, such as the ones used for the heaters and as was use on the 1.3 version of the board for FAN0, would probably be more tolerant to spikes despite having the same electrical spec as Q4. I believe that Q4 was changed in the 1.4 design of the board as they were running out of real-estate space on that part of the board - a lot has been crammed in that corner. Q4 will have unpredictable failure modes and if a short between Drain and Gate results then it is possible that the buffer chip U6 and even the micro U1 can become damaged (see my diagram above) as I believe happened in my case.

tablatronix, 2020-10-11:
@DaveJS46 I would but I already reworked all the mosfets , and ordered a new buffer, I just bought a new board. I doubt they would refund me since I replaced 3 mosfets on the board fixing it.

Gorgonbert, 2020-10-12:
I guess the FAN0 MOSFET on my SKR 1.4 Turbo Board is blown too since i can't turn the fan off anymore. So here I am and join this party!

I have very little knowledge in electronics. In some Youtube Video I learned that if you connect fans to MOSFETs you need a diode in reverse so when you turn the fan off the current that's produced by the spinning fan blade cannot go back into the MOSFET and destroy it... i can't see a diode in the schematic... could this be the issue? 

I have a blower style fan connected and the rotating part spins rather long after it's turned off, has a metal part on it and gives a vibe of "I'm a chunky big fan"

If somebody here knows electronics... can you help me what diode do i need and how do I have to connect it? 

I will try and replace the MOSFET with a model mentioned above and don't want to destroy it again because of a missing diode.

DaveJS46, 2020-10-16:
Wow! My pessimism was not justified - BTT have just re-imbursed me in full for my faulty SKR1.4 Turbo board. So many thanks to them - it took a bit of hoop jumping but worth it in the end.

mikegleasonjr, 2020-10-24:
@DaveJS46

> 2. Well, I'm still arguing with BTT over them supplying me a replacement board - let's just say that their Customer Services Dept. are not very customer focused and love the photo (even if it shows [...]

Just to be sure, I am waiting on the SKR Mini E3 V2.0 and seen ppl having the fan problems as well. So I think I will do the mod before installing the board.

You solder the CAP+ to FAN0 GND, and CAP- to FAN1 GND?

Thanks!

DaveJS46, 2020-10-24:
@mikegleasonjr 
Yes, that is what I did.

mikesuperrc, 2020-10-27:
G'day.
I'm have this same issue on my skr mini e3 v2 also. 
Could I safely use an AO3404 do you think?
I can't seem to find any of the other chips mentioned on here locally. 
Sorry I'm not very literate with the technical specs, etc. 

This is the data sheet for the AO3404. 
[AO3404_AlphaOmegaSemiconductors.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5444001/AO3404_AlphaOmegaSemiconductors.pdf)
[image]
[image]

CRTified, 2020-10-27:
@PrimeEpilogue 

> Could I safely use an AO3404 do you think?

It might work, but it has some properties that are worse than the PL4009. For example, the R_DS(on) is a bit worse (which might result in a "hotter" MOSFET), its V_GS(th) is a bit higher (which might worsen the switching behavior, as it turns on later and off sooner). The maximum ratings are not violated and the pinout is identical, so it will probably work, but it's probably not the best choice for replacement, given that it does not match (or improve) the original component in some aspects.

DaveJS46, 2020-10-27:
@PrimeEpilogue
 I would say that the spec is close enough that it should be OK. A couple of fans are only going to draw a few hundred milliamps so shouldn't stress the part too much. I would, however, caution changing these surface mount parts unless you are set up with a fine point soldering iron and are confident that you have the skills with it to work on them in very tight areas. Maybe just asking for a replacement board is the better option if you have any doubts?

mikegleasonjr, 2020-10-27:
@PrimeEpilogue May I ask if you were using the stock fan on you 3D printer?

PrimeEpilogue, 2020-10-27:
@CRTified I have found some AO3400 I saw above that you said ao3400a were OK as similar to the AO3400, are these still a good choice?

@DaveJS46
 I have some experience soldering surface mount parts. I have a ts100 soldering iron and have had to replace micro usb ports, etc. on flight controllers, etc for my mini quads.

@mikegleasonjr
 no I was using 2 12v 5015 in series. I'll be changing to 2 24v 5015 now

PrimeEpilogue, 2020-10-27:
@DaveJS46 would you recommend I do the capacitor mod while I'm at it? Does it matter what type of cap it is as long as it's still 10uF 50V?

DaveJS46, 2020-10-28:
@PrimeEpilogue
 Well, as I say, the cap mod has worked for me, so it is your choice. I don't see the point of just changing the mosfet for a similar one without trying to do something to cut down the chances of it blowing again. I used a small electolytic but I guess tantalum bead would be OK too.

PrimeEpilogue, 2020-10-28:
@DaveJS46 OK thanks. Just wanted to make sure it was still going well before I did it.

javimagro, 2020-10-29:
> I guess the FAN0 MOSFET on my SKR 1.4 Turbo Board is blown too since i can't turn the fan off anymore. So here I am and join this party!
> 
> I have very little knowledge in electronics. In some Youtube Video I learned that if you connect fans to MOSFETs you need a diode in reverse so when you turn the fan off the current that's produced by the spinning fan blade cannot go back into the MOSFET and destroy it... i can't see a diode in the schematic... could this be the issue?
> 
> I have a blower style fan connected and the rotating part spins rather long after it's turned off, has a metal part on it and gives a vibe of "I'm a chunky big fan"
> 
> If somebody here knows electronics... can you help me what diode do i need and how do I have to connect it?
> 
> I will try and replace the MOSFET with a model mentioned above and don't want to destroy it again because of a missing diode.
> 
> EDIT: Replaced it with a IRLML6344TRPBF and now it works like a charm... i ordered 10 of these... so for now I'll try without diode since i don't know what diode and how exactly to connect it (yeah, noob here)... we'll see... anyways BIG THANKS for this helpful thread... Have a nice day!

I have bought a SKR MINI v1.2 board and I have the same problem.  fan layer is alwais on and green led is alwais lighting.  But In my case I cannot heat bed or nozzle , in both case thermistor is reading the good temp but ist not increased so in few seconds Marlin stop the board.   Is this the same problem of the MOSFET ? or in you case the problem was only with the fan layer ??

ymurenko, 2020-12-09:
I think we should look for a way to prevent this issue altogether. Is there a reliable way to protect this mosfet from whatever plagues it? From what I understand, the PL4009 is just barely suited to running a 24v PWM fan, and any voltage spikes or potential noise can put it in jeopardy. The stock fans on the Ender 3 aren't exactly high quality, and neither are many of the 5015's people use, I take it that poor fan circuitry kills this frail mosfet?

@CRTified You and @DaveJS46 seem very knowledgeable with electronics, what do you guys think we can do to avoid having the mosfets die? DaveJS46 mentioned putting a cap between the grounds of FAN1 and FAN0, but I don't really understand how this would help, could you guys can explain the principle behind this solution? Wouldn't a bypass/decoupling capacitor on the FAN0 pins be a better a solution? 

I was thinking of putting a buck style circuit between the fan header and the fan (a cap, diode, and inductor) in order to reduce PWM noise and also protect against voltage spikes (two birds one stone hopefully). Another thought I had was slapping a tiny heatsink on this mosfet, since its power dissipation is only 1.4w at 25c, which seems to be very small especially with dual 5015 blowers, or is power dissipation unrelated to this issue?

Thanks

DaveJS46, 2020-12-10:
@ymurenko 
If you look at the diagram in my 28th September comment, the capacitor is in effect across the mosfet terminals Drain and Source. A capacitor reacts instantly to absorb very fast pulses - which is what my fans produce. Diodes my well do the same job but all have a switching time so react to the pulses less quickly.
If you fit a 10uF 50V capacitor before you connect and run the fan(s) you will at least have some protection, but if your fans have different characteristics and noise generating parameters from mine then possibly the protection will not be sufficient.
As the SKR earlier version boards used the same mosfet to drive FAN0 as the heating elements and it seems without failures then maybe the best solution is to remove the PL4009 and wire in its place a WSF3085 mounted on a small stripboard or similar.

mikegleasonjr, 2020-12-10:
I did the capacitor trick to my V2 board, so far so good...

ymurenko, 2020-12-10:
@DaveJS46 Ahh, that makes perfect sense, I guess I didn't look closely enough at the schematic. If I understand correctly, since FAN1 isn't controlled, you're just using its ground pin. I was confused for a moment because I didn't realize that this revision of the board has only one controlled fan.

I'll give it a shot, I'm using the stock blower that came with the printer, but if we wanted a universal solution for all fan configurations, would a schottky diode with appropriate specs be a better solution? (I've had some issues in the past where the PSU would click due to the inrush current current from at 10uf cap)

@mikegleasonjr I don't think that's a good idea on the V2.0 board, because it has two of these pl4009 mosfet instead of one, which means two controlled fan headers. You're not connecting the capacitor to ground by connecting it to FAN1 ground, because those pins go into the second mosfet's drain. If you wanted to do this solution on the V2.0 board you would need two caps, one for FAN0 and one for FAN1, and each connected to ground somewhere else on the board.

DaveJS46, 2020-12-11:
@ymurenko I know this topic is under the E3 board but as the mosfet failure occurs on the SKR 1.4 boards as well, the problem ended up being included here and my comments do, in fact, refer to the 1.4 board and its layout. Anyone who has an E3 board needs to be aware of the differences that you outline above.
With regard to using a fast acting schottky diode, well yes, it may well work for you - it is up to you to try that route. The problem generally is that to find a solution that works for all fans one needs to test them all and as hobbiests I don't think too many of us have access to more than one or two types of fan. We can but hope that BTT will eventually design the problem out on the next iteration of the boards.
As an aside, your comment that you had a PSU that had problems charging a 10uF cap - all I can say is that I can't imagine that PSU being able to drive hardly any load and certainly not a couple of fans. Most DC PSUs have smooting caps (i.e. across the output) way in excess of 10uF.

ymurenko, 2020-12-12:
@DaveJS46 Thanks for you help, I ended up using a 10uf 50v cap since I don't seem to have any schottky diodes. As for board layouts, the FAN0 configuration on the 1.4 is slightly different than on the 1.2, but connecting the capacitor between FAN0 gnd and FAN1 gnd is effectively still a connection between drain and source, so this solution still applies to the 1.2 board. 

I had little room to solder the cap on (the back of the board sits very low), so I opted to crimp the cap's positive leg into the fan connector, and the other leg into an empty connector. I drive my board fans from the hotend fan terminals, so I didn't need FAN1 for anything.

It's been running well so far. I've tested the fans at different speeds using hardware pwm, and with marlin's `SOFT_PWM` and `SOFT_PWM_SCALE` at each frequency. Of course, I'll never know if my fan mosfet was destined to fail, but I can confirm that this solution doesn't cause any issues on my board.

[image]

As I've mentioned before, you shouldn't do this if you use a 2.0 board, because both FAN1 and FAN0 are driven by mosfets. Connect the cap's negative straight to ground elsewhere on the board.

DaveJS46, 2020-12-12:
@ ymurenko You're welcome - I'm glad that you are happy with the mod. As you rightly say we don't know if a working mosfet is destined to fail or not. However, as I did run some tests on the fans connected to my bench PSU and observed the noise pulses on my 'scope and then saw how those pulses were greately smoothed (from 9V peak to about a volt peak) so I know that the cap keeps the Vds max in spec.

mikegleasonjr, 2020-12-13:
@ymurenko

Thanks I removed the CAP from my V2 board. Everything seems to work fine, both fans are controlled normally.

ChuckNoxis, 2021-01-09:
My part cooling fan is doing the same : going on when I power on my skr mini e3 v2 and staying always on at 24v :(

[20 more replies]