bigtreetech/BIGTREETECH-SKR-mini-E3 #239: Current firmware needs to be the non 512kb. the RCT6 only has 256kb

bgobrecht, 2020-05-03:
Current firmware needs to be the non 512kb. The current chip for the SKR mini E3 v1.2 RCT6 only has 256kb

https://www.st.com/en/microcontrollers-microprocessors/stm32f103.html

paul-1, 2020-05-03:
Spec sheets are not always correct.  ST is probably making the same chip and then down labeling them.   Of course this may not work for everyone, but it works fine in my tests.  If you don't want to use 512k, then disable stuff in the firmware and compile it using STM32F103RC_btt in platform.io

Here is the conversation......

https://www.reddit.com/r/ender3/comments/dsct5x/in_fact_the_flash_size_of_skr_mini_e3_v12_is_512kb/

HVJ17, 2020-05-03:
It already says this :

> 
> If you are using the official Marlin 2.0 version, you need the following additional changes
> After opening the project, go to the platformio.ini file and change the default environment from megaatmega2560 to STM32F103RC_btt_512K, env_default = STM32F103RC_btt_512K
> If you use the latest bugfix-2.0.x change the default environment from megaatmega2560 to STM32F103RC_btt_512K, env_default = STM32F103RC_btt_512K

swilkens, 2020-05-13:
For some boards, the used STM32F103 chip does appear to have the full 512k available. But not for all.

You can modify this quickly yourself.

In `platformio.ino`, change `default_envs` to one of these 4 options: 

> STM32F103RC_btt ............. RCT6 with 256K
> STM32F103RC_btt_USB ......... RCT6 with 256K (USB mass storage)
> STM32F103RC_btt_512K ........ RCT6 with 512K
> STM32F103RC_btt_512K_USB .... RCT6 with 512K (USB mass storage)

bgobrecht, 2020-05-13:
How do we know if the RCT6 chip has it without knowing? At this point it’s a gamble to find out. And since the boards don’t have the 4 layer cob can be volatile without warning from what others have posted. How do we find out if it is in fact a 512kb board without experimentation is the key in all of this since you can’t tell looking at the chip.

swilkens, 2020-05-13:
There is no direct way to check this, as the manufacturer sometimes let the bootloader report they have 256K while they actually have 512K available.

The only sure way to know is to use a firmware flash & verification tool that can flash the binary, read it back and verify what was written. This is what this guy does: https://www.youtube.com/watch?v=q0JEx3uzgSo

For normal users though, not really an option. The safe way to go would be to default to 256K and let users themselves try up to 512K if they want.

quarky42, 2020-11-03:
I get the difference between 256k and 512k, but what is the functional difference between the ones with USB Mass Storage and the ones without USB Mass Storage?   I haven't seen that explained very well anywhere.