bigtreetech/BIGTREETECH-SKR-mini-E3 #240: SKR Mini v1.2 upload

byxox, 2020-05-03:
Hello
I have a skr mini e3 v1.2 card. I downloaded the compatible marlin on the site. I have built seamlessly with Platformio. But when I upload through the platform, I get this error.

https://prnt.sc/sacei7
https://prnt.sc/sacep8
https://prnt.sc/sacexl

what could I be missing? 

I can see the device as a serial port on the platform.
https://prnt.sc/sacfal

paul-1, 2020-05-04:
You cannot update the board over serial.    You have to use the sdcard, as described n the manual

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2manual.pdf

byxox, 2020-05-04:
Yeah. As you said, it was updated when firmware.bin was thrown on the sd card. Thank you so much.

closed 2020-05-04 by byxox