bigtreetech/BIGTREETECH-SKR-mini-E3 #241: [FR] (M300 Command Support)

Lmstaple, 2020-05-04:
On the stock Creality board, as well as the silent creality board you could execute M300 commands after enabling the speaker in your configuration file. 

Whether it is through the stock screen or the TFT35, no matter what I do I can not get any noise to sound when I enable the speaker in the config file. 

I don't have much knowledge in terms of coding, but I wonder whether there is something missing that controls the speaker enabling in the greater Marlin folder, since there is sound with a buzzer but nothing with speaker, which should be on the same pins.

HVJ17, 2020-05-05:
Speaker is disabled in Btt fw.

Lmstaple, 2020-05-12:
I understand it is disabled by default.  I reference in the above comment that I have enabled it in my config file and flashed with no luck in getting any sound.

HVJ17, 2020-05-12:
The SPEAKER is not usable through the mini 3e 1.2.

Lmstaple, 2020-05-12:
Could you elaborate a little on why this is the case? What exactly causes
this issue? In terms of hardware it has a similar buzzer to that of the
stock board, so I can only assume it’s something software side. Is there a
way to fix this?

HVJ17, 2020-05-12:
I cannot. This is a hardware choice they made. I've not seen anyone able to get it working through Marlin.

bigtreetech, 2020-06-04:
Hello, after this PR https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/796, speaker in stm32f1 should work normal (including SKR-mini-E3-V1.0/V1.2/V2.0, E3-DIP-V1.0, mini-V1.1, etc...), you need to delete the maple library for stm32f1 in "C:\Users\<your usr name>\.platformio\packages\framework-arduinoststm32-maple" (delete this folder), enable SPEAKER in Marlin's configuration.h, Then compile again, vscode will download the latest maple dependency library automatic,  compile and update, and M300 should work normally

Lmstaple, 2020-06-04:
I am not sure if I am doing something incorrect, but when I deleted the Maple Library and built the firmware it now results in no noise across touchscreen mode and LCD mode. It does not register M300 commands nor does it recognize the typical clicking noise that accompanies using the knob on the LCD mode.

I have enabled the speaker, as well as the LCD Feedback frequency with it at default values. 

Also, in the section directly below this I have the Cr-10 Display enabled. Do I need to change this?

Lmstaple, 2020-06-04:
Update: I went into the Tone.cpp and manually changed the TONE_CHANNEL 8 to TONE_CHANNEL 4 like was mentioned in one of the linked threads. I don't fully understand why it wasn't updated in the reinstall that I did upon building the firmware, but perhaps I was too impatient and it simply has not been made public yet? 

Regardless, this fixed the issue, so I will be marking this closed. Thank you to everyone involved, can't wait to get my John Cena theme back lol.

closed 2020-06-04 by Lmstaple

jackharvest, 2020-06-24:
> Tone.cpp and manually changed the TONE_CHANNEL 8 to TONE_CHANNEL 4

I don't see Tone.cpp when modifying the firmware for SKR mini 1.2. Where was this exactly?

JdeV987, 2020-06-24:
> > Tone.cpp and manually changed the TONE_CHANNEL 8 to TONE_CHANNEL 4
> 
> I don't see Tone.cpp when modifying the firmware for SKR mini 1.2. Where was this exactly?

https://github.com/MarlinFirmware/Marlin/pull/18011#issuecomment-639199335

jackharvest, 2020-06-24:
> MarlinFirmware/Marlin#18011 (comment)

Thank you!

bennymamo, 2020-10-20:
I have applied the fix to get "#define SPEAKER" working (changed file C:\Users\benny\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\tone.cpp and changed TONE_CHANNEL from 8 to 4) and speaker works. Problem is that with this setup, Part cooling fan does not work correctly. Fan only starts when I enter some menu from LCD or similar action. To confirm that might be the issue, I reverted back changes and commented out SPEAKER and part fan is now working as it should. I will keep it without speaker. Buzzer is good enough for me. Thanks for the suggestion.

trengtor, 2020-12-19:
> Hello, after this PR [rogerclarkmelbourne/Arduino_STM32#796](https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/796), speaker in stm32f1 should work normal (including SKR-mini-E3-V1.0/V1.2/V2.0, E3-DIP-V1.0, mini-V1.1, etc...), you need to delete the maple library for stm32f1 in "C:\Users<your usr name>.platformio\packages\framework-arduinoststm32-maple" (delete this folder), enable SPEAKER in Marlin's configuration.h, Then compile again, vscode will download the latest maple dependency library automatic, compile and update, and M300 should work normally

Not working: compilation error instead auto library download

yourbuddydinec, 2021-03-12:
By deleting framework-arduinoststm32-maple i was able to get the speaker to work but a single tone sounds repeated and for a long time so over and over. Anyone run into this problem?