bigtreetech/BIGTREETECH-SKR-mini-E3 #242: [BUG] don't save again the Z height !

eiscorp, 2020-05-04:


eiscorp, 2020-05-04:
I will return to the creality board and put this board to the trash ! 3 month after i buy it the update don't update whatever !

HVJ17, 2020-05-05:
Whats the issue? Did you look for assistance?

eiscorp, 2020-05-05:
I would like to know hot save the Z offset ? i have put the eeprom.dat with the last firmware on sdcard board, i do "store settings". That's change all the time ! one time it's -4,2mm, one other time it's 8,4mm and one other it'3,8mm, ....

HVJ17, 2020-05-05:
This board, btt mini e3 1.2, does not need eeprom on sd. I assume this means you build your own firmware file? 

If so, make sure you start with the btt config files. There are a few known issues with MARLIN having issues. Don't get mad at the SKR board when its just Marlin.

eiscorp, 2020-05-05:
I have tried hvj17 firmware (-z offset is -18,4mm) he don't work properly !

eiscorp, 2020-05-05:
i turn on, after i do a AutoHome. at this moment i'm downing Z-axe till the piece of paper on the bed.
I take the mesure on the display -xx,xx mm and i put it to the Z offset.

That's the good procedure ?

eiscorp, 2020-05-05:
i think with your fw it's good.