bigtreetech/BIGTREETECH-SKR-mini-E3 #244: [BUG] Loud steppers after print

Brachyceran, 2020-05-07:
Ender 3 Pro user here. Im having my SKR mini E3 V1.2 for ~ 2 weeks. I printed a variety of models - large, small, short and over 12 hours long prints. PLA and PET-G. Worked out awesome. But today i encountered a weird issue.
I started a fairly large print:
- Fiberlogy PET-G black
- 245 degrees on hot end
- 73 degrees on heated bed
- 80 mm/s printing speed,
- took ~ 8 hours.
I observed the first layer to check if it sticks to the bed well which it did. Everything was fine on ~0.5mm Z height so i went to sleep. I woke up to see the print finished and I was happy with the quality of the print. Then i went ahead to start the next print (multi part project) so while I prepared the g-code, I pre-heated the printer but after a while, thermal runaway protection triggered so I restarted the printer. After double chencking the hot end and mainboard electrical connections, I pre-heated the printer which this time went without a problem. I started the print and when homing started, i heard loud stepper motor noises. It was never present when I printed using this board before. I stopped the print and flashed the firmware using the latest .bin file in basic form (no bltouch) but it hasnt solved the issue. I looked into printer configuration and tried swiching TMC drivers StealthChop mode off which weirdly solved the issue - motors were silent as normal. This setting does not presist after turning the printer on and off and every reboot i have to change it manually.

### Steps to reproduce

1. Print a model with the abovemensioned settings
2. After it finished, pre-heat the printer to the same temperatures
3. "Thermal runaway protection triggers" - no electrical issues found
4. Start the next print
5. Loud stepper motor noises are present

**Expected behavior**
Silent operation of the stepper motors

**Actual behavior**
Loud stepper motor noises. Grinding-like sound but the printer mechanically behaves as normal (though i did only a limited amount of testing to prove this).

### Additional Information
* Provide pictures or links to videos that clearly demonstrate the issue - on request in the event of a misunderstanding.

sfranke89, 2020-05-08:
i am actually having the same issue. using version 2.0.5.3 brand new board out of the box, does not happen on my other ender 3 and that has stealthchop enable but as expected the steppers are silent on that one.

bengomez31, 2020-05-08:
I’m having the same issue also with 2.0.5.3. Stealthchop enabled results in noisy steppers. As soon as I disable them they are nice and quiet. But settings won’t stay saved. Hopefully one of us figure something out soon.

paul-1, 2020-05-08:
Just move the jumper on the board, so StealthChop is default.    Then if you want to turn off stealthchop, that seems to work fine.

bengomez31, 2020-05-08:
excuse my ignorance, I've been trying to find where it is online.Iis that the jumper that is next to the spread label on the board or the choose? 
http://www.bigtree-tech.com/shop/product/show/id/6

chazzin, 2020-05-08:
I recently built firmware from latest Marlin bugfix-2.0.x and have it running but for some reason I do not understand I had to reduce all my steps by 1/2. I am running SKR-E3-mini v1.2 board. It must be related to the stepper motor setup but can't seem to figure out what it might be. Any one have a suggestion?

paul-1, 2020-05-08:
@bengomez31 yes that is the jumper,   The picture shows the correct position for stealthchop

@chazzin drivers change, it’s best to calibrate after firmware changes.   But you might want to start a new issue with more details specific to your concern.

bengomez31, 2020-05-08:
Thanks @paul-1, yes I moved the spread jumper to the same position in the photo. when I turned the machine on it still was noisy during auto home. I then went to the stealthchop settings on the control and noticed stealthchop was now set to OFF on the control. I changed them to the ON setting, saved settings and now seams to be powering up with stealthchop working correctly. Hope this also fixes the same problem others were having.

Brachyceran, 2020-05-10:
@paul-1 I checked the jumper on the motherboard and it was not in the position you mentoned so I changed it (no changes in firmware configuration) and now its fine. Quiet as it was before. Thank you.
Though it still puzzles me how it happened in the first place that it was silent and then out of nowhere i was no longer like that.

dturner, 2020-05-12:
It seems like the latest firmware also has `HYBRID_THRESHOLD` enables which will enable SpreadCycle mode (noisy) when the feedrate is above a certain value. I recompiled the firmware with this undefined and my printer has been silent ever since. 

My config is here in case anyone is interested. https://github.com/dturner/Marlin/tree/bugfix-2.0.x/Marlin

closed 2020-05-16 by Brachyceran