bigtreetech/BIGTREETECH-SKR-mini-E3 #245: Y motor overheating

Bermit523, 2020-05-08:
### Description

### Steps to reproduce

**Expected behavior**

**Actual behavior**

### Additional Information

futnuh, 2020-05-10:
Have you looked at the actual board? I just opened mine and the Y driver heat sink had actually migrated from where I placed it. Now that could just be a symptom of it overheating ... but it can't help.

Bermit523, 2020-05-10:
> 
> 
> Have you looked at the actual board? I just opened mine and the Y driver heat sink had actually migrated from where I placed it. Now that could just be a symptom of it overheating ... but it can't help.

The stepper driver is not the problem. The Y motor is the one that is overheating.

661020786, 2020-05-12:
I also noticed the Y motor getting worryingly hot.   My E3 is currently offline, but when I bring it back on I will try to provide some concrete/reproducible scenarios.  Just wanted to echo what Bermit523 reported.

Anubis-SF, 2020-05-14:
try adjusting the current in the firmware. I had noticed it had been set higher then it should have in my version of the firmware

Bermit523, 2020-05-14:
> 
> 
> try adjusting the current in the firmware. I had noticed it had been set higher then it should have in my version of the firmware

I already tried that. Almost half of the original value and still, the only motor that was overheating was the Y. The other ones were cool.

Anubis-SF, 2020-05-15:
What printer are you running it on?

Bermit523, 2020-05-15:
> 
> 
> What printer are you running it on?

Ender 3.

Bermit523, 2020-05-15:
For all of you who wrote into this post. I'm currently running marlin 2.0 bugfix and everything is fine, no problems with overheating. But still I would like to upgrade to the version 2.0.5.3.

paul-1, 2020-05-15:
What is the commit when you pulled the bug fix branch?   The bugfix-2.0.x branch is **always** newer than the 2.0.x branch or the released version.   So 2.0.5.3 is likely a downgrade.

Anubis-SF, 2020-05-16:
That was going to be my next question. For some reason as soon as I switched over to regular Marlin my problem went completely away. After some further adjustments my machine is working amazingly well..except for it refusing to work on only usb 2.0 ports and NOT on USB 3.0 ones. Thought that's an apparent hardware problem with this style board.

Bermit523, 2020-05-16:
> That was going to be my next question. For some reason as soon as I switched over to regular Marlin my problem went completely away. After some further adjustments my machine is working amazingly [...]

Do you mean the none bugfix version?

Bermit523, 2020-05-16:
> 
> 
> What is the commit when you pulled the bug fix branch? The bugfix-2.0.x branch is **always** newer than the 2.0.x branch or the released version. So 2.0.5.3 is likely a downgrade.

I didin't know, that the bugfix is newer (should've come to my mind). And the reason I choosed the 2.0.5.3 version was, that it was prepared here on github page of the skr mini e3.

Anubis-SF, 2020-05-17:
Nope. As in I download the vanilla marlin directly,  went through and made the necessary changes, then compiled my own version. Since I have an ezabl from th3d I had to make changes anyways so figured why not.

miki3421, 2020-05-18:
Hello, I have the problem too, I've installed the firmware.bin that is the version 2.0.5.3 and for a couple of days it worked well. Today it's very noisy the Y driver and over heating the Y motor. 
What is the solution?  
Thanks

Bermit523, 2020-05-19:
Everything is fine now. Now I have the marlin bugfix version and everything is working flawlessly. But the thing that worries me is, that I didn't manage to find the thing that caused Y motor to overheat. If anyone has any suggestion, write it here.

heximcz, 2020-05-19:
https://community.octoprint.org/t/pi-setup-serial-exceptions-still/12938/17 or cut the power cord in the USB cable (my solution).

Anubis-SF, 2020-05-20:
> Everything is fine now. Now I have the marlin bugfix version and everything is working flawlessly. But the thing that worries me is, that I didn't manage to find the thing that caused Y motor to overheat. If anyone has any suggestion, write it here.

I think its a combination of things, including have both linear motion and S curve enabled at the same time (a known issue by itself). I also know there was an issue with the recent batch of boards having the spread/stepper moder mode jumper pre-set incorrectly (including the boards I got) from the factory causing a bunch of problems by itself. If you are running the board with the built in 2209's, that jumper should be set towards the power terminals and not towards the 2209 drivers. 

Other then that, I have no idea as there wasn't a bunch of changes in the code when I ran a comparison of the two (unless its something deeper I missed)

Anubis-SF, 2020-05-20:
@heximcz yeah, the usb power issue is a royal pain in the ass, even more so with usb 3. On usb 3 I couldn't even get it to work even with the power pin pin disabled.

pjones112000, 2020-06-06:
Ok, this is starting to sound like my problem.  My Y-axis motor was seriously overheating which started to whine, then eventually started grinding and not reset position correctly producing really bad output.  Turned everything off, problem went away.  Downloaded and recompiled the latest firmware (only changed the display name to Ender 3 Pro to prove that it uploaded) and then everything was fine for a time then the X-axis started having the problem and only worse (I have video with audio)....power cycling did NOT solve the problem this time.  Switched back the the Creality3D mainboard and screen and the problem appears to have disappeared.  What should the spread/stepper jumper be set to?  That might be related to my problem....none of the diagrams I've seen actually explain what the jumper settings are and where they should be, only that the exist.  I've worried that the voltage might be set too high but I'm frankly a newbie at platformio and Marlin so I'm not exactly sure where to look for those settings or if in gcode where to look.  I might try switching back to the SKR Mini mainboard to see if the jumper helps if I knew whether the jumper is in the right place now or not.

[iCloud Photos.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4740565/iCloud.Photos.zip)

cloudmj, 2021-01-22:
I have the same problem with my 2.0.7.2 version - the stepper motors heat to around 47 deg. I change vref to 560/560/670 (x,y,z) and it is better now - around 41 deg.

spooky79, 2022-04-02:
Did the author solve the problem ?  
I have exactly the same symptom on the MKS E3 Robin board (Ender 3)  tested on Marlin 2.0.9.3 and 2.0.9.1 .
Random Y-axis motor overheating, missing steps