bigtreetech/BIGTREETECH-SKR-mini-E3 #247: Lates

vitalisb, 2020-05-09:
According to the latest marlin code compiled 512k file firmware.bin (bltouch and safe home) does not flash to SKR e3 mini v1.0

txt4nk, 2020-05-14:
Reply with your printer srtup and ill compile one and see if i get the same results. If i get something that will work for you ill post it here.

krakow2000, 2020-05-16:
Got the same problem, even if i wanna compile marlin for v1.0 i got error.

txt4nk, 2020-05-16:
Are you using SKR's firmware version or vanilla marlin?

txt4nk, 2020-05-16:
also, what software are you using to compile?

krakow2000, 2020-05-16:
Atom but in visual got the same problem. Yes, SKR firmware version for 1.0

txt4nk, 2020-05-16:
What printer and mods do you have? I'll try making one and see if I get the same results. If it works, I'll post the firmware bin file for you

krakow2000, 2020-05-16:
SKR e3 mini v1.0 tmc2209, BLTouch (-45, -15, 0)

txt4nk, 2020-05-16:
My E3 mini worked amazingly when I used vanilla marlin. I'm going to try that for you.

Can you look at the board and verify the processor model? I've seen RCT6 and RET6's floating around

txt4nk, 2020-05-16:
Also, is it an ender 3? This board is cross compatible with the CR10 and Ender 5, so I need to know the model for bed sizes and such. If you have a 5, do you have the standard 5 or 5 pro? (z screw changed on the pro model, required 800 steps instead of 400)

krakow2000, 2020-05-16:
Ender 3, RET6

txt4nk, 2020-05-16:
alright, standby.

txt4nk, 2020-05-16:
Below is the vanilla marlin 2.0.x bugfix full folder (for your safe keeping and tweaking if need-be) and also the firmware.bin (in a zip file so I could upload it) compiled from the linked project.

[Marlin-bugfix-2.0.x_krakow2000.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4638710/Marlin-bugfix-2.0.x_krakow2000.zip)
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4638711/firmware.zip)

krakow2000, 2020-05-16:
ty, will inform you after test :)

txt4nk, 2020-05-16:
Looking forward to it!

krakow2000, 2020-05-16:
soooo slow auto home :) rest work correct for now

kirelam, 2020-05-16:
Hi,

I have a few settings to speed up the homing and probing. Will mail them.

Regards,
Kirelam

txt4nk, 2020-05-16:
haha yeah I didn't fine tune any of that, was more focused on making sure the firmware would flash before spending more time on it. I can speed things up.

krakow2000, 2020-05-16:
:)

txt4nk, 2020-05-16:
Awesome. Does that project file compile for you? I'd like to know what caused your firmware to not work but mine did.. could you zip the folder you were working in and upload it? I'd like to inspect for differences.

krakow2000, 2020-05-16:
yes

txt4nk, 2020-05-16:
I have a suspicion that it is the processor setting in the platformio.ini or the board name in the configuration.h

krakow2000, 2020-05-16:
nah... board name correct, change in planformio STM32F103RE_btt_USB but still nothing :)

txt4nk, 2020-05-16:
I didn't use the USB processor, I used STM32F103RE_btt. I wonder if that difference really causes the problem

krakow2000, 2020-05-16:
i can show you old btt build

txt4nk, 2020-05-16:
yes please

krakow2000, 2020-05-16:
https://drive.google.com/open?id=1Tk-Qumot-EEqdfgCvtyqdHdpbZWcdtDa

txt4nk, 2020-05-16:
The processor was wrong, the serial ports were wrong and the board is correct.

I used STM32F103RE_btt, whereas they used STM32F103RC_bigtree_NOUSB
[image]

Primary serial port is 2, secondary serial port is -1. They got them backwards.
[image]