bigtreetech/BIGTREETECH-SKR-mini-E3 #248: Latest Firmware makes Steppers Grining noise

numpty9989, 2020-05-12:
I am testing the Default firmware NO BLTOUCH... and when homing the stepper make a horrible GRINDING noise. I have trialled other firmware and it's fine so BTT how did you break the firmware

dturner, 2020-05-12:
The TMC2209 drivers can operate in 2 modes: Stealthchop (quiet) and SpreadCycle (very noisy, grinding sound - sounds like something is horribly wrong with the motors but is apparently perfectly normal). 

In the Marlin firmware configuration there's a setting called `HYBRID_THRESHOLD` which determines the feedrate at which the drivers will change from Stealthchop to SpreadCycle. Your homing feedrate is almost certainly above this rate (I think by default it's 100mm/s) and is causing the drivers to operate in SpreadCycle mode, hence the horrible noise. 

To solve this you need to compile your own firmware and comment out `HYBRID_THRESHOLD`.  

[Here's my configuration](https://github.com/dturner/Marlin/tree/bugfix-2.0.x/Marlin) which is working pretty nicely for me (I have BLTOUCH though so you'll have to change it for your own setup). 

There's also a really nice guide on configuring your [firmware here](https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/).

Grrby, 2020-05-12:
I had the same issue with with very scratchy and noisy motors during one of my the complication of 2.0.5.3. And this is not connected with HYBRID_THRESHOLD what is disabled by default configuration of BTT SKR mini E3. 
I don't remember details because of many compilations but for sure something is wrong with TMC stepper lib.
BTW 
 I use HYBRID_THRESHOLD with E3 DIP TMC2130 on my another printer and its not so noisy.

numpty9989, 2020-05-12:
I have manually  compiled my own firmware also but for some reason when i start a print and it does the nozzle prime from cura.. its like half an inch off the bed to the left  amd really strange any ideas.. was working fine months ago then needed to make a change and its been messed up ever since.. eg is there a way to wipe all the firmware off the board so its just a board then kust upload a firmware just incase anything is conflickting between firmwares... if nit then any way to fix thr miss bed issue i have even attempted to put bed size as 220 and still same its like it thinks bed is correct and its not i am screaming trying to figure it out

RevBooyah, 2020-05-12:
I got a new board (old one died after 6 months) and had a similar issue - but even putting back the old firmware (quiet on my previous e3 mini v1.2 board) was still getting the noise. It was the SPREAD jumper being in the wrong place on the new board, so StealthChop was completely disabled.   Swapping the jumper AND making sure HYBRID_THRESHOLD was commented out got rid of the noise for me.

numpty9989, 2020-05-12:
Can you send pics of what you did with the jumper.. its a new thing so am wondering if my board has died or is dying

Grrby, 2020-05-12:
You can find a description of SPREAD jumper in the board manual:
[](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2manual.pdf)

swilkens, 2020-05-13:
> I got a new board (old one died after 6 months) and had a similar issue - but even putting back the old firmware (quiet on my previous e3 mini v1.2 board) was still getting the noise. It was the [...]

Multiple reports of the same issue on the Marlin repo, it seems a batch of the boards went out the door with this jumper position.

@bigtreetech - was this jumper position change intentional or a production issue?

swilkens, 2020-05-13:
For reference, this is how it should be. look at the position of the "Spread" jumper on the left side.

[image]

numpty9989, 2020-05-13:
Mine is same as photo does that mean mine is ok???

Grrby, 2020-05-13:
Yes

Casual-NoVa, 2020-05-18:
I was experiencing the same issue. Ender 3 Pro was quiet with the mini E3 straight from the box. Added a BLTouch, so grabbed the precompiled from BTT and after flashing the noise started. Regardless of any firmware changes (precompiled or own build), as soon as the steppers started moving, the "grinding" would start. Doing a "Disable Steppers" on the printer and they were quiet again.

I checked that `HYBRID_THRESHOLD` was disabled, however my SPREAD jumper was in the incorrect position (closer to drivers rather than further away). But, changing the jumper did not resolve the noise.

Doing a bit more searching led to a similar reported issue [MarlinFirmware/Marlin/issues/13092](https://github.com/MarlinFirmware/Marlin/issues/13092) about StealthChop not being enabled.

Testing it out with `M122` or `M569` right after turning on the printer proved StealthChop was false / using SpreadCycle.

Doing the fix mentioned (`M502` to reload settings from configuration.h followed by a `M500` to save them) and things are quiet again.

Hope that helps anyone finding themselves in a similar situation.

kendoka15, 2020-06-03:
Well *expletive of your choosing*, I finally found the right thread. Thanks for the jumper tip, the noise was driving me crazy :)

galardao, 2020-06-04:
Today I received another v1.2 card, and now the problems are the noises even with the new fw the drives seem to break.

someone got the solution

Silvanor42, 2020-09-03:
> I was experiencing the same issue. Ender 3 Pro was quiet with the mini E3 straight from the box. Added a BLTouch, so grabbed the precompiled from BTT and after flashing the noise started. Regardless [...]

Thank you!!!
You saved me a lot of headache