bigtreetech/BIGTREETECH-SKR-mini-E3 #249: [BUG] (short description)

Masterzior, 2020-05-13:
### Description

Bltouch attached to SKr mini e3 1.2 is giving me black holes and weird values

### Steps to reproduce

Just normal bedleveling. I use Bedvizualizer plugin in octoprint.

**Expected behavior**
Not to give me these dips

**Actual behavior**
Sometime it gives values that are absolutely insanely off . gotten worse over time

### Additional Information
I can zip my config if needed. Been using 2.0.5 but recently went back to 2.0.3 after ghostlycrowd said it worked for him . But didnt work for me. Ive tried replacing cable aswell but i think its firmware cause it makes no sense
[image]

Thinkersbluff, 2020-05-13:
Any chance you are using the BLTOUCH HS mode?  I had to disable that to get the probe to read correctly consistently.

Reply-to: reply@reply.github.com
Cc: subscribed@noreply.github.com
Subject: [bigtreetech/BIGTREETECH-SKR-mini-E3] [BUG] Bltouch Gives off weird values and Dips (#249)

[Jzed50c]<https://user-images.githubusercontent.com/23239792/81848470-766f8400-9555-11ea-873b-44b79c3f7396.jpeg>

Masterzior, 2020-05-13:
> Any chance you are using the BLTOUCH HS mode? I had to disable that to get the probe to read correctly consistently.

How do i check if im using HS mode? I dont recall enabling that put ive dug myself pretty deep into the hole so maybe?

Masterzior, 2020-05-13:
Checked my firmware, i do not have HS mode enabled

Thinkersbluff, 2020-05-13:
I also disabled the "endstops are interrupt-capable" setting in my firmware, not being sure about the SKR mini.

Only other issues I have had with my BLTOUCH I traced to physical issues:

  1.  I discovered one BLTouch mounting screw had worked loose. The pin was not perfectly vertical, and got stuck at some test points, delaying the trigger event.  I could see when the probe hesitated and I manually edited the mesh to fix those points.
  2.  I had not pushed the heatbreak far enough into the finned cooling body of my Micro Swiss extruder, and the nozzle was fouling the bed before the probe could trigger.
  3.  I messed-up the extension harness and one repaired wire was intermittent, as the extruder carriage moved left/right.

Otherwise, my BLTouch has worked flawlessly, whether configured as the Z-min switch or as a probe.

Good luck.

Cc: badback42@outlook.com; comment@noreply.github.com

heximcz, 2020-05-13:
Check wire connection to SKR. Dupont connector is too high for good contact. I connected every pin individually and I dont have a problem. 
[image]

dturner, 2021-03-17:
@ViktorLindgren95 Did you ever resolve this issue? I am having the same problem: 

[image]

ViktorLindgren95, 2021-03-17:
I think tired alot of things . I think i decided it was a mechanical error
and fastened the bed , double checked the axis etc . I don't really
remember what fixed it I'm sorry . I could check my notes but i know it
eventually fixed itself . It might have ended up going back to an old
version .

dturner, 2021-03-17:
This Marlin issue looks to be the same https://github.com/MarlinFirmware/Marlin/issues/18526. Maybe you updated the firmware?

ViktorLindgren95, 2021-03-17:
> This Marlin issue looks to be the same MarlinFirmware/Marlin#18526
> <https://github.com/MarlinFirmware/Marlin/issues/18526>. Maybe you
> updated the firmware?

I can check when I get home. But check firstly make sure your bed is not
wobbling . I remember my did that slightly and would explain the weird dips
. Is the dip consistent or at random spots !?

dturner, 2021-03-17:
Thanks. The bed isn't wobbling, and the dip is at random points. Sometimes it works perfectly for all 9 probe points, sometimes (say 1 in 3 times) 1 or 2 probe points are far too deep.

ViktorLindgren95, 2021-03-17:
Is it always on the edges ? Check that your x axis is even. If not what's
your current version etc . I might still have my config file

dturner, 2021-03-17:
I upgraded from Marlin 2.0.5 to 2.0.7.2 and the problem seems to have been fixed, almost certainly due to this PR: https://github.com/MarlinFirmware/Marlin/pull/18637. 

So to anyone who runs into this issue, just upgrade Marlin :)