bigtreetech/BIGTREETECH-SKR-mini-E3 #252: Failing to compile v1.2 firmware

clfaye, 2020-05-14:
I get the following "file not found error" when I attempt to compile:

FileNotFoundError: [Errno 2] No such file or directory: 'G:\\User Folders\\Downloads\\BIGTREETECH-SKR-mini-E3-master\\BIGTREETECH-SKR-mini-E3-master\\firmware\\V1.2\\Marlin-2.0.x-SKR-Mini-E3-V1.2\\.pio\\libdeps\\STM32F103RC_btt_512K\\_tmp_installing-lall0_9x-package\\.\\examples\\StrandtestArduinoBLECallback\\StrandtestArduinoBLECallback.ino':
  File "C:\Users\fhe55\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\fhe55\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\fhe55\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\fhe55\.platformio\platforms\ststm32\builder\main.py", line 98:
    target_elf = env.BuildProgram()
  File "C:\Users\fhe55\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\fhe55\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\fhe55\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\fhe55\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\fhe55\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\fhe55\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 1036:
    project.install_dependencies()
  File "C:\Users\fhe55\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 868:
    lm.install(uri)
  File "c:\users\fhe55\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 275:
    pkg_dir = BasePkgManager.install(
  File "c:\users\fhe55\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 700:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "c:\users\fhe55\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 126:
    return self._install_from_url(
  File "c:\users\fhe55\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 482:
    self.unpack(dlpath, tmp_dir)
  File "c:\users\fhe55\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 222:
    return fu.unpack(dest_dir, with_progress=False)
  File "c:\users\fhe55\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 149:
    self._unpacker.extract_item(item, dest_dir)
  File "c:\users\fhe55\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 82:
    super(TARArchive, self).extract_item(item, dest_dir)
  File "c:\users\fhe55\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 39:
    self._afo.extract(item, dest_dir)
  File "c:\users\fhe55\.platformio\penv\lib\tarfile.py", line 2065:
    self._extract_member(tarinfo, os.path.join(path, tarinfo.name),
  File "c:\users\fhe55\.platformio\penv\lib\tarfile.py", line 2137:
    self.makefile(tarinfo, targetpath)
  File "c:\users\fhe55\.platformio\penv\lib\tarfile.py", line 2178:
    with bltn_open(targetpath, "wb") as target:

txt4nk, 2020-05-14:
What is your setup? I'll compile one for you.

jimmino, 2020-05-15:
It is written for missing SConscript.
Install Pytohn 3.8. Than download and install PIP. Use pip to install scons.

clfaye, 2020-05-15:
Thanks jimmino.  I have Python installed.  It looks like PIP is a standard part of Python.  I cannot figure out how to use PIP to install SConscript.

jimmino, 2020-05-15:
To check if pip is on, in cmd, run: (I expect you have python in PATH)
pip --version
If you see some, you are ok. To install scons run
pip install scons
Reopen VScode and you should be good to go.