bigtreetech/BIGTREETECH-SKR-mini-E3 #254: firmware updating starting on each boot

kn-dima, 2020-05-17:
bootloader upload firmware after each reset (power on) if firmware.bin present on card

**steps to reproduce**
1. Compile marlin with USB mass storage enabled
2. Write firmware.bin to stock 128Mb sdCard
3. Insert card into board
4. Power on board
Firmware uploaded and seems works.
USB drive appears in windows

**Expected behavior**
firmware.bin must be renamed to firmware.cur
next boot must be faster than at step 4 (about 10 sec)

**Actual behavior**
firmware.cur file with same size and date as firmware.bin appears at root of sdcard.
firmware.bin not deleted.
next boot take same time as at step 4 (about 10 sec)
firmware.cur disappears from sdcard (and appears after next reset/power on)

### Additional Information
actually i am using mini v1.1 board with mini e3 bootloader from [there](https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/issues/7#issuecomment-515892603)
because mini v1.1 bootloader have same issue.

tried two 128Mb cards with same result
tried skr v1.3 and it havent that problem
after manually deleting firmware.bin, boot takes about 3 sec.

thenewdays, 2020-05-18:
Yes I am having the same issue with skr mini 1.2 e3. Also need to make an eeprom. Dat file for it to save settings. All seems to be working good so I delete the firmware. Bin file after firmware. Cur is there.

txt4nk, 2020-05-18:
I've had this issue in the past, I resolved it by formatting the SD card before placing the new firmware binary file on it. I do this for all firmware updates now, it may be something worth adding to your standard firmware update procedure as well (if this corrects your problem, of course).

kn-dima, 2020-05-19:
> resolved it by formatting the SD card
Thank you!
It works for me too, but only with Mini E3 bootloader.
Mini v1.1 bootloader stil cant delete firmware.bin even after format.