bigtreetech/BIGTREETECH-SKR-mini-E3 #255: Missing firmware for SKR Mini E3 V2.0

JeremyKennedy, 2020-05-21:
This isn't quite a bug report, nor is it a support request. However I just received a SKR MINI E3 V2.0 and would like to tweak the firmware, however there is no stock `firmware.bin` file, and no example `Marlin` configuration. I tried tweaking the base Marlin config using a guide for a V1.2 board, but I got stuck with a blank screen on boot, so I think there must be some other changes needed to the config. This repo should have the Marlin config for V2.0, just like it has one for V1.2.

monstis, 2020-05-21:
Same for me. Damn Bltouch! I have tried so many variations to get a firmware that works. My v1.2 worked like a charm till2 weeks ago. The facebook page has a link to the firmware. But its just a bunch of pdf's
has anybody had success with getting any firmware onto the v2.0?

monstis, 2020-05-21:
As I wrote my previous comment. v2.0 firware appeared. I will try this!

JeremyKennedy, 2020-05-21:
Yay! This appears to be resolved by commit 8962900 here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0

monstis, 2020-05-21:
Indeed. Flashed and working fine.

JeremyKennedy, 2020-05-21:
After a quick look through the repo, looks like BTT added a new board config: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/Marlin-2.0.x-SKR-mini-E3-V2.0/Marlin/src/pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h

I'm going to try copying this into the base Marlin config and see if that works. I'd rather use the official Marlin repo as a base than this one, for ease of updating.

RepairSyl, 2020-05-22:
Hi the firmware firmware-bltouch.bin, for the cable conection,  use z min endstop or the all probe conection

thisiskeithb, 2020-05-24:
This board should be supported upstream in the main Marlin repo soon: https://github.com/MarlinFirmware/Marlin/pull/18088

And the config: https://github.com/MarlinFirmware/Configurations/pull/111

hadukennl, 2020-05-24:
> Hi the firmware firmware-bltouch.bin, for the cable conection, use z min endstop or the all probe conection

Neither works for bl touch

JeremyKennedy, 2020-05-25:
> This board should be supported upstream in the main Marlin repo soon: [MarlinFirmware/Marlin#18088](https://github.com/MarlinFirmware/Marlin/pull/18088)
> 
> And the config: [MarlinFirmware/Configurations#111](https://github.com/MarlinFirmware/Configurations/pull/111)

This is great news, thank you! I will test this out this week. 

Does the config you've posted differ at all from the config in this repo?

thisiskeithb, 2020-05-25:
> Does the config you've posted differ at all from the config in this repo?

Probably, but I didn’t compare them.

It’s based on the [SKR Mini E3 configs](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/BigTreeTech/) that I’ve already contributed to the Marlin Configurations repo (which are based on the stock Ender-3 config).

shyt47, 2020-05-28:
So i tried to flash the Bl touch bin and now my ender 3 pro just has a blank screen.

marcjwebb, 2020-05-28:
hello, what board do we choose in marlin for vscode? bltouch isnot working and bltouch bin file does nothing

marcjwebb, 2020-05-28:
also, is there a bootloader on this board as standard? tried one of the folders from one of the users here, however, seemed to act like there was no bootloader

HyperbolicTurtle, 2020-05-30:
I tried this, but my printer is just ramming into the end stop when I auto home, as if the end stop is not working. It works perfectly with the old board though.

Any suggestions?

hadukennl, 2020-05-30:
> I tried this, but my printer is just ramming into the end stop when I auto home, as if the end stop is not working. It works perfectly with the old board though.
> 
> Any suggestions?

Where are you bltouch and or zstop switch plugged into the v2 board?

afcrab, 2020-06-06:
> > I tried this, but my printer is just ramming into the end stop when I auto home, as if the end stop is not working. It works perfectly with the old board though.
> > Any suggestions?
> 
> Where are your bltouch and/or zstop switch plugged into the v2 board?

Tearing my hair out with this also with my Ender 3 pro. I have the BLTouch (clone that works fine on the original creality board) initialising (deploying and stowing when powered on and lights up red) but when I try to auto home it doesn't deploy and bottoms out.
The red z-stop two pin connector is plugged into the z-stop port next to the y-stop port and the three pin is connected using the middle 3 pins adjacent to the lcd port.
I have tried both the BTT BLTOUCH firmware and BLTOUCH with Z homing firmware.

I'm new to this folks so be gentle with me :)

[image]

txt4nk, 2020-06-06:
afcrab, you need to switch the blue and red wires in the dupont connector. Looking at the board in that same orientation, the colors should be yellow (top), red (middle), blue (bottom).

HyperbolicTurtle, 2020-06-13:
I'm not using any additional features such as BLTouch and everything is plugged in exactly where it was on the old board/according to the labels on the new board.

JeremyKennedy, 2020-06-13:
Yeah I didn't do any wire cutting or anything like that. I plugged my
BLTouch into the dedicated port as shown in pictures on this repo, but put
the two pin connector into the z-endstop plug, and just completely
unplugged that endstop. Of course I made the relevant configuration changes
in Marlin.

I also learned that the fan ports are swapped on the SKR vs the Ender,
specifically the part cooling fan and the board/controller fan. Test out
manual fan controls and you will probably see that if you plugged things
into the exact same ports, they need to be swapped.

JeremyKennedy, 2020-06-13:
@afcrab do you have `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` enabled?