bigtreetech/BIGTREETECH-SKR-mini-E3 #258: Failing to compile v2.0 firmware

portcqb, 2020-05-22:
SKR mini E3 v2.0 wont compile and throws these errors before failing in latest version of VSCode with PlatformIO 

In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:1:0:
Compiling .pio\build\STM32F103RC_btt_512K\lib061\USBComposite\USBMIDI.cpp.o
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
Compiling .pio\build\STM32F103RC_btt_512K\lib061\USBComposite\USBMassStorage.cpp.o
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
Compiling .pio\build\STM32F103RC_btt_512K\lib061\USBComposite\USBXBox360.cpp.o
                                           ^~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\STM32F103RC_btt_512K\lib061\USBComposite\usb_composite_serial.c.o
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_BusIO_Register.h:2:0,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp:1:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':     
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:160:79: error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'
   size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
                                                                               ^
In file included from C:\Users\cires\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire.h:42:0,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:1:
C:\Users\cires\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate: uint8 WireBase::requestFrom(uint8, int)
     uint8 requestFrom(uint8, int);
           ^~~~~~~~~~~
C:\Users\cires\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note:   candidate expects 2 arguments, 3 provided
C:\Users\cires\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate: uint8 WireBase::requestFrom(int, int)
     uint8 requestFrom(int, int);
           ^~~~~~~~~~~
C:\Users\cires\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note:   candidate expects 2 arguments, 3 provided
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':       
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:99:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'
     _spi->transfer(buffer, len);
                               ^
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:0,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:1:
Marlin\src\HAL\STM32F1/SPI.h:268:11: note: candidate: uint8_t SPIClass::transfer(uint8_t) const
   uint8_t transfer(uint8_t data) const;
           ^~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:42:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
*** [.pio\build\STM32F103RC_btt_512K\lib2a5\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o] Error 1
In file included from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:55:0,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope
                      BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                                           ^~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1/SPI.h:268:11: note:   candidate expects 1 argument, 2 provided
.pio\libdeps\STM32F103RC_btt_512K\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:115:23: error: 'SPI_BITORDER_LSBFIRST' was not declared in this scope
     if (_dataOrder == SPI_BITORDER_LSBFIRST) {
[... 28 lines cut ...]
                 from C:\Users\cires\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/avr/pgmspace.h:4,
                 from C:\Users\cires\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/WString.h:29,
                 from C:\Users\cires\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:47,
                 from C:\Users\cires\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:50,
                 from .pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:17:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.h:84:32: error: call to 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*)' uses the default argument for parameter 3, which is not yet defined
   Adafruit_SPIDevice spi_dev = NULL;
                                ^
*** [.pio\build\STM32F103RC_btt_512K\lib2a5\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\lib2a5\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp.o] Error 1
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp: In constructor 'Adafruit_MAX31865::Adafruit_MAX31865(int8_t, int8_t, int8_t, int8_t)':
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:38:76: error: call to 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, int8_t, int8_t, 
int8_t, uint32_t, BitOrder, uint8_t)' uses the default argument for parameter 6, which is not yet defined
   spi_dev = Adafruit_SPIDevice(spi_cs, spi_clk, spi_miso, spi_mosi, 1000000);
                                                                            ^
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp: In constructor 'Adafruit_MAX31865::Adafruit_MAX31865(int8_t)':
[... cut ...]
       Adafruit_SPIDevice(spi_cs, 1000000, SPI_BITORDER_MSBFIRST, SPI_MODE1);
                                           ^~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_btt_512K\Adafruit_MAX31865\Adafruit_MAX31865.cpp:49:43: note: suggested alternative: 'SPI_CR1_LSBFIRST'
       Adafruit_SPIDevice(spi_cs, 1000000, SPI_BITORDER_MSBFIRST, SPI_MODE1);
                                           ^~~~~~~~~~~~~~~~~~~~~
                                           SPI_CR1_LSBFIRST
*** [.pio\build\STM32F103RC_btt_512K\liba83\Adafruit_MAX31865\Adafruit_MAX31865.cpp.o] Error 1

FlyingT, 2020-05-22:
Already posted:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/257

FlyingT, 2020-05-22:
Found a Fix, comment out this:
https://prnt.sc/slxnu1

portcqb, 2020-05-22:
with the line commented out 2.0 still seems to fail with those errors so something else must be making mention to them

soothsayer011, 2020-05-22:
> 
> 
> with the line commented out 2.0 still seems to fail with those errors so something else must be making mention to them

clear your .pio directory and compile again

portcqb, 2020-05-22:
ok found the second issue in addition to commenting out the above line this needs to be changed from 
#Adafruit NeoPixel
  https://github.com/bigtreetech/Adafruit_NeoPixel

to
Adafruit_NeoPixel=https://github.com/bigtreetech/Adafruit_NeoPixel/archive/master.zip

turbogreen, 2020-07-29:
> Found a temporal workaround, comment out this:
> https://prnt.sc/slxnu1
OMG TY TY TY TY..i was finally able to compile the firmware..why did i have to do this though? i dont understand python or coding at all. im just googling errors as i go lol

CMendy, 2020-08-23:
https://github.com/CMendy/MarlinForBTT_SKR-Mini_E3_V2
I just created this repo with up to date Marlin firmware

oliver01000, 2020-08-27:
> https://github.com/CMendy/MarlinForBTT_SKR-Mini_E3_V2
> I just created this repo with up to date Marlin firmware
> 
> Its the firmware I am using on my SKR-mini E3 V2.0

I actually registered to GitHub just to say thank your for this link :)
I was searching for a solution for the entire day now and your Firmware works perfekt 👍

dodgedoc, 2020-08-28:
> https://github.com/CMendy/MarlinForBTT_SKR-Mini_E3_V2
> I just created this repo with up to date Marlin firmware
> 
> Its the firmware I am using on my SKR-mini E3 V2.0

I downloaded and tried to get the firmware to work and it failed with same error

CMendy, 2020-08-28:
Did you build with vs-code and platformIO ? because that code base in my
repo is correct, and working for others as well.

dodgedoc, 2020-08-28:
CMendy,
When you import the code into PlatformIO, what board do you select?

CMendy, 2020-08-28:
I use STM32F103RC_btt_512K
its also default in platformio.ini

CMendy, 2020-08-28:
dodgedoc, if you are having a problem with my code base, open an issue there:
https://github.com/CMendy/MarlinForBTT_SKR-Mini_E3_V2