bigtreetech/BIGTREETECH-SKR-mini-E3 #259: [v2.0 BLTouch] (Homing doesn't work)

hadukennl, 2020-05-24:
Just received a 2.0 e3 mini and installed the firmware. 

Many options missing compared to 1.2 firmware, but most importantly BL Touch as Z homing doesn't work.

Tried connecting to z-stop pin and using dedicated probe pins

polgaro, 2020-05-25:
Maybe i can help you...... there are a couple of things i had to do on top of what they said.

in platform.ini, you have to change line 32 to this:
Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip

In configuration.h you have to enable safe Z homing, for me is in line 1349:
#define Z_SAFE_HOMING 

I assume you already did the following as it's in the guide:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

The way it's configured, I had to disable MESH_BED_LEVELING and enable BILINEAR:

#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING ==> this was enabled 

Now, you have a BLTouch 2.0... some advanced configurations for the BLTouch are in the configuration_adv.h

you can switch to BLTOUCH_FORCE_SW_MODE and try

hadukennl, 2020-05-26:
Thank you, I'm trying to use the firmware.bin bltouch already compiled. It keeps trying to use the zsensor instead of the bltouch for z homing

hadukennl, 2020-05-26:
New firmware "z homing" uploaded, this appears to be resolved

closed 2020-05-26 by hadukennl