bigtreetech/BIGTREETECH-SKR-mini-E3 #260: SKR Mini e3 V2.0 temperature starts raising up and does not stop

pedroapq, 2020-05-24:
I just replaced my SKR Mini e3 V1.2 for V2.0. Powered on the printer and validate everything was running. Then I downloaded and updated the firmware for BLT from https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0. After update the hot end and heated bed starting to heat up and did not stopped.

I tested my original board, and works. Tested my old V1.2 and works. 

I read about issues with the hardware issues and temperature and ordered a replacement, Got the replacement.  I proceed to replaced the board with a new V2.0 and installed everything and validated. After that I updated the firmware and now only the bead heats up and does not stop.

Since I did not try the software that came originally with the board I cannot say it is hardware or software issues. Is there somewhere I can download the original firmware that came with the board to test if it is hardware or software issues?

soothsayer011, 2020-05-27:
Have you tried loading the firmware.bin in the v2 directory?

GitAPotato, 2020-08-04:
I also have this issue. All my other boards work 1.2, & 1.3 but this one.

OsamOBam, 2020-09-15:
I also got this problem and i checked that Adafruit_MAX31865 is not commented out.
But i do not know what is meant with loading the firmware.bin in the v2 drectory.
My printer reads out the temps for the hot end and the Bed but they are both heating up.
Do you know any way to fix it?

arnaduga, 2020-12-05:
Same kind of issue here, on a Ender 5.

Bed is heating up to 50 °C. Then, it does not heat anymore (target 60°C) and then, of course, it stopped and displaying BED HEATING FAILED.

I used the pre-compiled firmware integrating Marlin 2.0.7.2.

Works fine with the orginal motherboard.

jdrozdz, 2021-01-02:
> Same kind of issue here, on a Ender 5.
> 
> Bed is heating up to 50 °C. Then, it does not heat anymore (target 60°C) and then, of course, it stopped and displaying BED HEATING FAILED.
> 
> I used the pre-compiled firmware integrating Marlin 2.0.7.2.
> 
> Works fine with the orginal motherboard.

I had the same issue to sorted that comment `#define PIDTEMPBED` on line 519 and uncomment two lines below `#define BED_LIMIT_SWITCHING` and now works perfect
[image]

arnaduga, 2021-01-03:
Thanks @jdrozdz for your help. I did not jump to a custom firmware compilation yet, but ... I may try soon :)

In the meantime, I used a french version of the firmware pre-compiled (ender5.fr) but I do not know if they change this settings or not. I'll try to figure it out.

lfzamora81, 2021-05-23:
I have the same issue with an Ender 3. Bought SKR Mini E3 V2.0, installed it, turned it on and immediately noticed the board heating up. I thought I had crossed the HB wires and so swapped them back and forth. The only difference is the bed heats up faster with the wires installed the wrong way but it inevitably heats up without stopping either way. The monitor also does report that it knows the temp is going up but there's not way to stop it. Will try compiling custom firmware and tweak around a bit to see if I can fix it.

lfzamora81, 2021-05-23:
> I have the same issue with an Ender 3. Bought SKR Mini E3 V2.0, installed it, turned it on and immediately noticed the board heating up. I thought I had crossed the HB wires and so swapped them back [...]

So it looks like it was a wiring issue, not an issue with the board. I wondered why the fan wasn't going, not remembering that it gets constant power. So the bed got constant power and the fan none. Paid extra attention to the wires, actually following them from the bed, extruder, fan, etc. along to the board.