bigtreetech/BIGTREETECH-SKR-mini-E3 #262: [BUG] (Failed to Compile, [XYZ]_HOME_BUMP_MM now called HOMING_BUMP-MM)

matt142, 2020-05-25:
I'm new here so a little help please. It looks like one of my files is out of date but I don't know where the correct file is and or how to fix this. The description section is a copy and paste of my build failure. After the description is as many details as I can provide. 

### Description

In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\STM32F1\dogm\../../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:21:
Marlin\src\HAL\STM32F1\dogm\../../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
    ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1420:25: error: 'HOMING_BUMP_MM' was not declared in this scope
 constexpr float hbm[] = HOMING_BUMP_MM;
                         ^~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1420:25: note: suggested alternative: 'X_HOME_BUMP_MM'
 constexpr float hbm[] = HOMING_BUMP_MM;
                         ^~~~~~~~~~~~~~
                         X_HOME_BUMP_MM
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1421:1: error: static assertion failed: HOMING_BUMP_MM requires X, Y, and Z elements.
 static_assert(COUNT(hbm) == XYZ, "HOMING_BUMP_MM requires X, Y, and Z elements.");
 ^~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1422:1: error: non-constant condition for static assertion
 static_assert(hbm[X_AXIS] >= 0, "HOMING_BUMP_MM.X must be greater than or equal to 0.");
 ^~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1423:1: error: non-constant condition for static assertion
 static_assert(hbm[Y_AXIS] >= 0, "HOMING_BUMP_MM.Y must be greater than or equal to 0.");
 ^~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1420:25: error: 'HOMING_BUMP_MM' was not declared in this scope
 constexpr float hbm[] = HOMING_BUMP_MM;
                         ^~~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1420:25: note: suggested alternative: 'X_HOME_BUMP_MM'
 constexpr float hbm[] = HOMING_BUMP_MM;
                         ^~~~~~~~~~~~~~
                         X_HOME_BUMP_MM
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1421:1: error: static assertion failed: HOMING_BUMP_MM requires X, Y, and Z elements.
 static_assert(COUNT(hbm) == XYZ, "HOMING_BUMP_MM requires X, Y, and Z elements.");
 ^~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1422:1: error: non-constant condition for static assertion
 static_assert(hbm[X_AXIS] >= 0, "HOMING_BUMP_MM.X must be greater than or equal to 0.");
 ^~~~~~~~~~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1423:1: error: non-constant condition for static assertion
 static_assert(hbm[Y_AXIS] >= 0, "HOMING_BUMP_MM.Y must be greater than or equal to 0.");
 ^~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:40:0,
                 from Marlin\src\HAL\STM32F1\eeprom_sdcard.cpp:30:
[... 113 lines cut ...]
 static_assert(hbm[Y_AXIS] >= 0, "HOMING_BUMP_MM.Y must be greater than or equal to 0.");
 ^~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_wired.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_sdcard.cpp.o] Error 1

### Steps to reproduce

1. Using VSCode with latest PlatformIO. as of 5/24/2020
2. Download latest Marlin and configurations. as of 5/24/2020
3. Opened Ender 5 configuration and tried to build it.
4. Errors indicate that one of my configuration files has changed and dose not match. 

**Expected behavior**
Should build and be ready for .bin upload to my SD card for use. 

**Actual behavior**
Failed to build do to above errors

### Additional Information

Marlin version I am using Bug Fix 2.0.x
Configuration I am using Ender 5 2.0.5

What did I do wrong?

konmer, 2020-05-26:
Same here, when compiling from the current Marlin branch

Rate93, 2020-05-27:
A similar problem, is there a solution?

konmer, 2020-05-27:
After Update Platformio ( Updates -> Update all )Firmware Build Succeeded

Depz42, 2020-06-02:
I got this to work by opening configuration_adv.h and changing this:
`// Homing hits each endstop, retracts by these distances, then does a slower bump.

//#define X_HOME_BUMP_MM 5

//#define Y_HOME_BUMP_MM 5

//#define Z_HOME_BUMP_MM 2

#define HOMING_BUMP_MM      { 5, 5, 2 }`

MACKNUS-INC, 2020-06-03:
> I got this to work by opening configuration_adv.h and changing this:
> 
> ```
> // Homing hits each endstop, retracts by these distances, then does a slower bump.
> 
> //#define X_HOME_BUMP_MM 5
> 
> //#define Y_HOME_BUMP_MM 5
> 
> //#define Z_HOME_BUMP_MM 2
> 
> #define HOMING_BUMP_MM      { 5, 5, 2 }
> ```

This worked for me! Thanks bud