bigtreetech/BIGTREETECH-SKR-mini-E3 #263: Schematics of SKR mini E3 v2.0 is missing

riaqn, 2020-05-25:
what the title says.

txt4nk, 2020-05-26:
You mean the pin diagram?

riaqn, 2020-05-26:
I mean this:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20mini%20E3%20V1.2sch.pdf

txt4nk, 2020-05-26:
I was unaware they even offered those, that'd be nice to have handy. Guess its just a waiting game at this point.

Coffee0297, 2020-06-25:
@bigtreetech please upload schematics of this board. i wanna know if you have fixd the issues that the 1.2 had. 
i dont care if you pm me it or upload it. i just need it asap. thanks

kind reagards Tonny Møller

saikhurana98, 2021-06-23:
I accidentally fried my skr board it's not really a cheap replacement from where I come from so it'll be nice if there's some help available. @bigtreetech please post the schematics for V2. it'll be very usefull.

Cheo7811, 2021-06-23:
People tend to mix the models.

SKR E3 Mini V2.0 diagrams is here :
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0SCHpdf.PDF

SKR 2 diagram is here: 
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0SCHpdf.PDF