bigtreetech/BIGTREETECH-SKR-mini-E3 #266: Ender 5 with SKR Mini e3 V2.0 - Z direction is flipped

sdjerdj, 2020-05-28:
### Description
Replaced the stock Creality board with the SKR mini E3 v2.0 on my Ender 5.
After the install used the firmware for Ender 5 (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-ender5.bin)
XY movement and auto-homing for X and Y axes worked as expected however the Z movement is inverted.

### Steps to reproduce
Install the firmware from https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-ender5.bin

**Expected behavior**
As part of auto-homing the bed is supposed to go down a bit then up until the Z stop switch activates 

**Actual behavior**
Bed goes down until it hits the control box

txt4nk, 2020-05-28:
I can compile a firmware for you in the morning that inverts the motor. Please post a detailed description of your printer setup (including all mods and whatever info goes with them ie: bltouch, include offsets)

sdjerdj, 2020-05-28:
This is a stock Ender 5. No bl touch, Only mod are the PSU, bowden tube (none of this is affecting the firmware :) ) and now of course the new motherboard.

cmer, 2020-05-28:
@txt4nk can you please compile every version with this fix? same issue here, but I have a bltouch.

txt4nk, 2020-05-28:
sure I can do one for both of you. stand by.

txt4nk, 2020-05-28:
I need to know if your ender 5's are pro's or not. The pro has an updated lead screw design that needs a firmware change.

cmer, 2020-05-28:
I have an Ender 5 standard. 

BLTouch offset:

`define NOZZLE_TO_PROBE_OFFSET { -45, -14, -1.4 }`

sdjerdj, 2020-05-28:
I have a standard Ender 5 but it came with the upgraded z axis lead screw (800 steps) which I believe is the way going forward. Even if this is off it can be easily fixed through a menu or a M92 command

txt4nk, 2020-05-28:
Alright, I'll make both of yours with the correct settings as soon as the bugs get worked through. I may reverse engineer some of BTT's settings tonight if I find the free time.

mysticprysm, 2020-05-30:
Hey so I'm extremely new with messing with firmware and everything software side except for the slicer, but I've installed my 2.0 board in an ender 3 pro, and I'm having some issues with the BL touch firmware. Anyway you could help me out?

txt4nk, 2020-05-30:
Sure, whats the issue?

heatherlynn1962, 2020-06-01:
Having the flipped z access issue with ender 5 as well. It was working until I reflashed the firmware for BOROUGH to work for z stop. I tried to go back but the other firmware did the same. Any idea what happened?

txt4nk, 2020-06-01:
Yeah, there's a trigger in the firmware to invert the direction of rhe z motor.

txt4nk, 2020-06-01:
This is the section that inverts the axis directions. The setup below is from my personal Ender 5 Pro, and it is functioning as it should in regards to the stepper direction. If these settings were to make the Z axis move the wrong way, you would say false on #define INVERT_Z_DIR.

// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR true

// @section extruder

// For direct drive extruder v9 set to true, for geared extruder set to false.
#define INVERT_E0_DIR true

sdjerdj, 2020-06-01:
@txt4nk Thank you for all the investigation. I managed to recompile the firmware base on the BTT repo with some customization for my E5. 
That said someone needs to recompile and update the firmware (based on the BTT repo) for the E5-s so less experienced people can start using the V2.0 board in a plug and play manner.

txt4nk, 2020-06-01:
As soon as bugfix 2.0.x works for the V2, I'll gladly make some good working bin files for yall if its needed.

kakos73, 2020-06-07:
hello, is it possible to have configuration files( configuration.h configuration_adv.h etc) with the .bin ?
it's for learn and adapt....

cmer, 2020-06-10:
Hi what’s the status on this? I’d like to install my board. Thanks.

txt4nk, 2020-06-10:
Let me download the latest bugfix right now to see if it compiles

mysticprysm, 2020-06-10:
Works for me. Got it all figured out. At least on an ender 3 pro.

txt4nk, 2020-06-10:
@cmer Hey Carl, here is the firmware binary file, and also the entire project folder in case you want to change settings or teach yourself how to compile. 
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4761793/firmware.zip)
[Marlin-bugfix-2.0.x_E3P, MINI E3 V2.0, BLT.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4761794/Marlin-bugfix-2.0.x_E3P.MINI.E3.V2.0.BLT.zip)

cmer, 2020-06-10:
Amazing thanks!

txt4nk, 2020-06-10:
no problem, let me know how it works for you.

txt4nk, 2020-06-11:
@CMER I mis-read comments, what I posted is for an ender 3. Give me a few to make a new file for you.

txt4nk, 2020-06-11:
@cmer **these** should do the trick  lol sorry about that.

[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4761868/firmware.zip)
[Marlin-bugfix-2.0.x_CE5, MINI E3 V2.0, BLT.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4761869/Marlin-bugfix-2.0.x_CE5.MINI.E3.V2.0.BLT.zip)

cmer, 2020-06-12:
@txt4nk thanks for the firmware. Oddly, Marlin seems to think I have a tiny bed? I have tried recompiling myself with your config files but I get the same thing...

The line you see on this image is supposed to go all around the bed, but somehow it thinks my Y is very narrow?

[image]

txt4nk, 2020-06-12:
Oh man, this is the first time I've ever seen this issue! When your bltouch probes, does it probe the entire bed? Can you use the move axis menu to move the extruder to all 4 corners?

cmer, 2020-06-12:
No, it only probes up to that position. It never comes to the front of the bed. When moving manually, it thinks Y=0 is just a couple inches from the back of the bed. it doesn't come all the way to the front.

txt4nk, 2020-06-12:
Okay let me get to the office and look through the firmware to see if there are any bugs I can find. Give me a little bit

cmer, 2020-06-12:
I tried something else. I disabled steppers and manually moved (with my hands) the Y axis all the way to the front of the bed. Now if I move Y axis through the Marlin menu, +220 just moves back a couple of inches.

Could it be a unit problem? 220mm = ~55mm on my Y axis, or pretty much 1/4 of what it should be.

txt4nk, 2020-06-12:
If your esteps for the y axis are only 1/2 of what they should be, this will happen. Thats the first thing I'll be looking at

txt4nk, 2020-06-12:
When you move the y axis with the menu, does it move really slow, or does it move at normal speeds?

cmer, 2020-06-12:
Seems slower than X, but “very slow”? See for yourself. https://www.dropbox.com/s/kqr8uk52alrr2r2/Video%202020-06-12%2C%204%2058%2035%20PM.mov?dl=0

txt4nk, 2020-06-12:
okay let me look at the firmware a bit more.. I'm not seeing anything immediately alarming.. I am comparing it to my working ender 5's firmware and it all seems normal.. Going to give it another detailed look over.

txt4nk, 2020-06-12:
Can you go to Menu > Configuration > Advanced Settings > Steps/mm and verify your Y axis steps?

txt4nk, 2020-06-12:
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4773279/firmware.zip)
Give this a try..

cmer, 2020-06-12:
X and Y are at 80. Z is 400.

cmer, 2020-06-12:
Tried your FW. Same thing unfortunately.

txt4nk, 2020-06-12:
hmm.. All of the exact same settings are working perfectly on mine (E3 Mini V1.2). I'm wondering if this is another V2.0 bug. This may be work posting about over on the marlin github. Feel free to tag me and I'll give as much input as possible as well.

cmer, 2020-06-13:
Interestingly enough @txt4nk, my Y works fine with  https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-ender5.bin -- the Z is inverted, however. I just need to figure out what the difference is between that one and mine now...

txt4nk, 2020-06-13:
Its still got to be a bug, one that BTT seemed to have ironed out but not shared. Just wish they would share their project files instead of bins only

mysticprysm, 2020-06-13:
I could try giving you my compiled Marlin. Works perfect on an ender 3. And
a 5 is just a bigger 3 right?

txt4nk, 2020-06-13:
Looks like you can download their entire project. At that point all you have to do is change the direction of the stepper. Open the config.h and search for invert. You'll find the section to invert the stepper direction

txt4nk, 2020-06-13:
> I could try giving you my compiled Marlin. Works perfect on an ender 3. And a 5 is just a bigger 3 right?
> […](#)

Its actually not that simple. The ender 5 uses x and y max switches, the ender 3 uses and and y min switches. The software end stops are different, and the bed size is different (in the software, physically it is the same size)

cmer, 2020-06-13:
I’ll try rebuilding my config from the BTT config and hopefully it’ll work. At least I know it’s likely not a bug in Marlin or a defective board.

txt4nk, 2020-06-13:
There is definitely a bug in the firmware. As I mentioned, BTT seem to have worked their way through it. The firmwares I sent you work on my ender 5 after changing the board to the V1.2. The V2.0 was just added last week so we can't expect perfection out of the gate

mysticprysm, 2020-06-14:
I don't own an under 5, but if it's not a bug within Marlin itself, it
seems like you should just be able to go grab the Marlin stuff and
configure your own. I know I said I haven't under three and that's not
quite the same, but if Marlin has support for the board now natively, then
everything should be fine. I believe I've seen one or two people in a
couple Facebook groups also saying they've got theirs working on an under 5
so it's doable.

txt4nk, 2020-06-14:
The firmwares that are showing the problem are the latest marlin 2.0.x bugfix. Not the BTT files. If you read back in the thread. The exact firmware i built for him works perfectly fine on my ender 5 (only change is the board as im on a V1.2). This tells me it is a bug with the board info within the marlin firmware.

cmer, 2020-06-14:
Ok I have somehow got it to work. I have no clue why it works now. I took the vanilla Marlin config file and rebuilt it from scratch.

This commit works: https://github.com/cmer/creality_ender5_config_and_notes/commit/83aae28058b7ae6b8cf9e455c590fd0e5217a67a

You can see the diff between my old and new config files.

txt4nk, 2020-06-14:
Try using https://www.diffchecker.com/

txt4nk, 2020-06-14:
One thing I immediately see is your bed size is not correct, I know physically it is 235mm but the tried and true software size is 220mm. Also, you need to uncomment restore abl after g28 and add comments to #define speaker.

Plasticjesus, 2020-06-24:
I am having the same issue with the Ender 5 Pro precompiled firmware being inverted. I have tried to compile my own off both Vanilla Marlin and BTT baselines. While I can get each of them to compile and flash fine each has its own set of issues. One of which is the same Issue that @cmer  was having with the printer only using the back half of the bed. Other issues with different versions include trying to home off the bed etc, Can somebody please provide me with working Example for the Ender 5 Pro with BLtouch.

txt4nk, 2020-06-24:
Have you tried using cmer's firmware he has posted on his repo? https://github.com/cmer/creality_ender5_config_and_notes/tree/master/marlin-bugfix-2.0.x

cmer, 2020-06-24:
You can find my bin and al my config on my repo. This should get you up and running in no time.

Plasticjesus, 2020-06-24:
> 
> 
> You can find my bin and al my config on my repo. This should get you up and running in no time.

Awesome, I am new to GIt up so I was manually combing through your Diffs, I see I can get the whole thing as a Zip now, I will try to give it a go. Much appreciated!@

kakos73, 2020-06-24:
i can't download your bin.....it make 67o

Plasticjesus, 2020-06-24:
> 
> 
> You can find my bin and al my config on my repo. This should get you up and running in no time.

I can confirm, with some minor tweaks, I was able to get it going with your config files. Comparing yours and the previous copy I compiled, I still cannot for the life of me figure out what caused the back half of the bed issues, everything else was user error. Anyhow I really appreciate it.

cmer, 2020-06-24:
I was never able to figure out what the half bed issue was either. I wonder if it was a bug in a specific bugfix release.

umop-apisdn-wl, 2020-06-24:
did anyone figure out the inverted Y axis issue on the ender 5? just flashed the firmware-ender-5.bin and having this issue.

cmer, 2020-06-25:
Read above ^^^

umop-apisdn-wl, 2020-06-25:
i did. i see you have one for the bl touch. Will that work if i do not have one?

[69 more replies]