bigtreetech/BIGTREETECH-SKR-mini-E3 #268: [BUG] v 2.0 mainboard wont autolevel using BLTouch

benutne, 2020-05-30:
### Description

Autoleveling bed fails after first touch point.  No errors on LCD.  

### Steps to reproduce

1. Try to autolevel from printer

**Expected behavior**
Autolevel hits 25 points and stops

**Actual behavior**
Autolevel stops after the first point

### Additional Information
This is using the DEFAULT BLTOUCH FIRMWARE.  I've had the board a few days, compiled my own firmware, flashed it.  I'd autoleveled the bed a few times, screwed up a bunch, then gave up and went to be.  Then this morning auto leveling exhibited the aforementioned behavior.  So I flashed back to the firmware from GitHub and the issue still exists.

hadukennl, 2020-05-30:
It seems like your firmware and connections may be switched.  Make sure you plug bltouch into the z-switch plug location on the edge of the board with x/y/etc. Use the z-probe-as-homing firmware.  Make sure you do a factory reset after the new firmware as well

Alan-81, 2020-05-30:
I have installed mine yesterday and done the steps you have suggested but still have the same issue.

benutne, 2020-05-30:
> It seems like your firmware and connections may be switched. Make sure you plug bltouch into the z-switch plug location on the edge of the board with x/y/etc. Use the z-probe-as-homing firmware. Make sure you do a factory reset after the new firmware as well

I haven't changed any plugs or wires since I got the thing and up till yesterday, ABL was working fine.  How to I perform a factory reset?

abacabie, 2020-05-30:
You can send a M502 to factory reset then a M500 to save changes.
I don`t think i have read of anyone getting ABL working on the V2 board yet and there is no guides up yet.

benutne, 2020-05-30:
> You can send a M502 to factory reset then a M500 to save changes.
> I don`t think i have read of anyone getting ABL working on the V2 board yet and there is no guides or videos at the moment.

Thanks.  I'll try that later. The thing I don't get is, IT WAS WORKING.  Then suddenly it stopped.  I don't think it was actually doing anything with the info it gathered because the high spot in my bed was still really squished but at the very least it was performing the full 25 point test.

abacabie, 2020-05-30:
If you have any luck it would be great if left a comment, i`m waiting of the V2 to arrive and i ordered a Trianglelab  3dtouch at the same time.

Alan-81, 2020-05-30:
I have the same set up you're getting and The auto bed leveling doesn't work at all, there isn't much help out there neither to resolve it.

hadukennl, 2020-05-30:
Use this firmware. Before you start, factory default everything and reset eeprom. https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch-for-z-homing.bin

Make sure you have the bltouch hooked up and the black/white pair is plugged into the z-axis switch when you flash the firmware

EFISCompMon, 2020-05-31:
Same issue here, just stopped working. Installed the board yesterday and flashed "firmware-bltouch-for-z-homing.bin" via SD card.
Yesterday: Auto-Home worked fine. Bed leveling went through the entire sequence.
Today: Auto-Home still works fine. Bed Leveling just stops after touching 1st point.
Tried to reflash the same firmware several times, M502 then M500 commands via octoprint, no fix

benutne, 2020-05-31:
> If you have any luck it would be great if you left a comment, i`m waiting of the V2 to arrive and i ordered a Trianglelab 3dtouch at the same time.

M502 followed by M500, unplugged the Octoprint, then shut down the printer.  Turned the printer back on and now it looks like it is going to complete all 25 touch points.

hadukennl, 2020-05-31:
I have removed OctoPrint from my setup for the time being as it's a variable I can't manage right now.

The M502 then M500 is the Marlin EEPROM reset or initialization. It's in the menus somewhere as well

benutne, 2020-05-31:
Well, I can verify that the touch point test completed but it does JACK SQUAT.  I printed a single layer X pattern and some parts of it are so thin you can see through them while other parts didn't stick to the bed at all.  I know this is a "new" board and all but I would think they could get their firmware right by now.

vexter77, 2020-05-31:
I started over, Reflashed the zhoming firmware. I reset defaults.  Im using v3.1 of the bltouch. I used this mount and visualy adjusted it to get a "close" home. Then calculated my Z off set using these instructions.  I used M420 S1 as my cura command.  Its working now. 

Is your Z-offset correct?

Getting the Z-offset right is critical to get your first layer. Use this process to find and set it:

With a firmware based on Marlin 1.1.x the procedure is a bit different as the behaviour of G92 in pre 1.1.x versions was buggy:

    Heat your printer up to your printing temperature and allow a few minutes for it to expand and settle
    Reset the existing Z-offset to zero

    M851 Z0

    Home all axes

    G28

    Move the nozzle to the middle of the bed

    G1 X110 Y110

    (if your bed is 220 x 220)
    Turn off the software endstops with

    M211 S0

    Move the nozzle down so it is just gripping a piece of standard printer paper
    Set the Z-offset to the displayed value. E.g. if the printer displays a Z-Value of -1.23 enter

    M851 Z-1.23

    Store it to the EEPROM

    M500

    Important notice! Enable the endstops again with

    M211 S1

    or the printer head will collide with the bed on the next

    G28

benutne, 2020-05-31:
> I started over, Reflashed the zhoming firmware. I reset defaults. Im using v3.1 of the bltouch. I used this mount https://www.thingiverse.com/thing:3148733 and visualy adjusted it to get a "close" [...]

I'll try all that tomorrow.  I'm not sure it is going to help though.  Some parts of the print are great.  Stuck well, look good.  In another corner of the bed, the print is so thin I can see through it.  Regardless of my Z offset, shouldn't the print be uniformly thick or thin if the bed leveling software were working?

Alan-81, 2020-05-31:
I have to say this is all a little beyond me, I have no idea what these things mean or do, so maybe its best i remove it, and level the old was with a piece of paper.

EFISCompMon, 2020-05-31:
Posting this again (edited my original post) so anyone that anyone following gets my solution.

Good old fashioned user error for me. Figured out that my bed level in the first probe point corner was out of the BLTouch trigger limits compared to the center of the bed (I was monkeying around underneath the bed, didn't realize I had moved it that much). Manually re-leveled the bed and all is good again
For anyone that is searching for this in the future here are the codes I got in the octoprint terminal when it would fail
Send: M112
Send: N2 M11235
Send: N3 M104 T0 S034
Send: N4 M140 S0*97

abacabie, 2020-05-31:
> I have to say this is all a little beyond me, I have no idea what these things mean or do, so maybe its best i remove it, and level the old was with a piece of paper.

That was me a few weeks ago with my first printer, just stick with it and you'll see that the same commands get used pretty often and anything your not sure of just google, i do that a lot when i look at something and say to myself "what the heck does that mean!".

benutne, 2020-05-31:
I'm now 90% sure that the ABL feature of the 2.0 board does not work at all.  Getting the bed to complete the test is a pain, but after running M502 to wipe all settings, correcting my Z offset, and then M500 to save.  The test will complete (at least a few times, after five or so it starts quitting before completing the test).  But the issue is that the board is doing NOTHING with that data.  The Z motor does not move at all while printing.  And my board is pretty high in the center, so I know it isn't perfectly level.

vexter77, 2020-05-31:
I can run the test and it completes. I'm printing.  But I really dont think BLtouch is actually adjusting. Its printing because I have a new bed and everything including the printers structure and the bench that it sits on is level, I checked with a level. The z axis is level and stays level through its range of motion.

benutne, 2020-05-31:
> I can run the test and it completes. I'm printing. But I really dont think BLtouch is actually adjusting. Its printing because I have a new bed and everything including the printers structure and [...]

Yup.  Just to check again, I strapped a huge 1.5 ft long zip tie to the Z axis.  No matter how minute the movements are, it should show up on the end of the zip tie.  Nada.  No movement.  ABL on the 2.0 board just doesn't work.

mshaks01, 2020-06-01:
I'm having an issue on the v2.0 board with a BLTouch 3.1 -- it will get through 20 points .. once it gets to point 21 (top left corner)  it touches and keeps moving until the z end stop kicks in -- I dont have my BLTouch connected to the z-end stop - my prints come out great but i'm not getting the full mesh as it doesn't ever complete

benutne, 2020-06-01:
> I'm having an issue on the v2.0 board with a BLTouch 3.1 -- it will get through 20 points .. once it gets to point 21 (top left corner) it touches and keeps moving until the z end stop kicks in -- I [...]

You can get a full mesh.  Take note of any settings you have stored since flashing, like Z-Offset, E-steps, temp preheats, etc.  Run a M502 using Octoprint (or whatever you use) then a M500 to write the settings.  Before you do anything else, make those changes you wrote down.  Again, however you decide to (using the printer or using a console connection) and save again (M500).  You'll be able to complete a full 25 point touch test.  But it won't matter, as the printer just does not make any adjustments to the Z level while printing.  The firmware is broken.

Teyn, 2020-06-01:
> I'm having an issue on the v2.0 board with a BLTouch 3.1 -- it will get through 20 points .. once it gets to point 21 (top left corner) it touches and keeps moving until the z end stop kicks in -- I [...]

What do you have set in your firmware? I wanted to set mine up so it's not using the z-end stop, but putting #define Z_MIN_PROBE_PIN PC14 just wouldn't work for me

mshaks01, 2020-06-02:
BTW.. as an update it started to work right after i posted this.. with no changes -- What i did do though is do a manual Bed level and then afterwards it started to go through all the points -- but like Benutne and others mentioned .. it does not seem to be adjusting when it prints.. on top of it.. i think my extruder just busted a gear.. but that's another issue.. :) i'm keeping an eye on this.. hoping there will be some updates to this.. really enjoying the silence this board gives..

abacabie, 2020-06-02:
If anyone has any success with ABL and the V2 it would be great if they could update this thread.

mshaks01, 2020-06-02:
> What do you have set in your firmware? I wanted to set mine up so it's not using the z-end stop, but putting #define Z_MIN_PROBE_PIN PC14 just wouldn't work for me

in the PINS file it's  -- #define Z_MIN_PROBE_PIN                     PC14

benutne, 2020-06-02:
I gave up for now and ordered a v1.2 board. I'll see if I can set up the v2.0 board on my older Anet A8 which I don't require to be in working order and can hack on a bit.

EFISCompMon, 2020-06-04:
> 
> 
> If anyone has any success with ABL and the V2 it would be great if they could update this thread.

ABL  works fine for me using the precompiled version (BLtouch with z-homing firmware).

benutne, 2020-06-04:
> > If anyone has any success with ABL and the V2 it would be great if they could update this thread.
> 
> ABL works fine for me using the precompiled version (BLtouch with z-homing firmware).

Can you verify that?  Is the Z stepper motor moving at all during a layer?  I had to put a big flag on mine to verify it wasn't as the Z motion when adjusting is so minute normally.

Ahmedk1995, 2020-06-06:
> precompiled

I just got my board, in the process of installing it. I'll try this out and let you know if it works or not

Ahmedk1995, 2020-06-06:
> Can you verify that? Is the Z stepper motor moving at all during a layer? I had to put a big flag on mine to verify it wasn't as the Z motion when adjusting is so minute normally.

I just got my board, in the process of installing it. I'll try this out and let you know if it works or not

Ahmedk1995, 2020-06-06:
Nope, even on the default firmware with bed leveling, the probe goes through the points but the z stepper never moves during the print.
Something is REALLY wrong here.

Teyn, 2020-06-06:
I guess the best we can do is wait for BTT/Marlin to fix this. Quite upsetting because as of right now, I would've been better with a previous version of the board.

EFISCompMon, 2020-06-07:
> Can you verify that? Is the Z stepper motor moving at all during a layer? I had to put a big flag on mine to verify it wasn't as the Z motion when adjusting is so minute normally.

I can confirm 100% that mine is working just fine. Printed through OctoPrint and direct from the SD card (just to exclude octoprint) with a "flag" attached to the z-rod.

In both cases the flag was clearly moving in response to bed level changes. I printed a single layer 200x200 empty square with 2 walls.

vexter77, 2020-06-08:
> 
> 
> > > > If anyone has any success with ABL and the V2 it would be great if they could update this thread.
> > > 
> > > 
> > > ABL works fine for me using the precompiled version (BLtouch with z-homing firmware).
> > 
> > 
> > Can you verify that? Is the Z stepper motor moving at all during a layer? I had to put a big flag on mine to verify it wasn't as the Z motion when adjusting is so minute normally.
> 
> I can confirm 100% that mine is working just fine. Printed through OctoPrint and direct from the SD card (just to exclude octoprint) with a "flag" attached to the z-rod.
> 
> In both cases the flag was clearly moving in response to bed level changes. I printed a single layer 200x200 empty square with 2 walls.
> 
> Edit: Just for a sanity check, I deliberately un-leveled my bed to exaggerate the movement and still worked fine

I would like to know what I did differently than you? I've read or watched everything google could find.

EFISCompMon, 2020-06-08:
> I would like to know what I did differently than you? I've read or watched everything google could find.

You are using the "firmware-bltouch-for-z-homing.bin" precompiled firmware?
Can you post your starting g-code as well?

Ahmedk1995, 2020-06-08:
> I would like to know what I did differently than you? I've read or watched everything google could find.

I'm also curious as to what you're doing differently, could you possibly post your starting gcode as I have tried every combination with no avail.

EFISCompMon, 2020-06-08:
> I'm also curious as to what you're doing differently, could you possibly post your starting gcode as I have tried every combination with no avail.

Here's my start code. Grabbed it from a Reddit thread when I got my BLTouch and modified it slightly.
`; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S140; start warming extruder to 140
G28 ; Home all axes
M117 Auto-Level Running
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
M117 Purging
G1 Z5.0 F3000 ; move z up little to prevent scratching of surface
G1 X0.1 Y20 Z0.3 F5000.0 ; move to start-line position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; draw 2nd line
G92 E0 ; reset extruder
G1 Z5.0 F3000 ; move z up little to prevent scratching of surface
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
M117 Printing Started
; End of custom start `GCode``

Ahmedk1995, 2020-06-09:
> Here's my start code. Grabbed it from a Reddit thread when I got my BLTouch and modified it slightly.
> ; Ender 3 Custom Start G-code
> M140 S{material_bed_temperature_layer_0} ; Set Heat Bed [...]

Yep, really not sure what your code is doing different than mine, tried yours and still absolutely nothing.
You may be the only lucky one in this thread to have a working ABL on the V2.0 board :/

Teyn, 2020-06-09:
I flashed the precomipled firmware and reset to defaulfs, then adjusted my offsets and then ran the probing (25 probes). Once that finished I started a print with a slightly off level bed and monitored my Z axis movements.
The Z axis is still not moving and the print is obviously imperfect / not sticking on one side.

This is on Ender3 Pro bought in 2020 with genuine BLTouch.

tippedhatnstogy, 2020-06-09:
I also can't get the BL Touch to work with my BTT SKR MINI E3 V2.0. The probe seems to work just fine but the ABL doesn't seem to adjust anything. I have tried both firmware versions for the BL and both wiring as Z and in the servo.

After 12 hr of trying I'm throwing in the towel.

txt4nk, 2020-06-09:
There is a bug with bilinear abl, head over to the marlin github and you'll see countless posts about bilinear abl not working.

Ahmedk1995, 2020-06-09:
Great news everyone, I have solved the mystery, ABL is now working perfectly.
The key is to NOT use the precompiled firmware (Only worked for one person here) and to NOT use the marlin repository from the BTT Github, instead head over to the latest bugfix Marlin repository on the official Marlin Github from here 
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x
Download it then download the BTT SKR E3 Mini V2.0 Configuration files from here
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/BigTreeTech/SKR%20Mini%20E3%202.0
(You only need configuration.h and configuration_adv.h)

Now comes the good part, take those configuration files and drop them in the marlin build, then open them in your text editor of choice (I use VSCode since I can compile there)

You have to change a few things,
In Configuration.h:
UNCOMMENT `#define BLTOUCH`
SET `#define NOZZLE_TO_PROBE_OFFSET { X, Y, Z}` to the proper offset for your setup
SET `#define PROBING_MARGIN` to whatever you like, mine was 30 to avoid binder clips
COMMENT `#define MESH_BED_LEVELING` and UNCOMMENT `#define AUTO_BED_LEVELING_BILINEAR`
UNCOMMENT `#define RESTORE_LEVELING_AFTER_G28`
SET `#define GRID_MAX_POINTS_X` to a number between 3-7, I used 3 for testing purposes (9 points), but you can keep it at 5 for 25 points
UNCOMMENT `#define Z_SAFE_HOMING`
SET `#define Z_MIN_PROBE_ENDSTOP_INVERTING` to true
UNCOMMENT `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

In Configuration_adv.h:
There is not much to change in here, but I would go to the BLTOUCH section and just make sure the only thing that is uncommented is `#define BLTOUCH_SET_5V_MODE`

Last thing, compile, build and flash your firmware. Finally add G29 in your start Gcode (preferably AFTER G28), and what I did to test it is I used M111 S8 to dry run a big print, and while it was probing I used a stick under the probe to offset it by a lot. That resulted in movement in the z stepper during the print.

Hope this helps everybody

abacabie, 2020-06-09:
I can also confirm ABL working correctly with the Bugfix version of marlin, using a Trianglelab 3D touch and the skr mini e3 v2, only difference is my Z stop port on the board is broken so my 3D touch is connected to the EO-STOP, when connected to Z_PROBE it wouldn't trigger.

Ahmedk1995, 2020-06-09:
> I can also confirm ABL working correctly with the Bugfix version of marlin, using a Trianglelab 3D touch and the skr mini e3 v2, only difference is my Z stop port on the board is broken so my 3D touch is connected to the EO-STOP, when connected to Z_PROBE it wouldn't trigger.

I'm glad it helped you. Interesting about the z stop port, mine works fine. Should work fine for most people but I guess if somebody has a broken z stop port they could also use your solution :)

abacabie, 2020-06-09:
i could help, i defined the Z stop pin to PC15 in pins_BTT_SKR_MINI_E3 pins file, the board came like that, i`m waiting to hear back from BTT and the seller i got the board from, unfortunately it was an Aliexpress buy, funnily enough there is a green QC stick next to the port....its hilarious.

tippedhatnstogy, 2020-06-09:
Ahmedk1995 I'd just like to double check are you using the BL touch with or with out z switch and/or is it wired as Z-probe or z end stop.

Sorry if I muddy the water but I'm way new. 
Thanks for your help ether way I'm working through your fix now, fingers crossed.

abacabie, 2020-06-09:
I have the Trianglelab 3D touch working with Z end stop, Z-probe wouldn't trigger for me, thats using the bugfix marlin.

Ahmedk1995, 2020-06-09:
> Ahmedk1995 I'd just like to double check are you using the BL touch with or with out z switch and/or is it wired as Z-probe or z end stop.
> 
> Sorry if I muddy the water but I'm way new.
> Thanks for your help ether way I'm working through your fix now, fingers crossed.

No z switch, the idea is that the white and black wires from the bl touch go into the z stop port where the original mechanical switch used to be. It's not mandatory but that's the setup I used. Make sure you enable z safe homing if you use the bl touch as the z stop

tippedhatnstogy, 2020-06-10:
So it took my all day to learn how and to do, and there may be some tweaks needed but, it is working. 
Huge Thanks to Ahmedk1995.
I can confirm that the Z is adjusting as the printer moves, and that the Firmware is working.

Ahmedk1995, 2020-06-10:
> So it took me all day to learn how to and to do it, and there may be some tweaks needed but, it is working.
> Huge Thanks to Ahmedk1995.
> I can confirm that the Z is adjusting as the printer moves, and that the Firmware is working.

Glad to hear it worked, I shouldn't be taking any credit, the thanks goes to the Marlin dev team.
And take your time with this, it's definitely a learning curve, I'm quite new to it as well.

mshaks01, 2020-06-11:
Do there need to be any changes made to the Pins files in **marlin\src\pins\stm32f1\**? I dont see the 2.0 board defined there like i do in the BTT firmware?

abacabie, 2020-06-11:
No, most of the pins are defined in the file "pins_BTT_SKR_MINI_E3.h".

JPCossu, 2020-06-12:
> Glad to hear it worked, I shouldn't be taking any credit, the thanks goes to the Marlin dev team.
> And take your time with this, it's definitely a learning curve, I'm quite new to it as well.

Hi, can you share your .bin, I still don't know how to compile firmware. That would be nice of you.

Ahmedk1995, 2020-06-12:
> Hi, can you share your .bin, I still don't know how to compile firmware. That would be nice of you.

I can, it is attached however I would warn you that I have a lot of stuff set up differently due to the fact that I installed a BMG extruder and upped my rates, what I can do for you is tell me what mods you have on your printer and I can compile a firmware for you.
Other option is learning to compile it yourself which honestly isn't hard when using VSCode + PlatformIO + Auto build marlin
If you want to learn (would make your life easier), watch this video it shows exactly how to do it
https://www.youtube.com/watch?v=W6zYvRgGr3Q

tippedhatnstogy, 2020-06-12:
Feel free to try this, I also recommend that you do your own. 
I have learned the last couple days, so **use at your own risk**, I barely know what I'm doing.

But having been in that spot, "I just want it to work, and not have to take it all apart again." I'd like to try and help.

I would recommend you make your own, but if you want push forward and at least have the printer working again, go ahead.

But please save two copies of mine and then try to make your own that way you have a feeling of a safety net as backup,

All above is assuming my file I'm trying to attach to this post works.

I second the use of the "Auto Build Marlin" tool. I was watching some out dated tutorials and had to discover Auto Marlin own my own.  

[my zip STM32F103RC_btt_512K_USB.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4769886/my.zip.STM32F103RC_btt_512K_USB.zip)

JPCossu, 2020-06-12:
> > > > So it took me all day to learn how to and to do it, and there may be some tweaks needed but, it is working.
> > > > Huge Thanks to Ahmedk1995.
> > > > I can confirm that the Z is adjusting as the printer moves, and that the Firmware is working.
> > > 
> > > 
> > > Glad to hear it worked, I shouldn't be taking any credit, the thanks goes to the Marlin dev team.
> > > And take your time with this, it's definitely a learning curve, I'm quite new to it as well.
> > 
> > 
> > Hi, can you share your .bin, I still don't know how to compile firmware. That would be nice of you.
> 
> I can, however I would warn you that I have a lot of stuff set up differently due to the fact that I installed a BMG extruder and upped my rates along with a few different mods to the firmware, what I can do for you is tell me what mods you have on your printer and I can compile a firmware for you.
> Other option is learning to compile it yourself which honestly isn't hard when using VSCode + PlatformIO + Auto build marlin
> If you want to learn (would make your life easier), watch this video it shows exactly how to do it
> https://www.youtube.com/watch?v=W6zYvRgGr3Q

Thank you very much. I'll watch the video, of course. I would however like to be able to install this V2 on my ender 3 PRO, I only have a BLTouch installed in its original position (sensor on Zendstop). For the rest I can manage, I just need a working firmware that allows me to resume my impressions quickly.

Sorry, English is not my original language.

CRS99, 2020-06-13:
Sorry Newb here, can someone tell me where am I suppose to plug the BLtouch 3 pin into? The V2.0 board has the Z-Probe pins, but there are 5 pins.

Thank you in advance!

Ramoss007, 2020-06-13:
> Feel free to try this, I also recommend that you do your own.
> I have learned the last couple days, so **use at your own risk**, I barely know what I'm doing.
>
> But having been in that spot, "I just [...]

Hi, I will be trying this this evening thanks!

I am wondering if you can share the Marlin files as I would like to copy over my older ender 3 ABL settings to the E3 (Mesh size, Probe travel speed etc). I could just about config the old Ender 3 but this is slightly daunghting for me lol.

Thanks

tippedhatnstogy, 2020-06-13:
> Hi, I will be trying this this evening thanks!
>
> I am wondering if you can share the Marlin files as I would like to copy over my older ender 3 ABL settings to the E3 (Mesh size, Probe travel speed [...]

> Great news everyone, I have solved the mystery, ABL is now working perfectly.
> The key is to NOT use the precompiled firmware (Only worked for one person here) and to NOT use the marlin repository [...]

> > The key is to NOT use the precompiled firmware (Only worked for one person here) and to NOT use the marlin repository from the BTT Github, instead head over to the latest bugfix Marlin repository on the official Marlin Github from here
> > https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x
> > Download it then download the BTT SKR E3 Mini V2.0 Configuration files from here
> > https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/BigTreeTech/SKR%20Mini%20E3%202.0
> > (You only need configuration.h and configuration_adv.h)
> 
> https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/BigTreeTech/SKR%20Mini%20E3%202.0 is a 404.
> They have now moved it to:
> https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0

Here are the posts with the Marlin Files for you to make and tweak what you need.

[58 more replies]