bigtreetech/BIGTREETECH-SKR-mini-E3 #270: Mid print freeze, SD card corrupted

Erahhon, 2020-05-30:
Can someone tell me what could cause that?

Ender 3 freezes during printing in random moment. I cannot do anything, screen does not respond. When I checked micro SD card, files were gone (like the SD card corrupted). I had to upload new firmware.bin and upload to main board. In the beginning main board worked fine but I have this problem for 2 weeks right now. First card broke down, 2nd and 3rd i have tried but files just disappear. More to that even thermal runaway does not in that case temperature of hot end and bed were rising every time issue occured.

I use firmware from github for z-homing
Cura 4.6.1
Different SD cards up to 8GB of memory
Bltouch
SKR mini E3 v1,2
Ender 3
Bltouch v3.1 from Antclabs

kadluna, 2020-06-23:
I have the same problem. :-(

Erahhon, 2020-06-23:
Till now i did not find a way to fix this. However, I found solution that skip micro-SD card. Try to use Pronterface or program like this where you can upload gcode and sent request for print from computer. I do not know what causes this, but for now this is my fix of the problem. G-code is read by printer from computer not micro-sd card.

kadluna, 2020-06-24:
Many thx for your comment, Erahhon.
I decided to switch the board to Creality 1.1.5.
I am looking forward to a print without freezing.

Zonark, 2020-06-25:
Did you ever find a solution to this? To be honest I think it’s Cura causing it I keep making it to about 70 percent on my prints and boom lockup

kadluna, 2020-06-25:
I use Cura for a long time and I never had this issues on the original Melzi boards.