bigtreetech/BIGTREETECH-SKR-mini-E3 #271: Has anyone got a functional Firmware for the SKR mini E3 V1.2 and Ender 5?

MonCalamari, 2020-05-30:
Has anyone got a functional Firmware for the SKR mini E3 V1.2 and Ender 5? I am a total noob and cant get this combo working. If someone has the marlin files and a preconfigured firmware can you please post ist?

Actually the printer is stock, but i have a BL Touch here which i will install in future.

txt4nk, 2020-05-30:
I can compile a firmware for you if you'd like

txt4nk, 2020-05-30:
Instead of waiting, I just made one for the new and old. You can pick whichever binary file to use that matches your printer specs.

[CE5_SKR ME3 V1.2.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4706660/CE5_SKR.ME3.V1.2.zip)

MonCalamari, 2020-05-31:
Thank you for your work! 

My ender has got the old screw. I have uploaded the Firmware and it works. 

There are only 2 minor "Problems": 
1)  The Auto Bed Leveling is missing (but i can do that manually)
2) The Auto-Homing is in the upper right corner, the print starts in the lower left corner, so the hotend moves across the whole bed and produce a little string of filament across the bed.

Can these things be changed?

txt4nk, 2020-05-31:
Yes they can. When you said you would be installing the bltouch in the future I thought you were requesting a firmware without it. Ill be getting to work a little later today, ill get it compiled then. What lead screw do you have?

MonCalamari, 2020-05-31:
The bltouch is not mounted at the moment. In the original firmware was a menu for "bed leveling". what it was doing is go to one corner of the bed, then you adjust the height of the bed for that corner, click next, the hotend moves to the next corner, you adjust again and so on. So not a big thing as i can do that manualy with moving the x an y axis from the menu.

Is it possible that you can post the source so i can work trough it and se what you have changed? I hope to learn from it.

Also i think this will help other ender 5 users too.

txt4nk, 2020-05-31:
Absolutely, I will post the entire project file for you.

MonCalamari, 2020-05-31:
Thanks, great to have user like you that are willing to help.

txt4nk, 2020-05-31:
I was once in your shoes, took a lot of trial and error as no one really helped me much on the firmware side of it. Im happy to help.

txt4nk, 2020-05-31:
Okay so for learning purposes, I will give you a step by step.

1. Download the latest bugfix firmware & extract to your desktop https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip
2. Download the latest configurations https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x
3. Find your printer in the configurations file (config\examples\Creality\Ender-5), copy the Configuration.h, Configuration_adv.h, _Statusscreen.h & _Bootscreen.h files to your Marlin folder (Marlin-bugfix-2.0.x\Marlin)

Below are all of the changes I made to make this firmware.

Configuration.h
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1
#define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
#define MESH_BED_LEVELING
#define RESTORE_LEVELING_AFTER_G28
#define LCD_BED_LEVELING
#define Z_SAFE_HOMING
//#define SPEAKER

platformio.ini
default_envs = STM32F103RC_btt_512K

Once those changes are made, click compile and if it succeeds, you will find firmware.bin in the folder below.
Marlin-bugfix-2.0.x\.pio\build\STM32F103RC_btt_512K

Here's the entire project file. I do have a firmware.bin created in there already, but try making your own.
[Marlin-bugfix-2.0.x_MonCalamari.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4707899/Marlin-bugfix-2.0.x_MonCalamari.zip)