bigtreetech/BIGTREETECH-SKR-mini-E3 #272: Can anyone Compile a functional firmware for SKR mini e3 1.2?

TheRealRazr, 2020-05-31:
Hey, Very new to this. I've got my ender-3 and just finished installing my board. now I've found out that the latest firmware doesn't work? 

I saw some people saying that they could compile working firmware for others and would really appreciate it if someone could do that for me.

I'm also trying to setup z-homing.

txt4nk, 2020-05-31:
Post your full setup, with all mods (x y & z offsets for abl probe if present).

HVJ17, 2020-05-31:
Yeah. People have lots of working firmware. Stock works too. Not sure what issues you are having.

Do you have more specific information and details?

qwewer0, 2020-06-01:
[Guide]( https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/ ) for the board.

jakubgrzybek, 2020-06-02:
Post your full setup and your expectations. If you're using bltouch / etc you should also post your offsets from the nozzle. I have written software for my ender 3 pro from the ground (every little tweak :D) for about a month and it's running better than Prusa with pets fang's bullseye 🤣 . I can compile my software for you. If you want to use bltouch as z axis stop - Like I did, remember to connect two wires from bltouch in the z axis stop pins and setup it in marlin. Also I have designed the filament sensor mount to fit under z axis screws (u just use your z axis switch and screw it in), but its in alpha version (it works fantastich, but I will add something in final version).

jakubgrzybek, 2020-06-02:
> Yeah. People have lots of working firmware. Stock works too. Not sure what issues you are having.
> 
> Do you have more specific information and details?

Stock (compiled on GitHub) firmware has a lot of lacks, such as simple "M48 V4" command (probing test to verify if your abl works repeatable) and much, much more. It's loud as F$$$ 😄 compared to personalized hybrid threshold (depends on your preferences). For me it's like BTT team do not know exactly what they are doing with marlin.

HVJ17, 2020-06-04:
Use mine if you like.

txt4nk, 2020-06-04:
Im going to assume he got it working since he went radio silence on us.

vbrabble, 2020-06-06:
Folks I'm in the same boat so hopefully, I can revive this thread. I'm using the exact same setup as the guide listed by qwewer0 above. I moved my folder to the c drive and still can't get it to build.

The only firmware that seems to work appears to be louder than the one I had tried from teaching tech which had homing and leveling issues.
thanks

txt4nk, 2020-06-06:
If you post your exact printer setup I'll build one for you tomorrow when I get to work.

vbrabble, 2020-06-06:
> If you post your exact printer setup I'll build one for you tomorrow when I get to work.

My board is the SKR Mini E3, BLTouch with the stock metal mount, and the BTT touch screen for the ender 3.
I appreciate the response.

Any idea why I can't get it to build? I was able to the other night, but I had issues with setting Z offsets so I went back and repeated everything - so I thought.

txt4nk, 2020-06-06:
What errors are you getting when trying to compile?

vbrabble, 2020-06-06:
There are 2 problems listed: 
Cannot find "f:\Marlin-2.0x\include". [9,17]
Cannot find "f:\Marlin-2.0x\include". [70,21]
The error I'm getting is: STM32F103RC_btt_512K      FAILED    00:00:04.823
thanks

txt4nk, 2020-06-06:
Have you tried downloading the latest bugfix?

txt4nk, 2020-06-06:
Also, its looking for the f drive not the c drive.. I'd start fresh one more time beginning with extracting the latest bug fix to your desktop and modifying everything from scratch again, no copying and pasting from the folder thats failing.

vbrabble, 2020-06-06:
I'm not sure if I used the Marlin bug fix as that I downloaded from their GITHUB site. I just dowloaded from their marlin.org site. I'll give that a quick try.

BTW the guide I used is -
https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/

PIDTEMPBED is not enabled
S Curve Acceleration is enabled and not LIN_ADVANCED
BILINEAR Bed leveling is enabled

vbrabble, 2020-06-06:
> Also, its looking for the f drive not the c drive.. I'd start fresh one more time beginning with extracting the latest bug fix to your desktop and modifying everything from scratch again, no copying and pasting from the folder thats failing.

Thanks! I'll give that a shot

vbrabble, 2020-06-06:
Another question - Under the Bugfix config examples for BTT, the only have a V2 folder and no V1.0 or V1.2. I have V1.2 board. Does that make a difference?

vbrabble, 2020-06-06:
10-4 -thx

txt4nk, 2020-06-06:
I thought I had it in a file on my phone.. its gonna have to wait until tomorrow morning when I head into the office.

HVJ17, 2020-06-06:
Look at my repro.

vbrabble, 2020-06-06:
> I thought I had it in a file on my phone.. its gonna have to wait until tomorrow morning when I head into the office.

No worries thanks!

vbrabble, 2020-06-06:
> Look at my repro.

Thanks! I'll check it out

vbrabble, 2020-06-06:
OMG! I got it to build and flash! I used a combination of HVJ17's repro, the guide qwewer0 posted, and the advice from txt4nk. There are still some issues I need to work out. When I try to go to another screen so I can set the z offset. My screen goes back to the home screen before I can do something. The ABL is taking a long time, I think it is set to 7x7. I going to reduce to either 4x4 or 3x3.

@txt4nk - I would still like to try your settings, so at your convenience please share.
Again thanks, folks! I learned a lot!
Best Wishes!
Vic B.

txt4nk, 2020-06-06:
Thats great to hear Vic! Isn't it a super satisfying feeling to have gone from needing help to figuring it out and having something workable? I'll gladly share my settings with you tomorrow. Could you do me a favor and share your BLtouch offsets?

vbrabble, 2020-06-06:
> Thats great to hear Vic! Isn't it a super satisfying feeling to have gone from needing help to figuring it out and having something workable? I'll gladly share my settings with you tomorrow. Could you do me a favor and share your BLtouch offsets?

Satisfying indeed! I spent a lot of late nights over the last week trying to get to this point. I was convinced that I would never figure out how to use VSCode but it is not nearly as bad as I thought.

At the moment I took my BLTOUCH settings from HVJ17 (#define NOZZLE_TO_PROBE_OFFSET { -44, -6, 0 }. I haven't set my z offset yet, but I will post as soon as I dial it in. 

I am using the metal mount shipped with the BLTouch on the stock hotend. I printed the wolfsfang and installed it, but my results were worse so I reverted back to stock. I'm sure it was operator error.

I also figured out my screen would go back to the home screen. It was either the LCD Timeout to status or the bootscreen timeout. I reverted back to the example settings.

Anyway thanks again and I will get that z offset posted up shortly.

txt4nk, 2020-06-06:
Last I checked, the creality bltouch mount offsets are -44, -16. Have you actually measured it?

vbrabble, 2020-06-06:
I haven't measure yet. I was just trying to get out the gate with a general setting with a successful build and flash. So now I am in the tweaking phase. Looks like the -6 may have been a typo.
The z offset I came up with was -2.8. I may have my manual levels cinched down too much.

txt4nk, 2020-06-06:
I would suggest doing a manual level, then auto home. After the auto home, move your nozzle 44mm to the left (since you have a 44mm offset) so your nozzle is in the center of the print bed. Adjust your z offset from there and you should be good at that point.

vbrabble, 2020-06-06:
Got it. My nozzle is definitely not centered after auto homing, but the BLTOUCH is. I'm about to call for the evening and hit it hard again in the morning.
I'll report out my settings upon nailing them down. Thanks again!

txt4nk, 2020-06-06:
Thats exactly how its supposed to be, the probe needs to probe the center of the bed when auto homing. 

Have a good night

HVJ17, 2020-06-06:
-2.8 is normal. Mine is -4
It is set to 7xy prob.
-6 offset is correct. I have default mount.

vbrabble, 2020-06-06:
Good morning! 
Thank you for the info.
I made a test print overnight to test functionality.
Accuracy is off about  0.13-.3 mm on a 25.4  mm test cube. I also see what I believe is a Z banding issue. Neither is related to the firmware I think. I'm going to have to remove my fan to get an accurate position of my nozzle to get precise measurements

vbrabble, 2020-06-06:
Using tape to mark on I came up with -45.06, -4.62

txt4nk, 2020-06-06:
vrabble,

Here's a quick compile for an Ender 3 w/ an SKR Mini E3 V1.2, BLTouch (-45, -4.6, 0).

[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4740990/firmware.zip)
[Marlin-bugfix-2.0.x_vbrabble.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4740991/Marlin-bugfix-2.0.x_vbrabble.zip)

vbrabble, 2020-06-06:
Thank you very much! I give it a go!

vbrabble, 2020-06-07:
> vrabble,
>
> Here's a quick compile for an Ender 3 w/ an SKR Mini E3 V1.2, BLTouch (-45, -4.6, 0).
>
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4740990/firmware.zip) [...]

txt4nk,
Thank you very much. your compile works great and my test print was sharp. I will admit that I "cheated" and installed the pre-compile directly.

I do plan to go in the full firmware and modify the number number of leveling points from 7 to either 3 or 4 though.

Again, I appreciate your assistance (and everyone else's contribution too). I learned a lot!
Best Wishes!
Vic B.

iceyinferno, 2020-06-26:
Sorry to bring up a month old thread but I am struggling to create any new firmware, where i did not have issues in the past.  I attempted to reinstall vscode and platformio on a completely different computer and am having the same issue no matter what version of marlin I attempt to build in env:STM32F103RC I receive an error : 

Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory

What's strange is that even my old working project is effected by this as well as any and all attempts to try and start from scratch.  It doesn't want to build in that environment.  I'm very lost and could use any and all help.  Thank you for your time.  I have pasted an output of the most recent terminal window of an attempt to build vbrabble's zipped marlin environment.
[builderror.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4838977/builderror.txt)

Rdavis200, 2020-07-04:
> If you post your exact printer setup I'll build one for you tomorrow when I get to work.
> 
> **edit**
> You just put your setup in with an edit. I'll compile one tomorrow.

> vrabble,
>
> Here's a quick compile for an Ender 3 w/ an SKR Mini E3 V1.2, BLTouch (-45, -4.6, 0).
>
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4740990/firmware.zip) [...]

Hello,  I am having the same issues as above.  I even tried to compile your files here and they will not work

vbrabble, 2020-07-04:
hvj17 and txt4nk helped me out a lot. I actually tweaked the build txtnk gave me slightly using the instructions from https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/

Here is my most current. Making sure the folder is in the C:Drive.

[Marlin-bugfix-2.0.x_vbrabble.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4871994/Marlin-bugfix-2.0.x_vbrabble.zip)

iceyinferno, 2020-07-04:
Ensure you're using the LATEST bugfix version from Marlin's website.  There is an bug that was introduced.  Read more here: https://github.com/MarlinFirmware/Marlin/issues/18441

Rdavis200, 2020-07-04:
Thanks for the help.

I Have an Ender 3 Pro with BLTouch , and a Micro Swiss direct drive all metal hot end, which my extrude retracts instead of extrudes with the BTT BLTouch firmware on Git Hub page.   I started clean using the bugfix.  I have yet to get anything compiled.  Attached are my errors.

[cid:image002.png@01D65213.59DD9B40]

Any thoughts would be appreciated.

Thanks,

Rob

GUYVER1066, 2020-07-16:
> vrabble,
>
> Here's a quick compile for an Ender 3 w/ an SKR Mini E3 V1.2, BLTouch (-45, -4.6, 0).
>
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4740990/firmware.zip) [...]

h itried your bin file but i cant home or bed level any ideas

txt4nk, 2020-07-16:
try the latest version the vrabble made https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4871994/Marlin-bugfix-2.0.x_vbrabble.zip

Rdavis200, 2020-07-16:
Thank you for the help,

I have been able to create and run my own firmware now.  I believe it was file location in VSCode that was not allowing me to compile it.

Thanks again,

Rob

GUYVER1066, 2020-07-16:
Link does not work  Sent from Mail for Windows 10 From: Rdavis200Sent: 16 July 2020 16:20To: bigtreetech/BIGTREETECH-SKR-mini-E3Cc: GUYVER1066; CommentSubject: Re: [bigtreetech/BIGTREETECH-SKR-mini-E3] Can anyone Compile a functional firmware for SKR mini e3 1.2? (#272) Thank you for the help, I have been able to create and run my own firmware now. I believe it was file location in VSCode that was not allowing me to compile it. Thanks again, Rob From: txt4nk <notifications@github.com> Sent: Thursday, July 16, 2020 10:13 AM To: bigtreetech/BIGTREETECH-SKR-mini-E3 <BIGTREETECH-SKR-mini-E3@noreply.github.com> Cc: Rdavis200 <robsmusic@live.com>; Comment <comment@noreply.github.com> Subject: Re: [bigtreetech/BIGTREETECH-SKR-mini-E3] Can anyone Compile a functional firmware for SKR mini e3 1.2? (#272) try the latest version the vrabble made https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4871994/Marlin-bugfix-2.0.x_vbrabble.zip<https://eur06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fbigtreetech%2FBIGTREETECH-SKR-mini-E3%2Ffiles%2F4871994%2FMarlin-bugfix-2.0.x_vbrabble.zip&data=02%7C01%7C%7C86199fcee5374f8f198508d829924825%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637305055530926380&sdata=jgoQFzrgv6rwYg3297UBQaNW31LE9i%2F9rashikjy%2F9Q%3D&reserved=0>