bigtreetech/BIGTREETECH-SKR-mini-E3 #273: [FR] Document the jumpers that come with the SRS-mini-E3 2.0 Board

roman-ku, 2020-05-31:
**Is your feature request related to a problem? Please describe.**
I would like all the jumpers to be documented in the 2.0 manual. Including X-Diag, Y-Diag, Z-Diag, E0-Diag, etc.

**Describe the solution you'd like**
No documentation for jumpers and no schematic to check so not sure the point of the jumpers

**Describe alternatives you've considered**
Tried searching the internet for an explanation but since board is new, there is no explanation that exists

**Additional context**
Add any other context or screenshots about the feature request here.

Grrby, 2020-06-01:
All jumpers are clearly described in the board manual. In fact no so many of them 4xDIAG (Endstop) and x1 for +5V power/USB.
For schematic we have to wait a  while probably.
[https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0%20Instruction%20Manual.pdf](url)

roman-ku, 2020-06-02:
>All jumpers are clearly described in the board manual

Unless you have a different manual than I do, they are not clearly described. Not to mention the manual is written in broken English. But based on your comment I was able to figure it out I think.

I recommend for the authors to add a table list this:

| Jumper Name                 | Purpose                                                                                                                                                           |
|-----------------------------|-------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| X-DIAG                      | If jumper set, sensorless homing will be used for X-AXIS (Note: cannot use external end stop if this jumper is set)                                               |
| Y-DIAG                      | If jumper set, sensorless homing will be used for Y-AXIS (Note: cannot use external end stop if this jumper is set)                                               |
| Z-DIAG                      | If jumper set, sensorless homing will be used for Z-AXIS (Note: cannot use external end stop if this jumper is set)                                               |
| E0-DIAG                     | ???, not sure what E0-STOP is for, the original control board doesn't have this                                                                                   |
| Power selection for BLtouch | If jumper set between pins +5V and PWR: The onboard +5V power supply will be used for BLtouch; If jumper set between pins PWR and VOUT need to use external BIGTREETECH DCDC5V V1.0 Module |

Also I think your manual link is broken.

Grrby, 2020-06-02:
[](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0%20Instruction%20Manual.pdf)

Plasticjesus, 2020-06-26:
My 2 cents here. The documentation on the V2 board is horrible, the released firmware does not even work on the Ender 5 pro. The initial setup is a very poor customer experience.

aztrip, 2020-07-03:
Received my new v2.0 board w/ TFT display, but no manual included. There should be a printable manual available online.

nyjeeper, 2020-08-17:
Thanks for these comments. The last time I used jumpers was on a 486 PC. Its been a while. I am upgrading a CR-10 and bought the SKR Mini E3 v2 as a drop in replacement. 4 jumpers in the bag, and a 5th on the board, but its not 100% clear to me what to do with them. If I want to continue using the OEM mechanical limit switches, then I leave the 4 jumpers off, correct? I don't yet have a BLtouch, so there's no jumper required for it.

fraserw, 2020-09-22:
@Plasticjesus how did you get your board working with the ender 5 pro? I have failed in my attempts. I uploaded the ender-5 (non BLT touch) firmware, but XYZ directions were reversed when auto homing from the LCD panel. Swapped the pins on those steppers, but the directions are reversed when printing from octoprint (even after selecting invert axis for all three). Thus far I have failed to print anything.

What did you do to get this to work with the E5 pro?

agenttatdaddy, 2021-02-27:
Did anyone solve this issue? I have an Ender 5 Plus, and I also have the issue of reversed homing.

woodwaker, 2021-05-17:
Thanks for the post.  I was trying to figure out what the 5Volt jumper did.  I was hoping it would block the Raspberry PI from powering the board.  I'm going to have to build a USB power blocker.
TH3D has firmware that you can compile for the board and modify for the Ender 5 Plus.  I'm trying to switch to Klipper

jemmyhuang999, 2021-05-26:
What's numbers for x diag pin & y diag pin?

woodwaker, 2021-05-26:
jemmyhuang999, I'm not sure what you are asking.  In Klipper the x pin and y pins are the following in the printer.cfg file
[BIGTREETECH SKR mini E3 V 2.cfg.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/6548951/BIGTREETECH.SKR.mini.E3.V.2.cfg.txt)