bigtreetech/BIGTREETECH-SKR-mini-E3 #274: [BUG] Build fails in Adafruit_I2CDevice::read

scudderfish, 2020-05-31:
### Description

### Steps to reproduce

1. git clone https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3.git
2. cd BIGTREETECH-SKR-mini-E3/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2
3. pio run

**Expected behavior**
Build completes and firmware.bin generated

**Actual behavior**
Build fails with 
```
.pio/libdeps/STM32F103RC_btt_512K/Adafruit BusIO_ID6214/Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':
.pio/libdeps/STM32F103RC_btt_512K/Adafruit BusIO_ID6214/Adafruit_I2CDevice.cpp:160:79: error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'
   size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
                                                                               ^
In file included from /home/dgs/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/Wire/Wire.h:42:0,
                 from .pio/libdeps/STM32F103RC_btt_512K/Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1,
                 from .pio/libdeps/STM32F103RC_btt_512K/Adafruit BusIO_ID6214/Adafruit_I2CDevice.cpp:1:
/home/dgs/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/Wire/utility/WireBase.h:101:11: note: candidate: uint8 WireBase::requestFrom(uint8, int)
     uint8 requestFrom(uint8, int);
           ^~~~~~~~~~~
/home/dgs/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/Wire/utility/WireBase.h:101:11: note:   candidate expects 2 arguments, 3 provided
/home/dgs/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/Wire/utility/WireBase.h:106:11: note: candidate: uint8 WireBase::requestFrom(int, int)
     uint8 requestFrom(int, int);
           ^~~~~~~~~~~
/home/dgs/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/Wire/utility/WireBase.h:106:11: note:   candidate expects 2 arguments, 3 provided
Archiving .pio/build/STM32F103RC_btt_512K/lib38f/libSailfishLCD.a
.pio/libdeps/STM32F103RC_btt_512K/Adafruit BusIO_ID6214/Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':
.pio/libdeps/STM32F103RC_btt_512K/Adafruit BusIO_ID6214/Adafruit_SPIDevice.cpp:99:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'
     _spi->transfer(buffer, len);
                               ^
In file included from .pio/libdeps/STM32F103RC_btt_512K/Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:0,
                 from .pio/libdeps/STM32F103RC_btt_512K/Adafruit BusIO_ID6214/Adafruit_SPIDevice.cpp:1:
Marlin/src/HAL/STM32F1/SPI.h:268:11: note: candidate: uint8_t SPIClass::transfer(uint8_t) const
   uint8_t transfer(uint8_t data) const;
           ^~~~~~~~
Marlin/src/HAL/STM32F1/SPI.h:268:11: note:   candidate expects 1 argument, 2 provided
```

Zonark, 2020-06-05:
I also am having that error

txt4nk, 2020-06-05:
vanilla marlin bugfix works fine. you can use it to create your firmware or you need to modify the adafruit folder (search adafruit in the marlin github.. there are several threads about it).

scudderfish, 2020-06-06:
This is fixed by commit b5f912681e5a0777a49c467a0250abd459b16b04

closed 2020-06-06 by scudderfish