bigtreetech/BIGTREETECH-SKR-mini-E3 #275: [BUG] Unable to compile V1.2

Gnomish8, 2020-05-31:
### Description

/include dir missing causing compiling to fail

### Steps to reproduce

1. Download from github
2. navigate to /BIGTREETECH-SKR-mini-E3-master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2
3. See there is no "include" folder here
4. Open the project
5. Navigate to c_cpp_properties.json
6. Notice error on lines 9 and 69 related to this folder not existing
7. Attempt to compile
8. Get errors because this folder is missing

**Expected behavior**
Github repository has all necessary folders to compile

**Actual behavior**
Folder is missing causing compiling to fail.

txt4nk, 2020-05-31:
Why don't you take a crack at using the latest bugfix? I never use the BTT firmwares, the latest bugfix is always your best bet if you are cable to compile it.

Gnomish8, 2020-05-31:
@txt4nk -- ended up using the latest bugfix and compiled without issue.

txt4nk, 2020-06-01:
> @txt4nk -- ended up using the latest bugfix and compiled without issue.

Awesome, glad to hear it.

closed 2020-06-01 by Gnomish8