bigtreetech/BIGTREETECH-SKR-mini-E3 #277: SKR mini E3 V2.0 fan always on... again!

guildow, 2020-06-02:
Sorry for my english. I speak french.

I've buy new V 2.0 SKR Mini E3 to replace the V1.2. But the FAN0 connector is always on, and the FAN1 is always off. I return the board and now I trie another one but with same issue. After about 1h running, FAN0 are shutting down and never work again (I think the mosfet was burn).

I try change FAN_MIN_PWM and FAN_MAX_PWM in Configuration_adv.h but no change. I also noticed the D10 green light was decrease when I plug the fan (I try with 24v or 12v fans).

Would it be better to return it (again!) and re-order V1.2?

Please help

houseofbugs, 2020-06-02:
You shorted out the FAN fet. Check your wiring and the fan itself.

guildow, 2020-06-02:
> You shorted out the FAN fet. Check your wiring and the fan itself.

It's almost impossible to invert the fan. This board fit ride on Ender 3. And the fan work perfectly with 1.2 version of the board. But I check anyway and the connector are the right side.

galardao, 2020-06-02:
@guildow 

See the video of the link, if your measurements are approximately equal to the video, it is because your card has the fan control defect mosfet.

https://youtu.be/s7zhgiU1xNs

guildow, 2020-06-02:
> @guildow
> 
> See the video of the link, if your measurements are approximately equal to the video, it is because your card has the fan control defect mosfet.
> 
> https://youtu.be/s7zhgiU1xNs

Thanks, but this procedure is for 1.2 release of the board. I've the release 2.0. I'm not sure it's the same and without schematic, it's heavy to know.

houseofbugs, 2020-06-02:
If a short occours on a fet they usually fail off or on. It can happen that they can fail but its not too common.

AndyJanes, 2020-06-04:
I'm having the same problem in that I swapped out my Mini E3 v1.2 for a v2 and all fans are on 100% all of the time.  All fans were put back to stock parts before the swap (was using Noctua and buck converters on the v1.2) and apart from the noise being pretty terrible I can't control either fan with M106, M107 or via the control panel.  Tried stock firmware from here and also custom firmware from updated marlin that I was previously running on my 1.2.

How would I test for a failed mosfet if that is the likely suspect?  A bit annoying considering it arrived 2 days ago and has only been in the printer for about an hour at most!

Thanks

guildow, 2020-06-05:
I've try 2 release 2.0 board, and both have the same issue with fan. I returned both to Amazon and re-order release 1.2. I install it this evening and all of the printer work perfectly. Plug&play!

I think BTT have real desing error on 2.0 SKR mini E3. 

Sorry again for my english

siddolo, 2020-06-10:
same issue here. Mini E3 v2.0

closed 2020-06-17 by guildow

funkspoc, 2020-06-28:
The SKR Mini e3 V2.0 has two FAN Connectors. FAN0 is always on while FAN1 is controllable.

 
[image]

The following Changes were made to allow the Parts Cooling Fan to be controlled by TFT35 LCD/G-Code/Octoprint.
File Location: \Marlin-bugfix-2.0.x - FANFIX\Marlin\Configuration_adv.h
Line: 374
Task: Disable/Comment out the Use Controller FAN

//#define USE_CONTROLLER_FAN

File Location: \Marlin\src\pins\stm32f1\pins_BTT_SKR_MINI_E3_common.h
Line:97
Task: Change FAN_PIN from PC6 to PC7
------------------------------------------------------------------------
#ifdef SKR_MINI_E3_V2
  #define FAN_PIN                           PC7
#else
  #define FAN_PIN                           PA8   // "FAN0"
#endif
--------------------------------------------------------------------------
File Location: \Marlin\src\pins\stm32f1\pins_BTT_SKR_MINI_E3_V2_0.h
Line:44
Task: Define FAN1_PIN as PC6 instead of PC7
------------------------------------------------------------------------
#define FAN1_PIN                           PC6
------------------------------------------------------------------------

Now with these changes. You will be able to start the fan by sending M106 G-Code and Stop it by Using M107 G-Code. In addition, you will be able to control it from the TFT35 Screen by going to the Fan Icon

morgalion, 2020-07-02:
Hello,
I've the skr mini e3 V2.0 with recent custom marlin-bugfix-2.0 flashed.
I have same hardware install : yellow/blue cables on FAN1 as showed in the official cabling youtube video : https://www.youtube.com/watch?v=krz-plzJUH8
Note : fans locations are inverted compared to stock ender 3 motherboard (v1.1.3), confusing layout for newbies i admit...

I can control both part cooling and motherboard FANs with LCD without problem. Motherboard fan is only active if stepper motors are active (by default, idle time is 60second, so its stopping correctly after 60sec without movements, fan is stopped at startup too).

The only modification i made is uncommenting : **#define FAN_SOFT_PWM** in configuration.h because I found the control and sound of fans better.

So, sorry i think it should work out of the box, at least with last marlin bugfix (i didnt try stock bigtreetech firmware)...

aurror, 2020-07-04:
Same here, FAN0 is always on, but FAN1 is working/controllable. SKR mini E3 V2.0

ashtonias, 2020-08-09:
I have an ender 3 pro as well. After some testing they were inverted. So fan0 was always on and fan1 was controllable. I swapped them... Problem solved.

piterpw, 2020-08-20:
I just changed the settings in the marlin as described by funspoc and that's it. Changing plugs in places, in my humble opinion, is not a good idea.

bmfkai, 2020-09-01:
I just flipped the fans around.  Part fan on 0, board fan on 1 and everything works as it should and controllable on a TFT35

Rackir, 2020-09-01:
I mounted the SKR mini E3 V2 on my Cr10-S. The cooling parts fan was perfectly controllable for about 1 month. Yesterday, when I plugged on the printer, the fan started at 100% speed and i cannot switch it off with TFT or G-Code. I didn't make any change on the firmware or the wiring, so I believe it must be a burned mosfet.

Has anyone had the same experience?

ptmoy2, 2020-09-11:
I am really confused.  Funkspoc's diagram shows motherboard fan on fan0, and parts fan on fan1.  But Teaching Tech and Ruiraptor's installation videos both say to plug parts fan on fan0, and motherboard fan on fan1.  I installed per Teaching Tech and Ruiraptor's instructions.

As soon as I turned power on, the parts fan (i.e., the one plugged into fan0) runs at full speed.  So it seems like fan0 is always on as mentioned by others on this thread.  So what the correct way of doing this?

twinsenfan, 2020-09-11:
I believe I had the same issue as you guys, got a E3 mini 2.0 with my hypercube. Here is how I got is resolved.

1. Keep all pin files as default, BTT_SKR_MINI.E3.h, BTT_SKR_MINI.E3_V2_0.h
2. Connect your extruder cooling fan to FAN1(PC7) on the motherboard
3. Connect your part cooling fan to FAN0(PC6) on the motherboard

Edit marlin 2.0 Configuration_adv.h, change
#define E0_AUTO_FAN_PIN -1
to
#define E0_AUTO_FAN_PIN PC7

With the default fan temperature as 50, above setting will only turn the extruder fan on when it is over 50 degree.

You can control the parts cooling fan with M106/M107

Let me know this does not work for you.

CaBeTuX, 2020-10-02:
In Configuration_adv.h you can see by default (last Marlin 2.0 source of this BTT repo):

`/**
 * Controller Fan
 * To cool down the stepper drivers and MOSFETs.
 *
 * The fan turns on automatically whenever any driver is enabled and turns
 * off (or reduces to idle speed) shortly after drivers are turned off.
 */
//#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  //#define CONTROLLER_FAN_PIN -1        // Set a custom pin for the controller fan
  //#define CONTROLLER_FAN_USE_Z_ONLY    // With this option only the Z axis is considered
  //#define CONTROLLER_FAN_IGNORE_Z      // Ignore Z stepper. Useful when stepper timeout is disabled.
  #define CONTROLLERFAN_SPEED_MIN      0 // (0-255) Minimum speed. (If set below this value the fan is turned off.)
  #define CONTROLLERFAN_SPEED_ACTIVE 255 // (0-255) Active speed, used when any motor is enabled
  #define CONTROLLERFAN_SPEED_IDLE     0 // (0-255) Idle speed, used when motors are disabled
  #define CONTROLLERFAN_IDLE_TIME     60 // (seconds) Extra time to keep the fan running after disabling motors
  //#define CONTROLLER_FAN_EDITABLE      // Enable M710 configurable settings
  #if ENABLED(CONTROLLER_FAN_EDITABLE)
    #define CONTROLLER_FAN_MENU          // Enable the Controller Fan submenu
  #endif
#endif`

so... uncomment the line (remove the //):

`#define USE_CONTROLLER_FAN`

and ENJOY! :)

CaBeTuX, 2020-10-02:
```
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN PC7        // Set a custom pin for the controller fan
  //#define CONTROLLER_FAN_USE_Z_ONLY    // With this option only the Z axis is considered
  //#define CONTROLLER_FAN_IGNORE_Z      // Ignore Z stepper. Useful when stepper timeout is disabled.
  #define CONTROLLERFAN_SPEED_MIN      0 // (0-255) Minimum speed. (If set below this value the fan is turned off.)
  #define CONTROLLERFAN_SPEED_ACTIVE 255 // (0-255) Active speed, used when any motor is enabled
  #define CONTROLLERFAN_SPEED_IDLE     0 // (0-255) Idle speed, used when motors are disabled
  #define CONTROLLERFAN_IDLE_TIME     60 // (seconds) Extra time to keep the fan running after disabling motors
  //#define CONTROLLER_FAN_EDITABLE      // Enable M710 configurable settings
  #if ENABLED(CONTROLLER_FAN_EDITABLE)
    #define CONTROLLER_FAN_MENU          // Enable the Controller Fan submenu
  #endif
#endif
```

bleakreaper, 2020-11-07:
so i did the same think like @funkspoc say but didn't work, i was worried that mosfet is burn but i tried the kay3ds cheetah firmware and fan0 and fan1 work! :) when i increase or decrease speed of the fan the led now is blinking instead of make less or more bright.

bleakreaper, 2020-11-07:
ok so in my case i remove // from #define FAN_SOFT_PWM in configuration.h and fan0 is now controllable.

damy1976, 2021-01-09:
Sorry, i wrote basic english. me too have problem. i already new update firmware.. but always ON fan 0 and 1 both too.. cannot stop.. what abot it? i send show my video youtube:  https://youtu.be/dDTbQYqRqps

thank you and best regards

bltDefender, 2021-04-05:
had same problem with freshly configured and flashed Marlin 2. Parts cooling connected to F0 worked without any problem, extruder's cooling connected to F1 did not work ootb. Had to change #define E0_AUTO_FAN_PIN PC7 in Configuration_adv.h as suggested earlier. F1 runs automatically at 50dc (in my case) and F0 is controllable via gcode.

mantusma, 2021-07-18:
I found a strange behavior of the firmware settings on my SKR E3 Mini v. 2.0
I have a hotend Fan and a part cooling fan. And I want to use #define FAST_PWM_FAN in configuration.h

Approach1:
If I define the hotend Fan fan as CONTROLLER FAN with:

Configuration_adv.h
#define USE_CONTROLLER_FAN
   #define CONTROLLERFAN_SPEED_MIN      255

and 

#define E0_AUTO_FAN_PIN -1
#define COOLER_FAN_PIN -1

Than the hotend Fan is always on and the (Fast) PWM control of the part cooling fan works perfect.

Approach2:
But If I uncomment the CONTROLLER FAN and define the correct pins for the Fans than the PWM control doesn´t work with FAST_PWM_FAN.
Configuration_adv.h
//#define USE_CONTROLLER_FAN
and 
#define E0_AUTO_FAN_PIN PC7  //changed from -1 to PC7
#define COOLER_FAN_PIN PC6   //changed from -1 to PC6

Than the Temperature control of the hotend Fan works but PWM control of the COOLER_FAN is broken (100=on, 1 to 99%=off).

ErickONunes, 2022-07-21:
Hello, I had a problem when connecting 2 coolers, I let one wire touch the other and there was a short circuit, now the coolers are only connected directly, would there be a way to solve this problem?
A solution I thought would be to change the location of the handboard cooler to this other one that has shorted and is connected directly, would it be possible?

Thanks.