bigtreetech/BIGTREETECH-SKR-mini-E3 #278: file for the ender extender 400mmx400mmx250mm

vossproducts, 2020-06-04:
I am new to this process, but i loaded in the compiled OS for my SKR E3 mini 1.2 using the BL touch for Z homing .Bin file and it worked perfectly. I would like to add the same setup to my ender 3 with the ender extender kit (400mmx400mmx250mm print bed) but cannot find out how to load the original bin file into VS Code to change the print bed dimensions. is this possible?

txt4nk, 2020-06-05:
No it's not, you'll need to download Marlin Bugfix and compile your own binary file using Atom and Platform IO. There's a ton of videos on it on youtube.

HVJ17, 2020-06-07:
You cannot edit the .bin. You need to build a new one. clone the BTT 2.0 git. change the dimensions.

vossproducts, 2020-06-08:
So I can clone the existing BBT bltouch for z homing file and make changes to it? or do i have to start from scratch and download marlin 2.0 and do a completely new build.