bigtreetech/BIGTREETECH-SKR-mini-E3 #280: skr mini e3 v2.0  not triggering with Z axis

abacabie, 2020-06-06:
Not sure if this is a bug or hardware, i fitted a new mini e3 v2 to my  ender 3 pro today and all axis home fine apart from Z that doesn't trigger, i sent a M119 and Z is open while X and Y are triggered, i checked the end stop with a multi meter and it works as it should, i also tried swapping with the Y endstop and manually pressing the end stop while sending m119 and it still came back as open, i also tried flashing the new firmware from the BTT repository , this is a stock ender 3 pro with no bltouch.
there is not much info about this new board and wondered if there was something in the firmware that could cause this.

txt4nk, 2020-06-06:
Have  you tried compiling your own firmware using the latest bugfix version of Marlin?

abacabie, 2020-06-06:
thanks for the reply, no, i haven't tried that yet, i didn't even know i could, something to do with the boards for the V2 not been included with marlin yet, i will look into it, i also measured the voltage from the end-stops on the board and X and Y are 3.3v but Z reads 0v so it doesn't look good, i got the V2 from aliexpress as well so support is a no go.

txt4nk, 2020-06-07:
Yeah its not on the "official" release yet, but it is now available in the bugfix version.

And I've seen other aliexpress sourced boards have issues as well, I made the mistake of ordering on from aliexpress and it arrived with a tmc connection error from the get go. I have a theory that they buy the RMA's boards from retailers or BTT.. Where are you located?

abacabie, 2020-06-07:
In the UK, i think if i order again it will just be amazon, at the moment i`m trying to see if i can get a 3d touch working from the Z probe, it works as expected from the Z end-stop but just doesn't trigger.

txt4nk, 2020-06-07:
Are you positive you've got the 3 wire dupont connector wired correctly? This is common when switching to these boards because you've got to swap 2 of the wires around in the connector. If you can, post a picture of your wiring

abacabie, 2020-06-07:
Thanks for the reply, i flashed the bugfix version of marlin with stock configs and no bl touch but the Z endstop still doesn't  trigger, i checked the voltage on the endstop pins again and X and Y read 3.3v but Z reads 0v, that kind of tells me its a hardware issue, i may come back to it when more info gets released and try getting ABL working with the 5 wires in the Z probe pins as it looks like the Z endstop pins are dead, the wiring i used for the 3d touch was the same as whats in the manual for the V2, the 3d touch is from Trianglelab, i have messaged the seller but i`m expecting the usual "send it back" response", i have swapped back to the creality board while i figure out if i should get another BTT board or try something else.

txt4nk, 2020-06-07:
Sounds like a solid plan. Still though, check your wiring when you install the next BTT board. Going from the creality board to a BTT board, you've got to swap pins in the 3 pin dupont. Blue and yellow must be on the outer pins and red in the middle.

MimeoNoe, 2020-06-07:
Same problem here, and I switched the pin to match the BTT board configuration. It only seems to occur using "firmware-bltouch-for-z-homing.bin". With "firmware-bltouch.bin" it triggers

hadukennl, 2020-06-07:
FWIW - I don't know if this is correct. It seemed to start working after I had it wired correctly, THEN flashed the new firmware. Again, I don't know if that is relevant or circumstantial.

txt4nk, 2020-06-07:
There are a ton of variables, anyone in this thread would greatly benefit from learning how to compile your own firmware. 3D make it on you tube has awesome tutorials.

abacabie, 2020-06-08:
I think that`s good advice, and something i was getting familiar with before my first printer (ender 3 pro) arrived 6 weeks ago, it really is a must.

smanton, 2021-01-21:
Did you ever resolve this issue? I have just installed a skr mini e3 v2.0 and my Y stop isn't triggering. I have sent a M119 code ot the printer and when the Z switch is closed Z and Y are triggered

taylorhughmorgan, 2021-01-25:
I had this same issue, except it is with **all** of my end-stops. I sent M119 and it showed all the end-stops were open, even if I was holding them down. I fixed this by removing all the red jumpers from x-diag, y-diag, and z-diag.

closed 2021-07-28 by abacabie