bigtreetech/BIGTREETECH-SKR-mini-E3 #281: [FR] Source configuration files for precompiled firmware.

siddolo, 2020-06-07:
Can you provide the configuration files used to built a precompiled firmware?

kakos73, 2020-06-11:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/266

look the last post

txt4nk, 2020-06-12:
@kakos73 I'm pretty sure @siddolo is referring to the configuration files that BTT used to create their firmware binary files. What I posted in #266 is the latest and greatest vanilla marlin 2.0.x bugfix that I had to go modify for the Ender 5 and the specific E3 mini board. If you know the correct items to change, its probably best to use the latest bugfix.

izeberq, 2020-07-20:
Hello,
Is it possible to share configuration for all boards/firmware configuration?
After compiling FW for SKR Mini 2.0 it attempts to use BLTouch for Z-Homing and ignores appropriate configuration.

jtrmal, 2020-07-20:
I agree, it would be even great documentation of things, if the configs were provided...