bigtreetech/BIGTREETECH-SKR-mini-E3 #282: [BUG] FAN0 and FAN1 ports not the same as on Creality Melzi board

patoberli, 2020-06-10:
### Description

The FAN0 and FAN1 ports on the SKR Mini E3 V1.2 (and probably also the other revisions) are swapped, compared to the Creality Melzi 1.1.4 board found on the Ender 3 and 5. 
In other words, the old Melzi FAN0 cable needs to be plugged into the FAN1 port on the E3 and vice versa. 
This is, as far as I could see, nowhere mentioned in the documentation (of the V1.2).

If this is not done, then the part cooling fan is always running at 100% on the hot-end, while the mainboard fan is only running when it's printing something.

### Steps to reproduce

1. Plug old FAN0 cable into new FAN0 port on the E3
2. Plug old FAN1 cable into new FAN1 port on the E3

**Expected behavior**
The documentation should correctly state that those two cables need to switch the position.

**Actual behavior**
Part cooling fan on the hot-end always running at 100%, mainboard fan only running when printing.

### Additional Information
Zip file with my configuration attached.
[configurations_patoberli.tar.gz](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4757291/configurations_patoberli.tar.gz)

abacabie, 2020-06-10:
I agree that it should be documented, if not known before hand i can see this leading to a lot of frustration for some users, as for as i can tell ( with my limited knowledge) its just a case of defining pin numbers with the V2 so not such why BTT do that on a board that is supposed to be a direct swap for the ender 3.

patoberli, 2020-06-11:
> This is well known. It is documented.

And where? At least not in the manual: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2manual.pdf

DarkSnart, 2021-02-13:
> This is well known. It is documented. 

It only becomes well known after one experiences the issue and performs a web-search. 

And what is the solution? FAN0 should be controllable as it is designated as Part Cooling. FAN1 is designated as the Controller fan and is only on when printing because that's when the Drivers need to be cooled. The Controller fan then turns off 60 seconds after printing has complete. 

In my case, FAN0 is constantly on, yet my display shows 0%. If I then raise the theoretical fan speed to 100%, the fan actually slows down. Output is 12.1v when showing 0% speed, and drops to 7.3v when showing 100%. Is this a MOSFET issue? My understanding is that if a MOSFET failure occurs it would be either on or off.

Thanks for any help in resolution!