bigtreetech/BIGTREETECH-SKR-mini-E3 #283: Firmware SKR mini E3 V2.0 with bl touch

JPCossu, 2020-06-12:
Hello everyone, is this released firmware up and running? 
I just received the V2 and I don't want any more problems like with the v1.2.

Thanks and take care

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch-for-z-homing.bin

vexter77, 2020-06-13:
That does not work.  However, my v2 is using BLtouch thanks to this thread.  https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/268

KL-XL, 2020-06-22:
It works like charm for me on my BTT SKR Mini E3 V2.0 on ender 3 :) You just need to use the pins correctly: the 3-pin Dupont connector goes to Z-Probe area and the 2-pin JST-XH connector goes to Z-Stop

Raymans2, 2020-06-24:
@KL-XL I'm planning to the same, however I'm unsure what direction the Dupont connector will be placed. Also the color/function order would be helpful since the SKR 1.2 requires you to switch the position of Red and Blue. So currently my pins on my BLTouch go Gnd(blue), VCC(red),Sig(yellow). I am also using the BTT SKR Mini E3 V2.0.

Thanks in advance

KLXL2019, 2020-06-24:
BTT SKR Mini E3 V2.0 doesn’t need pins switching. Just look at the pins map and plug your DuPont connector as it is into the Z-Probe slot accordingly. The cable colors are really unreliable.

Raymans2, 2020-06-25:
But there are five pins on the Zprobe header.

From: KLXL2019
Sent: Wednesday, June 24, 2020 7:57 PM
To: bigtreetech/BIGTREETECH-SKR-mini-E3
Cc: Raymans2; Comment
Subject: Re: [bigtreetech/BIGTREETECH-SKR-mini-E3] Firmware SKR mini E3 V2.0with bl touch (#283)

BTT SKR Mini E3 V2.0 doesn’t need pins switching. Just look at the pins map and plug your DuPont connector as it is into the Z-Probe slot accordingly. The cable colors are really unreliable.

KLXL2019, 2020-06-25:
Yes, but you only need to use 3 of them. If you look at the board from the top and z-stop is at the bottom and z-probe on the right, then you need to use bottom 3 pins of the z-probe

bobcroft, 2020-07-01:
The z probe connector pins viewed from the bottom edge of the board, where the thermocouple connectors are, Ground1, 5V, signal, on my BLtouch the wire colours are Brown (Gnd), Red (5V), Yellow(signal). These are the servo connections. The remaining two pins are Ground2 (4), Z probe in (5).  On my BLtouch the wire colours are black (Gnd) and white (Zprobe).  I do not know why people say the two pin connector goes to Z stop  rather than to pins 4 and 5 of the Z probe connector.  However,     I connect my z probe wire to pins 4 and 5 and it doesn't work.  So, I'll try connecting them to the Z stop pins.  Furthermore, the Bltouch extension cable has a 5 pin JST-XH connector which matches the Z probe connector.
If anyone can clear up this apparent anomaly I would be very grateful.

TinoFpv, 2020-07-04:
Nothing unusual here, There are several versions o f Bl touch, and 2 pins layout. I got the 3.1 and with the skr mini e3 v2 I had to switch the the first 2 probe cables.
You can choose from config.h if bl touch should work as z-switch or not. Anyway i have the firmware for ender 3 pro with BlT they made and it is working flawlessly.

bobcroft, 2020-07-05:
I do not think anyone suggested there was anything unusual, there is, however, plenty of opportunity for confusion.  In addition there are many people who are uncomfortable modifying files, particularly in something like platform.IO which is a complicated piece of software.  Based om my learning I will share the easiest way to achieve success.  My notes are based on the SKR mini E3 V2 installed on an Ender 3 Pro.  I used a genuine BLtouch V3.1.  I obtained the necessary bin file to flash to the mini board from the BTT github.  The github download also includes a very good manual showing the board connections in detail.  The BL touch V3.1 has the following wires for the servo, Brown - Gnd, Red - 5 volt, Yellow - signal, these are terminated in a 3 pin Dupont connector, for the probe it has the following wires, Black - Gnd and White - probe signal, these are terminated in a 2 way Dupont connector.
An extension cable will be required for the BLtouch depending on the intended use of the BL touch, see below.
On the mini V2 there is a dedicated Z probe connector which is NOT a Dupont connector (2.54 mm pitch), it is a 5 pin JST-XH connector (2.5 mm pitch).  I refer to the bottom of this connector being the end nearest the thermocouple sockets.
IF the BLtouch is to be used for bed levelling only and not as a Z stop then all five wires from the BLtouch are connected to the Z probe socket in the order listed above.   The correct firmware is the one for the v2 board with BLtouch.
IF the Bltouch is to be used for bed levelling AND as the Z stop the servo connections are made to the bottom 3 pins of the Z probe socket.  The black and white wires are connected to the two pin Z stop connector after the Z stop microswitch is removed.  The correct firmware is the one for the V2 board with BLtouch and Z stop.
The selected firmware is loaded onto a SD card and its name MUST be changed to firmware.bin.  The SD card is then inserted into the SKR mini SD card slot whilst the printer is powered off.  When the card is in the Sd card slot, the printer is powered up and the new firmware will load.  After a few seconds the BLtouch will appear in the LCD menu.
I hope the write helps someone because there is a lot of confusion caused by Youtube videos that refer to earlier versions of the board, BLtouch and firmware files.

wildmanlofland, 2020-07-11:
I was very excited when I read this post.  I had followed this guidance once before with no success but followed again because it all makes perfect sense.  I have a genuine BL Touch and do have the 5 pin JST plug with pins.  I crimped up the pins and plugged in the listed order, no success, BL Touch didn't function and probe tried to go into board.  Removed the black/white wires and unplugged the Z connection.  Plugged in there and changed software on SD card. Still not working.  I've tried for one week to compile my own code, never can get through without errors.
Was so looking forward to this V2.0 board.  My offset is -43, -8 if you could adjust your code and email me a copy.  I would be happy to have a working Ender 3.
Chan

bobcroft, 2020-07-11:
Hi Chan, I have not changed any code or added any offsets.  I just used the firmware downloaded from the BTT github.  I am using the original microswitch Z stop and the BLtouch for bed leveling.  I am still working on getting the correct Z offset when using bed leveling.  Sorry I can't help further.

wildmanlofland, 2020-07-11:
I wish I could get a compile to finish.  I can't for the life of me determine why it stops on compiling.  Always stops saying I have an Include problem, can't find files.  I even load the files, don't make changes and try to compile, it still stops.  I also tried running the github files.  If I tell it to home, it starts going down, BL probe is extended but when it touches it doesn't stop, keeps going to bottom out.  Says Fade is off and I'm still working on the Z offset.  I had this all working on my V1.2 board, maybe change back and sell this one.  Thanks for replying, have a blessed day.....Chan.

wildmanlofland, 2020-07-11:
Bob, I read your hookup to show you are connected to the new 5 pin connection for the BL Touch and the original Z  sensor connected as usual.  The firmware used is the standard firmware-bltouch.bin.  And when you press Home, where does your sensor go??

vexter77, 2020-07-11:
> I wish I could get a compile to finish. I can't for the life of me determine why it stops on compiling. Always stops saying I have an Include problem, can't find files. I even load the files, don't [...]

I had compiling issues. I also believe the precompiled firmware for blt is not working. But anyways my problem compiling was that I did not edit the platform.io.  

This bit of text from the below link is what fixed things. I needed a newer version of marlin that worked with my bltv3.1 and the v2 board. Using that to compile my own was the key.  If you follow that thread you will have a working BLT.  

"Sounds like your PlatformIO.ini file isn't set up for the SKR (that's the error I got when I forgot to change it). If you just downloaded the Marlin FW then added the example Configuration files you'll get that error. To fix that, edit PlatformIO.ini and set default_envs = STM32F103RC_btt_512K"

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/268

bobcroft, 2020-07-11:
Chan, when I press home the BLtouch sensor is over to the left near the X axis stepper.  That doesn't matter because I am homing to the microswitch.  IF one is using the BLtouch as a Z stop (homing sensor) then, using the alternative firmware, the BLtouch should home in the middle of the bed.

bobcroft, 2020-07-11:
Chan,  what you describe as 'I wish I could get a compile to finish' is exactly what happens when you use the wrong firmware from BTT.  You do not need to use platform.IO! It has its own issues.
If you want to use the BLtouch as a Z stop then you use the BTT firmware that includes Z stop in its title AND you connect the black and white wires to the Z stop connection on the main board after you remove the micro switch connection.  
Whichever of the BTT firmwares you use, one of two versions, you must change the name to firmware.bin on the SD card.  Power off the printer, insert the SD card and power up, it will then load the new firmware, and, after a couple of seconds it will bring up the LCD screen.

bobcroft, 2020-07-11:
Chan, another point is to ensure you get the correct firmware for the SKR mini E3 you are using, I used v2.0.  There are not as many Youtube videos for that board, there are more for V1 versions.

wildmanlofland, 2020-07-11:
Thanks for the comments Bob, I'll keep you posted, I do have it working with the Black/white plugged into Zplug.  On home it goes to center and checks twice, if I ask for leveling, it goes to 15 spots to check with BL.  I worked on setting my Z and discovered I had to switch off the z lock that would not let me go below 0 to slide the paper in to get the perfect match.  Got it done though with commands, and testing some now.  Still will play with my own firmware compiled on this computer by this Old man.   LOL...

TinoFpv, 2020-07-15:
Can we just have the conf.h of the Bltouch z_home bin version? Can BTT release it or not? I changed some stuff on my ender and want to apply values in the config.h now.

vexter77, 2020-07-15:
Scroll up. I've linked the answer above.

TinoFpv, 2020-07-15:
> Scroll up. I've linked the answer above.
> […](#)

On what? I don’t see nothing about what I askd.

bobcroft, 2020-07-15:
TinoFpv, I do not know if BTT releases the files that are complied to make their binaries.  If you look at post vexter77 made further up the thread where the issue reference #268 is given and follow that link then the files you will need are in that discussion.  I have read #268 and it is very informative but I have not, as yet, compiled my own firmware.  I believe wildmanlofland has had success with the BTT binaries. I guess nobody reading this thread knows the answer to your question.

bobcroft, 2020-07-15:
Hi Chan, glad you have had some success, I am an old man too!  Personally, at the moment, I prefer not to mess about with my own firmware as I have too many other things to do.  Best of luck and please post your results so that others can benefit.

TinoFpv, 2020-07-16:
Thanks, will try to write @ BTT directly, now how about the noise coming from X and Y axis pushing hard with driver current (advanced settings) on lcd menù set to 1000??
Ender 3 pro X Y axis current is 1.5A so i can imagine a good value for stock motors can be between 600-800...X axis is doing but running anyway hotter then Y.50-60 celsius, is there a reason to set it to 1000 with tmc 2209's?
Thanks

bobcroft, 2020-07-17:
TinoFpv,  I am not sure what you are asking or want to know.  I installed the SKR mini E3 v2.0 with BLtouch and the correct bin file.  It all works perfectly with no over heating of steppers.  It is just the standard set up from BTT.

TinoFpv, 2020-07-17:
Mine BTT fw too, but the X-axis motor becomes untouchable, I don't understand why BTT sets the Ender 3 stock motors current value to 1000 when it should be the half (580) is there something on the TMC2209 drivers special or why are they so high?...

bobcroft, 2020-07-17:
I'll be using my printer today I'll check the X axis motor.  I wasn't aware that BTT set the motor current higher.  I'll have a look at some of the Gcode for a print.  What slicer are you using.

TinoFpv, 2020-07-17:
Simplify 3d, see your Advanced setting>drivers>Tmc driver current (can't remember exactly where it is, I'm in office right now).

bobcroft, 2020-07-17:
Being a cheapskate I do not used the paid for Simplyfy 3D I use the Prusa slicer as I am still only a beginner.  I have done a lot of electronics work that why I find sorting out that side of printing as opposed to printing itself.  I'll check in Prusaslicer, or more specifically in the Gcode that is generated.  Do you know what the Gcodes are for the motor currents?

vexter77, 2020-07-17:
I really think you should go to the other thread and use the .h files
linked their and the bugfix marlin to compile your own.  Your stepper's
will be warm to the touch but not uncomfortable. ...  This problem seems to
be unique to you. Starting over sounds like the best plan.

TinoFpv, 2020-07-17:
> Please go to #268 and read every single post. Find the solution and build a working firmware. I am using the firmware I compiled using that thread and my steps are only slightly warm. My autobedleveling is working great. You will have to go read that thread yourself. I want to help, but only those who are willing to read.
> […](#)

I’m since January in 3d printing world, have compiled my own fw, I don’t need another fw, I’m only asking why here the values on the steppermotors are so high, I’m a newbie here, so who is not reading here??
I can make a new bin, sure, but no sense for a value that you can change from the lcd.

vexter77, 2020-07-17:
I was operating under the assumption that your issue is unique and using the same instructions I used would eliminate the problem for you.

vexter77, 2020-07-17:
I went through 268 and qouted the relevant information in a new reply. This might help you find what you need.

TinoFpv, 2020-07-17:
> I went through 268 and qouted the relevant information in a new reply. This might help you find what you need.

Np, Thanks man.

bobcroft, 2020-07-17:
Hi running the printer and checking the motor currents, X,Y,Z are set to 580.  E is 650.  These are the stock settings from the BTT firmware I presume because I haven't altered them.  The motors are only just warm.

bobcroft, 2020-07-17:
I checked the settings on the LCD panel BTW.  Maybe yours have got inadvertently set to higher values.  I would download the latest BTT firmware and reload.  At this stage of the game I cannot see the point in cobbling up a hybrid firmware because it just adds more opportunity for error into the mix.  I do plan to look at doing my own configs when I get more comfortable with the stock offering.  I do use visual Code and Platform.IO for other tasks but I won't pretend I like that latter because the slightest error configuring it throws up errors and the cause isn't immediately obvious.

ashtonias, 2020-08-25:
> Can we just have the conf.h (and the adv) of the Bltouch z_home bin version? Can BTT release it or not? I changed some stuff on my ender and want to apply values in the config.h now.
> Another little thing, the X Y axes are going way to hot with motor value set to 1000, i think the right value is 650 for X Y and 580 for Z?(Stock motors ender 3 pro 20')
> 
> Probably this one is the right one?
> https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/5293e3c12a7f64e0e4ca1d588535ab92024b17e5/firmware/V2.0/Marlin-2.0.x-SKR-mini-E3-V2.0/Marlin/Configuration.h

#define NOZZLE_TO_PROBE_OFFSET { -40, -10, -2.0 }

bobcroft, 2020-08-25:
BTT do provide the files you want as swell as the complied binaries.  I am adding a link to a very comprehensive post that may guide you further.
https://www.reddit.com/r/ender3/comments/h8y1ia/marlin_20x_guide_skr_mini_e3_v20_ender_3/

tablatronix, 2020-09-04:
So what is the consensus? Does the probe header not work at all , whyt not? I guess ill rewire it back to use the zstop pins, there should be a big notification here about this and how to configure it not just a firmware.bin.. Am I missing something ?

ashtonias, 2020-09-04:
> So what is the consensus? Does the probe header not work at all , why not? I guess ill rewire it back to use the zstop pins, there should be a big notification here about this and how to configure it not just a firmware.bin.. Am I missing something ?

I was never able to get it working. I was however able to get all of the changes I needed from the thread @bobcroft linked to compile marlin with the required changes in order to get it working using the white/black wires plugged into the Z stop headers on the board.

tablatronix, 2020-09-04:
Do you know if it needs `Z_MIN_PROBE_ENDSTOP_INVERTING` when using z_stop pin
It did on my creality board

bobcroft, 2020-09-04:
The probe header socket works fine for the servo Gnd, 5V and servo connection (PA1).  The socket doesn't work for the probe because it is apparently wired to the wrong pin on the microcontroller.  The probe does work properly when connected to the Z stop 2 pin socket.  Z_MIN_PROBE_ENDSTOP_INVERTING is set to false.

tablatronix, 2020-09-04:
Sorry to be clear, I am compiling marlin not using firmware, it is not a pin definition issue

bobcroft, 2020-09-04:
Hi Shawn

 

I Didn’t say it was a pin definition issue, I explained why the obvious option doesn’t work, well I think I did!  I presume you are modifying a standard Marlin to your needs as in modifying the config files.  Once I got the connections right I haven’t had any further issues with the Marlin settings.  On the Official Creality Ender 3 Facebook group a guy called Joe Lacsamana posted some configurations for the SKR mini E3 V2 board using Bltouch.  Might be worth comparing your settings to his, I can vouch for the fact his work.

 

I could send you copies of mine but I have made some changes on the printer that I haven’t put into the configs yet, like PID tuning and bed offsets because I have a direct drive.

 

Bob

tablatronix, 2020-09-04:
Right, Yeah I am investigating this
`The socket doesn't work for the probe (as the Z stop) because it is wired to the wrong pin on the microcontroller (PC14). `
and why and how to get it to work. and why the solution is to "do not use the probe header.." cause that is silly

bobcroft, 2020-09-04:
Sorry, I don't think I can help further.  I am content that it works for me and I try to help others by sharing what I did, with help from others, to get it working.  I most definitely do not say my way is right.  I am not really bothered that the Z probe socket doesn't work for all of the connections as long as I can get a working set up which I have.
Good luck with your quest.

tablatronix, 2020-09-04:
hah no problem @bobcroft ,I am not actually looking for support, I have it working using the z_pin header, 
 I am looking to fix this issue or make BTT address this and update the docs.

vexter77, 2020-09-04:
You are beating a dead horse.  BTT has apparently abandon this product and its customers.  New enders come with 32bit boards and quiet drivers.

superpsych0, 2020-12-22:
> Nothing unusual here, There are several versions of Bl touch, and 2 pins layout. I got the 3.1 and with the skr mini e3 v2 I had to switch the the first 2 probe wires.

By this, you mean the GND and + right? 

Goddamn it, friend one probe already plugging it into the board sight unseen, got the "you let the smoke out" smell and turned it right off... now got a replacement, am being very careful about the wiring.

I'm thinking of just cutting off the plastic connector and simply soldering the wires to the CORRECT leads. Any better ideas?

SoapyCariboo, 2021-03-04:
I have an Ender 3 - and just upgraded with a BTT SKR-mini E3 V2.0 plus, I also bought a Creality BLTouch V3.1. I used the most updated firmware from BTT/GITHub - firmware-bltouch.bin. My BLTouch, on startup turns red, deploys the probe 2 x and then stays red. On LCD I can see the addition of the BLTouch, but when I have it selftest, deploy and stow nothing happens. I am using the BLTouch supplied "Pinboard A" to attach to the SKR as per the instructions with it and I still can't get the BLtouch to work as advertised. I am new to this hobby and for now I would like to use the available firmware instead of compiling my own.
Any help or ideas would be appreciated.

panosseretis, 2021-11-03:
hello friends
I have BLTOUCH with 5 pins connector. Installed it on the default dedicated port for BLTOUCH that exists on the skr mini v2 board. I cant seem to make BLTOUCH work. When i put a ZOFFSET to what ever value... the nozzle stays about 2 mm above the bed. When i set it to auto home... it uses the Z, X and Y microswitch. It is behaving like it does not have BLtouch, although i have installed the firmware.
 Do i need to remove the Z endstop? 
 What i m doing wrong? 
 Is there a guide for BLTOUCH (5 pins connector) and firmaware-bltouch.bin file?

I m searching all over internet, with no luck. Can anyone help me?

Thank you

jakedoesgithub, 2022-08-03:
Hi it’s not you it’s the recompiled firmware it’s incorrect here are the older btt version that work https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0