bigtreetech/BIGTREETECH-SKR-mini-E3 #286: NeoPixels not turning off with M150

ericsnis, 2020-06-12:
### Description

I'm having a little trouble with NeoPixels.  They seem to work, except they don't turn off when I issue a M150 command.

I added a Adafruit NeoPixel 8 LED stick to a SKR mini E3 1.2 board [.](https://www.adafruit.com/product/1426).  The LEDs are powered by the 5V on the NeoPixel header which according to the notes is acceptable.  The LEDs cycle through the power on test just fine (Red -> OFF -> Green -> OFF -> Blue -> OFF).  The LEDs are all off after the power on test, so I know they do turn off by some command that is send by the board.  I can also set the colors to what I need with M150, but I cannot turn them off.  Issuing the M150 command without any options turns off about half of the LEDs but sets the remaining ones to seemingly random colors.

I also tried specifically setting brightness to 0 with M150 B255 P0 R255 U255 and setting all three color components to zero.  In each case the lights don't switch off.  The lights are apparently capable of being switched off since they turn off after the power on test.  I'm guessing there is something odd in the M150 handling so it isn't sending the appropriate data to the chips.

I am using Marlin 2.0.5.3. from GitHub compiled with the default NeoPixel GRB options in configuration.h, just changing the pin number and number of pixels.

The LED chips are WS2812B which are RGB, not RGBW.  The NeoPixel stick has capacitors onboard, so the requirement for that should be fulfilled.

### Steps to reproduce

1. Change NeoPixel library in platformio.ini to use https://github.com/bigtreetech/Adafruit_NeoPixel and comment out original Adafruit NeoPixel.  Also remove Adafruit NeoPixel from lib_ignore.
2. Compile Marlin 2.0.5.3 with NeoPixel options changed.
3. Use Octoprint terminal to turn on NeoPixels for white light with M150 B255 P127 R255 U255. Lights turn on with white color.
4. Send M150 command, lights switch to random color instead of off.

**Expected behavior**
NeoPixels should turn off following M150 command with no options.

**Actual behavior**
Random NeoPixel colors happen when M150 command is issued.

### Additional Information
Configuration.h is
`//#define RGB_LED
//#define RGBW_LED

#if EITHER(RGB_LED, RGBW_LED)
  //#define RGB_LED_R_PIN 34
  //#define RGB_LED_G_PIN 43
  //#define RGB_LED_B_PIN 35
  //#define RGB_LED_W_PIN -1
#endif

// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     PC7       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 8       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  // #define NEOPIXEL_BKGD_LED_INDEX  4               // Index of the LED to use
  // #define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif`

ericsnis, 2020-06-12:
I think that I fixed it. There is another issue, #123, where odd issues appeared communication based. The solution was to use a different NeoPixel library which included a delay.

Platformio.ini was changed to comment out Adafruit Neopixel and add `https://github.com/CommandoreBombardiero/Adafruit_NeoPixel`

I changed .pio\libdeps\STM32F103RC_btt_512K\Adafruit NeoPixel\Adafruit_NeoPixel.cpp to have delay.h which in my case the relative path was `#include <..\..\..\..\Marlin\src\HAL\shared\Delay.h>`

Not sure if this is something that should be addressed in the master branch or if this is a rare issue.

argus0209, 2020-07-01:
Where in Platformio.ini are ya'll adding "https://github.com/bigtreetech/Adafruit_NeoPixel" and what line are you removing adafruit neopixel from lib_ignore? i have like 22 lib_ignores in the PLatformio.ini.

Thanks for any info.

Yaha500, 2020-07-06:
# MEEB_3DP (STM32F103RCT6 with 512K)
#
[env:STM32F103RC_meeb]
platform          = ${common_stm32f1.platform}
extends           = common_stm32f1
board             = MEEB_3DP
platform_packages = tool-stm32duino
build_flags       = ${common_stm32f1.build_flags}
                    -DDEBUG_LEVEL=0
                    -DSS_TIMER=4
                    -DSTM32_FLASH_SIZE=512
                    -DHSE_VALUE=12000000U
                    -DUSE_USB_COMPOSITE
                    -DVECT_TAB_OFFSET=0x2000
                    -DGENERIC_BOOTLOADER
extra_scripts     = pre:buildroot/share/PlatformIO/scripts/STM32F1_create_variant.py
    buildroot/share/PlatformIO/scripts/STM32F103RC_MEEB_3DP.py
lib_deps          =
  TMCStepper@~0.7.0
  Adafruit MAX31865 library@~1.1.0
  U8glib-HAL@0.4.1
  Arduino-L6470@0.8.0
  SlowSoftI2CMaster
  LiquidTWI2@1.2.7
  #Adafruit NeoPixel=https://github.com/CommandoreBombardiero/Adafruit_NeoPixel
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SoftwareSerialM
  USBComposite for STM32F1@0.91
lib_ignore        = SPI
debug_tool        = stlink
upload_protocol   = dfu

Yaha500, 2020-07-06:
#
# The 'common' values are used for most Marlin builds
#
[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts      = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags        = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__ -fmerge-all-constants
lib_deps           =
  LiquidCrystal@1.5.0
  TMCStepper@~0.7.0
  Adafruit MAX31865 library@~1.1.0
  Adafruit NeoPixel=https://github.com/CommandoreBombardiero/Adafruit_NeoPixel/archive/master.zip
  U8glib-HAL@0.4.1
  Arduino-L6470@0.8.0
  SlowSoftI2CMaster
  LiquidTWI2@1.2.7
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip

# MEEB_3DP (STM32F103RCT6 with 512K)
#
[env:STM32F103RC_meeb]
platform          = ${common_stm32f1.platform}
extends           = common_stm32f1
board             = MEEB_3DP
platform_packages = tool-stm32duino
build_flags       = ${common_stm32f1.build_flags}
                    -DDEBUG_LEVEL=0
                    -DSS_TIMER=4
                    -DSTM32_FLASH_SIZE=512
                    -DHSE_VALUE=12000000U
                    -DUSE_USB_COMPOSITE
                    -DVECT_TAB_OFFSET=0x2000
                    -DGENERIC_BOOTLOADER
extra_scripts     = pre:buildroot/share/PlatformIO/scripts/STM32F1_create_variant.py
    buildroot/share/PlatformIO/scripts/STM32F103RC_MEEB_3DP.py
lib_deps          =
  TMCStepper@~0.7.0
  Adafruit MAX31865 library@~1.1.0
  U8glib-HAL@0.4.1
  Arduino-L6470@0.8.0
  SlowSoftI2CMaster
  LiquidTWI2@1.2.7
  **#Adafruit NeoPixel=https://github.com/ccccmagicboy/Adafruit_NeoPixel#meeb_3dp_use**
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SoftwareSerialM
  USBComposite for STM32F1@0.91
lib_ignore        = SPI
debug_tool        = stlink
upload_protocol   = dfu

imperialdub, 2020-07-28:
@ericsnis I had this same exact issue, and thanks for pointing out the fix. After hours and hours of research, I can confirm that my Neopixels are finally working as intended. THANK YOU!

georgebohnisch, 2020-08-07:
> I think that I fixed it. There is another issue, #123, where odd issues appeared communication based. The solution was to use a different NeoPixel library which included a delay.
>
> Platformio.ini was [...]

This resolved my issue with the colors being off after the initial startup colors. I'm using an official Adafruit 8 NeoPixel RGBW strip. Thanks!

HeyRay2, 2020-08-09:
I had the same issue with M150 on my SKR 1.4.

I was able to correct it by adding a 330ohm resistor on the data line right before the first pixel, likely indicating there was some noise in the data line causing signal issues to the pixels.

monstis, 2020-08-24:
> I think that I fixed it. There is another issue, #123, where odd issues appeared communication based. The solution was to use a different NeoPixel library which included a delay.
>
> Platformio.ini was [...]

This is the last three days of my life..... fixed - sorted - done - working. As soon as I saw "RED, GREEN, BLUE" I knew the struggle was over. It was the #include path in **Adafruit_NeoPixel.cpp.**  The Library - **Adafruit NeoPixel=https://github.com/CommandoreBombardiero/Adafruit_NeoPixel/archive/master.zip**. Also I had to use **#define NEOPIXEL_TYPE NEO_GRB** Thankyou very very much! BTT SKR mini E3 V2.0

abyMosa, 2020-09-08:
had the same issue, @monstis solution fixed it, big thanks.

DominikMaga, 2020-10-08:
in Marlin 2.0.7 in section [features] replace `NEOPIXEL_LED            = Adafruit NeoPixel@1.5.0` 
to `NEOPIXEL_LED            = NeoPixel=https://github.com/CommandoreBombardiero/Adafruit_NeoPixel/archive/master.zip`
and don't forget comment in Marlin\src\HAL\STM32F1\inc\SanityCheck.h `// #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"`

sakoula, 2020-10-15:
hi there! I tried Marlin 2.0.7.1 and currently NEOPIXEL is not working on my 1.2 SKR Mini E3

I have `platformio.ini`:

```
NEOPIXEL_LED = NeoPixel=https://github.com/CommandoreBombardiero/Adafruit_NeoPixel/archive/master.zip
                          src_filter=+<src/feature/leds/neopixel.cpp>
```

and commented out the SanityCheck.h.

I get a failing compiled message:

```
.pio/libdeps/STM32F103RC_btt_512K/NeoPixel/Adafruit_NeoPixel.cpp:51:10: fatal error: replace with path to Delay.h file: No such file or directory
 #include <replace with path to Delay.h file>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/STM32F103RC_btt_512K/lib8e7/Wire/SoftWire.cpp.o
*** [.pio/build/STM32F103RC_btt_512K/libc41/NeoPixel/Adafruit_NeoPixel.cpp.o] Error 1
================================================= [FAILED] Took 56.96 seconds =================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RC_btt_512K  FAILED    00:00:56.957
============================================ 1 failed, 0 succeeded in 00:00:56.957 ============================================
```

anyone who overcame this?

Thanks!

sakoula, 2020-10-15:
Hi everyone. It eventually worked following the @monstis post and doing the changes that @DominikMaga is pointing.

Up to now I have been using `https://github.com/bigtreetech/Adafruit_NeoPixel` library that did not require the `#include <..\..\..\..\Marlin\src\HAL\shared\Delay.h>` Delay.h header file. Do we know that other than compiling or booting is safe to use that?

Thanks!

monstis, 2020-10-15:
> Hi everyone. It eventually worked following the @monstis post and doing the changes that @DominikMaga is pointing.
>
> Up to now I have been using `https://github.com/bigtreetech/Adafruit_NeoPixel` [...]

I believe 'bigtree tech' and official 'Marlin Adafruit NeoPixel' versions dont play well with STM32F103RC boards. 'CommandoreBombardiero' link is best for the latest version of marlin (2.0.7) I have it setup and functioning well on my SKR Mini E3 V1.2 & V2.0. I am not an expert by any means. I cant take credit for the solution. This thread is where I found it. The only thing that doesnt work is the (event) green light after a successful print. It works from the SD card... Not from octoprint. This to be expected. I put a M150 command to turn the pixels green once the end gcode is triggered. (M150 U125 I255) 
[image]

877dev, 2020-12-19:
@ericsnis thank you!!
It's working for me now on Ender 3 Pro with SKR mini E3 v2.0.0 and 8 LED Neopixel.

You saved me a lot of time!