bigtreetech/BIGTREETECH-SKR-mini-E3 #287: [BUG] (short description)

Ramoss007, 2020-06-12:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Ramoss007, 2020-06-12:
> Running BL Touch V2.0 firmware with Z stop on Ender 3, cannot adjust Z Offset?

Tried to print a file, when it started I noticed the Z was lost from previous setting. Went to try to adjust Z stop but the nozzle does not move towards the bed? Numbers change but the Z axis no movement. 

All hardware works, reverted to non-BL touch firmware all ok.

Anyone have any ideas?

JPCossu, 2020-06-12:
I have the same problem, impossible to adjust the Zoffset on the SKR mini E3 V2, I don't find the solution either...

JPCossu, 2020-06-16:
Problem solved ! You have to activate the babysteeping in the firmware : uncomment : BABYSTEP_ZPROBE_OFFSET in config_adv

Ramoss007, 2020-06-21:
> Problem solved ! You have to activate the babysteeping in the firmware : uncomment : BABYSTEP_ZPROBE_OFFSET in config_adv

Thank you I was away will take a look this weekend. Thank you