bigtreetech/BIGTREETECH-SKR-mini-E3 #288: pin connector

romandie, 2020-06-13:
Hello, I have bltouch with z homming but i look the picture and i not have the pin in my card for gnd, yellow ans +5V 
my card : https://www.amazon.fr/gp/product/B082CJHKYC/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
picture : https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/bltouch-2.0-2.jpg
Where i connect the pin ? 
Thanks

CarenaAndrea, 2020-06-13:
your card is v1.2 not v2.0
here you can find hot to connect:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2

romandie, 2020-06-13:
Ok, but i put firmware-bltouch-for-z-homing.bin in my SD card, after i turn ON my printer and the screen say the printer is not connect ? And i have a another problem, i have a BLTOUCH RIGHT (44, -13) ? and E3DV6

romandie, 2020-06-13:
I'm really stupid. I put the SD card in the TFT and not in the motherboard. There was already a card inside. Now the problem I have is to configure my BLtouch which is on the right and my BMG direct drive extruder. Do a PID I think because I have an E3DV6. I guess I have to configure my marlin by hand and there with my configuration. This is the first time that I configure a Marlin so I don't know where to start.