bigtreetech/BIGTREETECH-SKR-mini-E3 #289: MicroSD Card Slot not recognizing any micro sd card

rambokd, 2020-06-14:
My newly installed SKR mini E3 will not recognized any micro sd card installed. I've rebooted, reset, re-format the micro sd card and the TFT35 v3 has no problem reading any of the sd cards and if I could flash the firmware any other way maybe it wouldn't be an issue but I'm thinking a replacement board is my only solution. The board has the factory Marlin 2.0bug fix installed and no changes have been made since I can not get any firmware updates..

deamonSLINKY, 2020-06-21:
Just installed my SKR Mini E3 today and I am having the same issue. Luckily I have a TFT35 as well and can print from that. When I try to open the onboard SD card I get an error message and the printer completely locks up. Not sure what to do since I cannot update firmware.

PiyiZeta, 2020-07-03:
I have SKR2.0 for E3 and TFT35, only can read the SD card in Marlin mode.

rambokd, 2020-07-03:
> I have SKR2.0 for E3 and TFT35, only can read the SD card in Marlin mode.

I have the TFT35 V3 and once the new V2 board was installed on an Ender 3, I had to initialize the media in Marlin then it showed up on the TFT. I have the UART speed at 115200 and the EXP 1 and rs232 plug in use. I think everything in the touch screen mode is reported via the UART rs232 connection and the exp1 ribbon cable is for Marlin.

PiyiZeta, 2020-07-03:
Thanks, another doubt if I print from udisk  (usb pendrive) using the tft usb port the auto resume function doesnt work. It only work printing from an sd card? Thanks

ljbeng, 2021-12-17:
I could not use the on-board sd to find the gcode files but I could use the on-board sd to do the firmware update.  Once I recompiled the latest today, I got 2 options for SD card and the on-board card works again!