bigtreetech/BIGTREETECH-SKR-mini-E3 #290: [BUG] USB device not recognized when plugging in thermistors

hr121, 2020-06-14:
### Bug Description

When plugging in the either of the temperature sensors to the board the device can no longer be recognised. However if i disconnect the temperature sensors it seems to connect again. So i think the issue is with the temperature sensor settings or something along that line. This is on a custom built printer.

### My Configuration

Link to config files: [https://drive.google.com/file/d/17LR1ErjJTCcIPhGYAJFluQ76doJpb8QW/view?usp=sharing]

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4776955/config.zip)

### Steps to Reproduce

With the config files linked above, having the temperature sensors disconnected the device is recognised fine and stepper motors can be controlled and homed. Then powering down the device, reconnecting the temperature probe and powering the device back on the device can no longer be connected to.

#### Additional Information

two white wires in the bottom left hand corner is the wires for the hotend thermal probe and the purple and blue wires next to that one is the wires for the heated bed.

[image]
[image]
[image]

mitrofan110, 2020-06-17:
One of the sensors probably has contact with the metal case of the printer ..

hr121, 2020-06-17:
The sensor was faulty

closed 2020-06-17 by hr121