bigtreetech/BIGTREETECH-SKR-mini-E3 #292: [BUG] TFT35 E3 V3.0 and SKR mini E3 no printer attached

vulbas, 2020-06-15:
Hello
according to the info i found, need to change the port.
Set "#define SERIAL_PORT -1" and "#define SERIAL_PORT_2 0". Set "#define BAUDRATE 115200"
can you update the firmware to make it work with the screen, bltouch for Z-homing

wdrovere, 2020-06-18:
Hello! I have the same problem, but I don't know how to insert these settings that you mentioned! how should I proceed? i really appreciate your help!

cvols, 2020-06-19:
@wdrovere You can find those variables in the Configuration.h file.  Click on the Marlin Folder and then you'll find the file there.

However I have these variables set as @vulbas said and I'm still having the "no printer attached" error.

Ramoss007, 2020-06-21:
When that happened to me, on the LCD menu I had to set the baud rate to 115200 and printer was found!

wdrovere, 2020-06-26:
@cvols i solved! the tft cable was inverted in my motherboard, now everything works great!

dissocio0, 2020-12-10:
my touch mode is working but the marlin emulation is not working ... i am working on an ender 5 tried with 2 skr 1.4t boards and now with a skr e3 2.0 / ft 35 3.0 e3...... i tried every configuration possible from config.h version2006-2008.... i have no clue what to do .. plus if i am not using unified version of marlin i get the kill halt error when i home my printer and the reality probe does not stove before homing, so i have to manually stove the probe from the menu and home ... and then i get the error .... i got it working once with the unified (no emulation mode) but it would not extrude unless i started a print job and the prints sucked and were terribly underextruded plus the mesh bed leveling (bilinear) seemed not to work since it was uneven (even if i manually leveled the bed almost perfectly beforehand and edited the mesh to perfection)... any help please??