bigtreetech/BIGTREETECH-SKR-mini-E3 #294: [FR]Z-hop after stop printing (v1.2)

st4szek, 2020-06-17:
Hi,
I have Ender 3. I installed SKR mini E3 V1.2. It works good. I flashed new firmware v2.0. I like it more than v1.2. But on v2 USB connection doesn't work. So I came back to 1.2. But then I realized that one little feature is missing in 1.2 comparing with 2. Head doesn't Z-hop after stopping the print. Can you enable it? Or can I use firmware v2.0 on board 1.2? If yes can you verify USB serial connection?
I didn't prepare the firmware by myself. I used pre-compiled bin files.

JPCossu, 2020-06-18:
hello the V2 firmware is reserved for the new mini e3 V2, you have to use the V1.2 firmware for your card.

st4szek, 2020-06-18:
@JPCossu thanks for clarifying. I was not sure. But anyway is there a possibility to add z-hop when stopping the print? How can I enable it in FW and flash my board? I've never done this. Is this like adruino programming? Or I need to burn bootloader on skr first?