bigtreetech/BIGTREETECH-SKR-mini-E3 #295: SKR mini E3 V2.0 Firmware compiled for a stock CR-10S

webwookie, 2020-06-19:
Can you compile a firmware for the CR-10S?

Hedda, 2020-07-01:
+1 feature request for official firmware support for Creality CR10s (stock hardware) upgraded with the new SKR Mini E3 V2.0

Hedda, 2020-07-01:
Also see feature request https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/309

Hedda, 2020-07-01:
Please also see this upstream request for config example files -> https://github.com/MarlinFirmware/Configurations/issues/141

Gamester17, 2020-10-01:
Please help test submitted Marlin config files in https://github.com/MarlinFirmware/Configurations/pull/237 and https://github.com/MarlinFirmware/Configurations/pull/238 that should work with stock CR-10 or stock CR-10S upgraded with BigTreeTech SKR Mini E3 2.0 board.

Gamester17, 2020-10-01:
+1 by the way for including pre-configured and pre-compiled Marlin Firmware binaries for stock CR-10S as well as for stock CR-10.

Gamester17, 2020-10-05:
FYI, MarlinFirmware Configurations upstream have now committed BigTreeTech SKR Mini E3 2.0 config examples for both stock Creality CR-10 and stock Creality CR-10S

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/BigTreeTech%20SKR%20Mini%20E3%202.0

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%20Mini%20E3%202.0

Now all that is missing is for @bigtreetech to provide precompiled SKR Mini E3 2.0 binary firmware files for CR10 and CR10S .

Drewvee2, 2021-06-20:
Hi there, is there anyone who get a skr on a cr10v2 pls ? i need help with the compilation. thx

PocketBees, 2022-03-11:
Does anyone have the compiled binary file? I've been fighting compiling errors for days.

Tobmoc, 2022-10-02:
> FYI, MarlinFirmware Configurations upstream have now committed BigTreeTech SKR Mini E3 2.0 config examples for both stock Creality CR-10 and stock Creality CR-10S [...]

Is that possible that you compile that? I have try many hours to compile the BIN files, but every try i get some errers. best regrads michael