bigtreetech/BIGTREETECH-SKR-mini-E3 #296: Been using klipper for a while. Cant flash marling back on it.

iH8teF1ames, 2020-06-22:
Ive been using klipper for a while now but decided to switch back to marlin for some features id like to be supported. When I load the SD card with the "firmware.bin" file it doesnt do anything. If I take it out and look at the contents, it has overwritten it with "FIRMWARE.CUR" as expected. ive tried both self compiled and the pre-compiled one from github. LCD stays blank and octoprint cannot see it. Any ideas?

Logo2001, 2020-06-26:
Maybe install the boot loader again.
Somewhere on the Github/Bigtreetech/YOUR BOARD described.

macrossactual, 2020-06-27:
> Ive been using klipper for a while now but decided to switch back to marlin for some features id like to be supported. When I load the SD card with the "firmware.bin" file it doesnt do anything. If [...]

I have had the same experience with attempting to install either of the 2.0 binaries on the mini. The status light does flash so it is writing to the board, and generating the .cur,  but the only way I've been able to get the mini to boot to a usable state is by flashing the 1.2 firmware.

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2