bigtreetech/BIGTREETECH-SKR-mini-E3 #297: No Printer Attached with New SKR Mini E3

mkn1116, 2020-06-22:
I had purchased a new SKR Mini E3 v1.2 but my TFT35 v3.0 was saying 'No Printer Attached'. I thought it was defective so I purchased the SKR Mini E3 v2.0 and had the same issue. I tried to use my TFT35 v2.0 again with same issue. I tried both boards with preloaded firmware and I updated the firmware and set the baud rate to 115200. I have an SKR Mini E3 v1.2 in my older Ender 3 that has been working since last year. I'm not sure what I'm missing with this new setup. Thanks for the help in advance!

Newbe-printer, 2020-06-23:
I am having the similar issue.  I have the V1.2 with TFT35 V3.0 .  It was working ok. I tried hooking up a BLTouch and ever since, the board isn't recognized by the TFT35.  I hooked up the original Ender 3 screen and it doesn't seem to recognize the skr mini.  I have tried updating the firmware on the skr but it won't recognize the firmware to auto update.  I am on a baud rate of 115200 and have the most up to date firmware on the skr mini.

Logo2001, 2020-06-26:
I think the port speeds do not match between TFT and board.
Please try the "Connection" item in the "Machine" menu.
Change to another port speed and wait a bit.
If nothing happens, try the next one.
Or am I misunderstanding you?

Logo2001, 2020-06-26:
[https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/292#issuecomment-649938961](url)
Cable Connection OK?

Newbe-printer, 2020-06-26:
I go to Menu -> Settings -> Machine but there isn't a "Connection" item to choose.  I have "Parameter" and "Custom" but neither have settings for port speed.

If I go to Menu-> Settings there is an item that shows connection speed.  I am assuming that is what you were referring to.  I am not sure how long 'a bit' is.  5 minutes should be long enough, i think.  Also, on this same page, there is a item to "Disconnect"  I did push that at one point.  I would assume that unplugging the printer and then restarting it would reconnect anything.

I have tried multiple speed settings and waited but my boards still are not talking to each other.  Other ideas?

Thank you for your help.  I am really new to this and I am struggling to get both components to work well together all the time.

David

Logo2001, 2020-06-26:
Ok, please check the correct Cable connection (5pin, 4+1) from Port "TFT" on Mainboard to RS232 Port on Display's backside. The one loose single plug (+1) is always the +5V.  Match both sides. May be one side is wrong?

Outlook for Android herunterladen

Newbe-printer, 2020-06-27:
I switched the connector on the mainboard.  The screen had lines across it and still showed the mainboard was not connected.  I could tell that the change I made wasn't correct.  I currently have the 5 pin connected to the mainboard with the +1 pin on the left if you look at the mainboard with the drivers at the top.  

I tried putting the original Ender 3 display back on and although it was getting power, nothing would show up on the screen.  

I am rather confused because the skr mini and the TFT was working.  I have tried downloading the TFT firmware again and that did not help.  I have tried installing the firmware again on the skr mini but it did not seem to be starting to install the firmware.  I tried holding the reset button for 10 seconds and that didn't start the upload.  I am wondering if something has happened to the mainboard.

Again, thank you for any advice you can give me.

David

Logo2001, 2020-06-27:
Your connection to the Mainboard is now correct. Please check the TFT connection also. I'm confused to the datasheet for the tft 35 3.0 (pin).

The print on Mainboard says 5v is left (RS232) and the legend says 5v is right.

I dont know what is correct. I Think is left.

Check it out and then test the connection speed Settings again.

Outlook for Android herunterladen

Logo2001, 2020-06-27:
I just remembered that the cabel has a 5 pin connector on. Now the connection in the Mainboard is wrong. Turn it the other way again.

Outlook for Android herunterladen

On Sat, Jun 27, 2020 at 10:32 PM +0200, <logo2001@freenet.de> wrote:

Your connection to the Mainboard is now correct. Please check the TFT connection also. I'm confused to the datasheet for the tft 35 3.0 (pin).

The print on Mainboard says 5v is left (RS232) and the legend says 5v is right.

I dont know what is correct. I Think is left.

Check it out and then test the connection speed Settings again.

Outlook for Android herunterladen

Logo2001, 2020-06-27:
Can you check the cable with a Multimeter?

May be a wire is broken.

On Sat, Jun 27, 2020 at 10:41 PM +0200, <logo2001@freenet.de> wrote:

I just remembered that the cabel has a 5 pin connector on. Now the connection in the Mainboard is wrong. Turn it the other way again.

Outlook for Android herunterladen

On Sat, Jun 27, 2020 at 10:32 PM +0200, <logo2001@freenet.de> wrote:

Your connection to the Mainboard is now correct. Please check the TFT connection also. I'm confused to the datasheet for the tft 35 3.0 (pin).

The print on Mainboard says 5v is left (RS232) and the legend says 5v is right.

I dont know what is correct. I Think is left.

Check it out and then test the connection speed Settings again.

Outlook for Android herunterladen

Newbe-printer, 2020-06-28:
As I thought about this situation a bit better, I realized it all started when I tried adding my new BL Touch to the printer.  I had downloaded the firmware with z homing onto the printer and that was when things stopped working. I put the regular firmware, without the BL Touch line active, and the skr mini board is now shown as attached.  I also had to put back the original screen board.  The TFT still isn't working properly for me.  The TFT will bring back up the screen to choose between the 2 types of interface.  When I choose one type of interface, it takes me to it and then immediately reverts back to the screen to choose the type of interface.

nilspear, 2020-07-01:
I had the same problem; running SKR Mini E3 v2.0 with TFT35 E3 v3.0 and BL Touch. 
This was the fix for me: 
        1) reinstalled marlin firmware -- no change
        2) clicked Menu
        3) clicked Settings
        4) click UART Speed until it reads 115200 -- Problem fixed!

mkn1116, 2020-07-09:
I spent over 3 hours the other night and had no resolution. I have a new SKR Mini E3 v1.2, new SKR Mini E3 v2.0, TFT35 v2.0, and TFT35 v3.0. I assembled them in multiple configurations and could not get the touch screens to establish a solid connection with No Printer Attached every time and no functionality with any feature. Changing the UART to 115200 did nothing. When I connected the original Creality board it worked fine. The next thing I'm going to try is to remove the SKR Mini E3 v1.2 from my working Ender 3 and see if that works. However, the issue is with the connection of the screen to the board and I'll use the other cable as well.

Newbe-printer, 2020-07-09:
mkn1116 - I never got the TFT35 V3.0 to consistently work with the skr mini e3 V1.2 .  It was hit and miss, mainly miss.  I gave up.  I put the original screen back in and downloaded the regular(non BLTouch) firmware and I have been running ever since.  I don't know how to make changes to the firmware so I went backwards with components until I got a working machine.

mkn1116, 2020-07-12:
Hey Newbe-printer - After installing the new SKR Mini E3 V1.2 I'm able to move/home the X, Y, and Z with both TFT35 V2.0 and TFT35 V3.0, the temps read 0/0 for both nozzle and hotend, and I can access the gcodes but it will not allow me to select a print. I had installed the original screen and saw the temps readings at room temp and was able to select the print and both nozzle and bed heated up to temp but it just pauses there and doesn't actually print anything. I have given up on the SKR Mini E3 V2.0 board. I'm getting really fed up with these new boards. I've been running my older Ender 3 with SKR Mini E3 V1.2 for almost a year without any hiccups.

cmmx34, 2020-07-15:
> I had the same problem; running SKR Mini E3 v2.0 with TFT35 E3 v3.0 and BL Touch.
> This was the fix for me:
> 1) reinstalled marlin firmware -- no change
> 2) clicked Menu
> 3) clicked Settings
> 4) click UART Speed until it reads 115200 -- Problem fixed!

Thank you. This solution worked for me.