bigtreetech/BIGTREETECH-SKR-mini-E3 #301: Fail to compile 1.2 ver

Claude-56, 2020-06-27:
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4840423/Configuration.zip)

Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\shared\backtrace\unwarmmem.cpp.o
~~~~~~~~~
compilation terminated.

I get the same error when working with Marlin autobuild or directly with PlatformIO

jjdmon, 2020-06-28:
Same issue compiling for the SKR mini E3 V2.0 using the PlatformIO IDE in Atom.  It updated some components and libraries when I started it and won't compile anymore.  Maybe I'll try VS Code again but I switched to Atom because VS Code compiled and the resulting firmware was inoperative.

Claude-56, 2020-06-28:
Thank you so much
In fact I managed to compile the BugFix version instead of the lastest release.

closed 2020-06-28 by Claude-56