bigtreetech/BIGTREETECH-SKR-mini-E3 #302: [BUG] Compiling Errors "libmaple/gpio.h" not found

BelSD, 2020-06-27:
I'm trying to copy with VS Code, which always works until now and suddenly impossible to compile, I always get the error that "libmaple / gpio.h" cannot be found ...

[image]

runninglennart, 2020-06-27:
I get the same error. I have changed the name in the platformio file to STM32F103RC_bigtree_512K and have replaced the configuration and configuration_av file, which should be all you have to do to get the firmware.bin file for the skr mini e3 v1.2 from original marlin 2.0. 
Ive also tried to use the firmware folder that is provided here, but even without any changes, it doesnt build for me (and gives me the same error as above).

michaelosbornegit, 2020-06-27:
Had this same issue with the latest Marlin release. Just fixed it by using the `bugfix-2.0.x` and following the BTT instructions on their github. Note that you need to also use the updated `bugfix-2.0.x` configurations for the correct `Configuration.h` and `Configuration_adv.h`. Running a test print right now with the bugfix firmware loaded.

runninglennart, 2020-06-27:
> Had this same issue with the latest Marlin release. Just fixed it by using the `bugfix-2.0.x` firmware from the Marlin download page and following the BTT instructions on their github. Note that you need to also use the updated `bugfix-2.0.x` configurations from the same download page for the correct `Configuration.h` and `Configuration_adv.h`. Running a test print right now with the bugfix firmware loaded.

Oh, i thought there werent any configs for BigTreeTech boards in that branch, but as it turns out, the E3 boards in bugfix-2.0.x are now under "Creality/Ender-3" and not under "BigTreeTech". Im gonna try your fix and see if it helps and keep you posted here. Thanks for the help!

runninglennart, 2020-06-27:
Alright, with your fix im able to build with the correct configurations. Now the reason for my firmware modifications are that i want to use NeoPixels, but with the changes mentioned in the BigTreeTech documentations ("go to the platformio.ini file comment out Adafruit NeoPixel@1.2.5 add https://github.com/bigtreetech/Adafruit_NeoPixel") dont seem to work and spit out following error: 
`Compiling .pio\build\STM32F103RC_btt_512K\src\src\core\serial.cpp.o
In file included from Marlin\src\feature/leds/neopixel.h:34:0,
                 from Marlin\src\feature/leds/leds.h:33,
                 from Marlin\src\MarlinCore.cpp:107:
.pio\libdeps\STM32F103RC_btt_512K\Adafruit NeoPixel_ID28/Adafruit_NeoPixel.h:361:3: error: 'GPIO_TypeDef' does not name a type
   GPIO_TypeDef *gpioPort;       ///< Output GPIO PORT
   ^~~~~~~~~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\MarlinCore.cpp.o] Error 1`

Does anyone know how to fix this, would be greatly appreciated.

mIRACLELive1, 2020-06-28:
```
[env:STM32F103RC_btt_512K]
platform          = ststm32@<6.2.0
```
because new library of ststm32 is v7.0, for Marlin need change back to 6.1.1

In VSCode - PlatformIO - Quick Access - Updates - Update All.

Platform Manager  
================  
Platform ST STM32
--------
Updating ststm32                         @ 7.0.0          [Up-to-date]

Platform ST STM32
--------
Updating ststm32                         @ 6.1.1          [Detached]

cmmx34, 2020-06-29:
> Oh, i thought there werent any configs for BigTreeTech boards in that branch, but as it turns out, the E3 boards in bugfix-2.0.x are now under "Creality/Ender-3" and not under "BigTreeTech". Im [...]

> Had this same issue with the latest Marlin release. Just fixed it by using the `bugfix-2.0.x` firmware from the Marlin download page and following the BTT instructions on their github. Note that you [...]

I followed your notes and was able to open the "bugfix" version in VSCode/PlatformIO and compile successfully with no changes. I brought over the Configuration.h and Configuration.adv.h files from BigTree and the compile failed with several errors. One said ""Oops! Select an STM32F1 board in 'Tools > Board.'" When I look up STM32F1, there are numerous variations listed, but none refer to BigTree. I've looked at the "Pin" file under BigTree's hardware folder to locate the board name and it's not listed there. Obviously, I'm new to all this and don't entirely understand how to pull all this together. All I want to do it get proper settings to my new BLTouch so it functions properly and get the SD card slot on the control board working from my new TFT35 screen.

HobbyADD, 2020-07-01:
I am sorry to ask this but I dont know what to do. I am very new to using and navigating github and firmware in general. However, I have the same problem as the OP. I compiled no problem back on Jun 9th. I wanted to make some changes and now today I cannot compile and I get this same error. Do I have to compile from scratch again with the bugfix marlin or is there a line I can change in the config.h file. I dont know enough about how this compiling works. All coding I have done was all in the local machine without calls to a website. So if nothing changed on the PC compiling the code should still work. I have changed nothing on my local computer. And I am not even sure how to pull the files from github to be honest. This whole site confuses me. What I downloaded came from Marlins website in a zip. and it doesnt have all the files that the 2.0.x zip has so I am unsure how to proceed.

runninglennart, 2020-07-01:
For the libmaple/gpio error take a look at this: https://reprap.org/forum/read.php?415,874307 it solved my problems and i got NeoPixels working

HobbyADD, 2020-07-01:
> For the libmaple/gpio error take a look at this: https://reprap.org/forum/read.php?415,874307 it solved my problems and i got NeoPixels working

Thank you thank you thank you! I was able to compile successfully.

Samusamu93, 2020-07-01:
Ho provato a seguire la guida elencata ma sono bloccato a questo :
Marlin \ src \ HAL \ STM32F1 \ dogm \ ../../../ inc /../ HAL /./ STM32F1 / fastio.h: 30: 10: errore fatale: libmaple / gpio.h: nessun file simile o directory:

`[env:STM32F103RC_btt_512K]
platform          = ${common_stm32f1.platform}
extends           = env:STM32F103RC_btt
board_upload.maximum_size=524288
build_flags       = ${env:STM32F103RC_btt.build_flags} -DSTM32_FLASH_SIZE=512`

Ho questo in platformio.ini e non riesco a trovare la riga da cambiare per compilare il firmware correttamente

[image]

> Per l'errore libmaple / gpio dai un'occhiata a questo: https://reprap.org/forum/read.php?415,874307 ha risolto i miei problemi e ho fatto funzionare NeoPixels

oldman4U, 2020-09-05:
@BelSD 

Could you please close this ticket and also https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/522 in case you do not need it anymore.

Thank you