bigtreetech/BIGTREETECH-SKR-mini-E3 #303: BLTOUCH succeeds G28, but fails G29

ulaimins, 2020-06-27:
**Setup:**
- SKR E3 Mini 1.2
- Genuine BLTOUCH V3.1

**Issue:**
BLTOUCH succeeds homing using G28, but bed level causes it to go to the first point and just stow/deploy the probe several times without leveling bed.

Worked prior to firmware upgrade.

[See gif here for a visual](https://gfycat.com/lonelyorangeelephant)

**Config Files:**
- [Configuration.h](https://easyupload.io/euttmd)
- [Configuration_adv.h](https://gofile.io/d/xIRzqQ)

Debug Output:

```
1. Send: M111 S32 <<< DEBUG START>>>
2. Recv: echo:DEBUG:LEVELING
3. Recv: ok
4. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
5. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
6. Send: G28 <<<HOMING G28 START>>>
7. Recv: >>> G28
8. Recv: Machine Type: Cartesian
9. Recv: Probe: BLTOUCH
10. Recv: Probe Offset X-45.00 Y-7.00 Z-1.85 (Left-Front & Below Nozzle)
11. Recv: Auto Bed Leveling: BILINEAR (disabled)
12. Recv: 
13. Recv: >>> homeaxis(X)
14. Recv: Home 1 Fast:
15. Recv: >>> do_homing_move(X, -345.00, [20.00])
16. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
17. Recv: echo:busy: processing
18. Recv:  T:20.62 /0.00 B:20.94 /0.00 @:0 B@:0
19. Recv: echo:busy: processing
20. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
21. Recv: echo:busy: processing
22. Recv:  T:20.62 /0.00 B:20.94 /0.00 @:0 B@:0
23. Recv: echo:busy: processing
24. Recv: <<< do_homing_move(X)
25. Recv: Move Away:
26. Recv: >>> do_homing_move(X, 5.00, [20.00])
27. Recv: <<< do_homing_move(X)
28. Recv: Home 2 Slow:
29. Recv: >>> do_homing_move(X, -10.00, 10.00)
30. Recv: <<< do_homing_move(X)
31. Recv: >>> set_axis_is_at_home(X)
32. Recv: Axis 88 home_offset = 0.00 position_shift = 0.00
33. Recv: > home_offset[X] = 0.00
34. Recv:   current_position= X0.00 Y122.00 Z11.85 :
35. Recv: <<< set_axis_is_at_home(X)
36. Recv:   current_position= X0.00 Y122.00 Z11.85 : sync_plan_position
37. Recv:   current_position= X0.00 Y122.00 Z11.85 : > AFTER set_axis_is_at_home
38. Recv: <<< homeaxis(X)
39. Recv: >>> homeaxis(Y)
40. Recv: Home 1 Fast:
41. Recv: >>> do_homing_move(Y, -345.00, [20.00])
42. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
43. Recv: echo:busy: processing
44. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
45. Recv: echo:busy: processing
46. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
47. Recv: echo:busy: processing
48. Recv: <<< do_homing_move(Y)
49. Recv: Move Away:
50. Recv: >>> do_homing_move(Y, 5.00, [20.00])
51. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
52. Recv: <<< do_homing_move(Y)
53. Recv: Home 2 Slow:
54. Recv: >>> do_homing_move(Y, -10.00, 10.00)
55. Recv: echo:busy: processing
56. Recv: <<< do_homing_move(Y)
57. Recv: >>> set_axis_is_at_home(Y)
58. Recv: Axis 89 home_offset = 0.00 position_shift = 0.00
59. Recv: > home_offset[Y] = 0.00
60. Recv:   current_position= X0.00 Y0.00 Z11.85 :
61. Recv: <<< set_axis_is_at_home(Y)
62. Recv:   current_position= X0.00 Y0.00 Z11.85 : sync_plan_position
63. Recv:   current_position= X0.00 Y0.00 Z11.85 : > AFTER set_axis_is_at_home
64. Recv: <<< homeaxis(Y)
65. Recv: BLTouch Command :160
66. Recv: BLTouch Command :90
67. Recv: last_written_mode - 1
68. Recv: config mode - BLTOUCH_SET_5V_MODE
69. Recv: home_z_safely >>>
70. Recv:   current_position= X0.00 Y0.00 Z11.85 : sync_plan_position
71. Recv:   destination= X160.00 Y122.00 Z13.70 : home_z_safely
72. Recv: >>> do_blocking_move_to X160.00 Y122.00 Z11.85
73. Recv: <<< do_blocking_move_to
74. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
75. Recv: echo:busy: processing
76. Recv: >>> homeaxis(Z)
77. Recv:   current_position= X160.00 Y122.00 Z11.85 : Probe::set_deployed
78. Recv: deploy: 1
79. Recv: Probe::move_z(10.00)
80. Recv: >>> do_blocking_move_to X160.00 Y122.00 Z11.85
81. Recv: <<< do_blocking_move_to
82. Recv: Home 1 Fast:
83. Recv: BLTouch DEPLOY requested
84. Recv: BLTouch Command :10
85. Recv: bltouch.deploy_proc() end
86. Recv: >>> do_homing_move(Z, -375.00, [4.00])
87. Recv: echo:busy: processing
88. Recv:  T:20.62 /0.00 B:20.94 /0.00 @:0 B@:0
89. Recv: echo:busy: processing
90. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
91. Recv: <<< do_homing_move(Z)
92. Recv: BLTouch STOW requested
93. Recv: BLTouch Command :90
94. Recv: bltouch.stow_proc() end
95. Recv: Move Away:
96. Recv: >>> do_homing_move(Z, 5.00, 4.00)
97. Recv: echo:busy: processing
98. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
99. Recv: <<< do_homing_move(Z)
100. Recv: Home 2 Slow:
101. Recv: BLTouch DEPLOY requested
102. Recv: BLTouch Command :10
103. Recv: bltouch.deploy_proc() end
104. Recv: >>> do_homing_move(Z, -10.00, 2.00)
105. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
106. Recv: echo:busy: processing
107. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
108. Recv: echo:busy: processing
109. Recv: <<< do_homing_move(Z)
110. Recv: BLTouch STOW requested
111. Recv: BLTouch Command :90
112. Recv: bltouch.stow_proc() end
113. Recv: >>> set_axis_is_at_home(Z)
114. Recv: *** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
115. Recv: > probe.offset.z = -1.85
116. Recv: Axis 90 home_offset = 0.00 position_shift = 0.00
117. Recv: > home_offset[Z] = 0.00
118. Recv:   current_position= X160.00 Y122.00 Z1.85 :
119. Recv: <<< set_axis_is_at_home(Z)
120. Recv:   current_position= X160.00 Y122.00 Z1.85 : sync_plan_position
121. Recv:   current_position= X160.00 Y122.00 Z1.85 : > AFTER set_axis_is_at_home
122. Recv:   current_position= X160.00 Y122.00 Z1.85 : Probe::set_deployed
123. Recv: deploy: 0
124. Recv: Probe::move_z(10.00)
125. Recv: >>> do_blocking_move_to X160.00 Y122.00 Z11.85
126. Recv: <<< do_blocking_move_to
127. Recv:  T:20.62 /0.00 B:20.94 /0.00 @:0 B@:0
128. Recv: echo:busy: processing
129. Recv: >>> do_blocking_move_to X160.00 Y122.00 Z11.85
130. Recv: <<< do_blocking_move_to
131. Recv: <<< homeaxis(Z)
132. Recv: <<< home_z_safely
133. Recv:   current_position= X160.00 Y122.00 Z11.85 : sync_plan_position
134. Recv: X:160.00 Y:122.00 Z:11.85 E:0.00 Count X:12800 Y:9760 Z:4740
135. Recv: <<< G28
136. Recv: ok
137. Recv:  T:20.62 /0.00 B:20.94 /0.00 @:0 B@:0
138. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
139. Send: G29 <<<BED LEVELING G29 START>>>
140. Recv:   current_position= X160.00 Y122.00 Z11.85 : >>> G29
141. Recv: Machine Type: Cartesian
142. Recv: Probe: BLTOUCH
143. Recv: Probe Offset X-45.00 Y-7.00 Z-1.85 (Left-Front & Below Nozzle)
144. Recv: Auto Bed Leveling: BILINEAR (disabled)
145. Recv: 
146. Recv:   current_position= X160.00 Y122.00 Z11.85 : Probe::set_deployed
147. Recv: deploy: 1
148. Recv: Probe::move_z(10.00)
149. Recv: >>> do_blocking_move_to X160.00 Y122.00 Z11.85
150. Recv: <<< do_blocking_move_to
151. Recv: reset_bed_level
152. Recv: >>> Probe::probe_at_point(20.00, 20.00, raise, 0, probe_relative)
153. Recv:   current_position= X160.00 Y122.00 Z11.85 :
154. Recv: >>> do_blocking_move_to X65.00 Y27.00 Z11.85
155. Recv: <<< do_blocking_move_to
156. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
157. Recv:   current_position= X65.00 Y27.00 Z11.85 : Probe::set_deployed
158. Recv: deploy: 1
159. Recv:   current_position= X65.00 Y27.00 Z11.85 : >>> Probe::run_z_probe
160. Recv:   current_position= X65.00 Y27.00 Z11.85 : >>> Probe::probe_down_to_z
161. Recv: BLTouch DEPLOY requested
162. Recv: BLTouch Command :10
163. Recv: BLTouch ALARM or TRIGGER after DEPLOY, recovering
164. Recv: BLTouch Command :160
165. Recv: BLTouch Command :90
166. Recv: BLTouch Command :10
167. Recv: BLTouch Command :90
168. Recv: BLTouch Command :10
169. Recv: BLTouch Recovery Failed
170. Recv: Error:!! STOP called because of BLTouch error - restart with M999
171. Recv: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
172. Recv: SLOW Probe fail!
173. Recv:   current_position= X65.00 Y27.00 Z11.85 : <<< run_z_probe
174. Recv:   current_position= X65.00 Y27.00 Z11.85 : Probe::set_deployed
175. Recv: deploy: 0
176. Recv: Probe::move_z(10.00)
177. Recv: >>> do_blocking_move_to X65.00 Y27.00 Z11.85
178. Recv: <<< do_blocking_move_to
179. Recv: Error:Probing Failed
180. Recv: <<< Probe::probe_at_point
181. Recv:   current_position= X65.00 Y27.00 Z11.85 : Probe::set_deployed
182. Recv: deploy: 0
183. Recv:   current_position= X65.00 Y27.00 Z11.85 : > probing complete
184. Recv: X:65.00 Y:27.00 Z:11.85 E:0.00 Count X:5200 Y:2160 Z:4740
185. Recv: <<< G29
186. Recv: ok
187. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
188. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
189. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0
190. Recv:  T:20.62 /0.00 B:20.62 /0.00 @:0 B@:0

```

Logo2001, 2020-06-27:
hi,
enable #define Endstoppullups
The BL-Touch is may be in Floating State.

ulaimins, 2020-06-27:
> hi,
> enable #define Endstoppullups
> The BL-Touch is may be in Floating State.

Thanks for the reply. That didn't fix the issue :(.

cdhunter1010, 2020-08-01:
look at line  #define z_probe_low_point change the value to a greater number or disable