bigtreetech/BIGTREETECH-SKR-mini-E3 #305: BLTouch auto leveling only probes lower left corner

Brandis13, 2020-06-28:
I added a BLTouch to my Ender 3 with SKR mini v1.3. Added the regular BLtouch firmware from this github, as I still use the standard switch for Z-stop. I also have a BTT touchscreen with the touchscreen UI installed.

I then manually leveled with the paper method in all 4 corners and Z offset is 0, if I understand the process correctly.

I then wanted to start Auto Bed Leveling, but the head only moves to the lower left corner of the bed, deploys the probe and slowly moves down, the probe reacts way before the nozzle even gets close to the bed (not sure if this is normal behaviour) and then I get the message "Settings Stored (629 bytes; crc 48408). Then the head moves up a bit and nothing else happens.

I was under the impression that Auto Bed Leveling with BLtouch would probe all 4 corners and maybe also the center? Am I doing something wrong?

Brandis13, 2020-06-28:
By the way, I also tested it in Marlin mode (with the classic UI, using the wheel to navigate menus) and when I select bed leveling, it also only probes the lower left corner.

ulaimins, 2020-06-29:
I have a similar issue. See #303