bigtreetech/BIGTREETECH-SKR-mini-E3 #309: [FR] BIGTREETECH SKR Mini E3 V2.0 compatibility with Creality CR-10 and Creality CR-10S?

Hedda, 2020-07-01:
**Is your feature request related to a problem? Please describe.**

Problem is that other than the EZBoard Lite by TH3D there is really no relativly inexpensive third-party drop-in replacement 32-bit mainboards with integrated stepping-drivers available for the stock Creality CR-10 and Creality CR-10s, both of which are still popular for those wanting larger print-beds than Creality Ender series.

I guess the main relative difference between CR10 and CR10S is dual Z-axis so might just need a Dual Cable LCD adapter/board and Z-axis dual adapter/board (upgrade/conversion kits) or other addon mods?

**Describe the solution you'd like**

@bigtreetech Please consider making SKR Mini E3 V2.0 mainboard fully compatibility with Creality CR-10 and Creality CR-10S.

Technically I would think it possible hardware-wise for BIGTREETECH to make the existing SKR Mini E3 V2.0 mainboards fully compatibility with stock Creality CR-10 and Creality CR-10s with custom firmware and possible some upgrade-kit with small hardware modifications(?) if needed to make it compatible with an upgrade/conversion kit containing Creality Dual Cable LCD Board and dual Z-axis stepping motor adapter for CR-10s if needed.

Personally I have a Creality CR-10s and would love if there was a relatively inexpensive drop-in replacement 32-bit mainboard with integrated stepping-drivers available for it, and I definitely believe that there is a good-sized marked among CR-10/CR-10S owners if you could officially make the SKR Mini E3 V2.0 mainboard fully compatible with CR-10 and CR-10S.

**Describe alternatives you've considered**

EZBoard Lite by TH3D is an existing drop-in replacement 32-bit mainboard that is compatible with both Creality CR-10 series , Ender 3 and Ender 5 series, but the downside is that it is much more expensive and only ships from the United States, (there is no resellers outside the United States, like for example in the European Union).

https://www.th3dstudio.com/product/ezboard-lite/

EZBoard Lite by TH3D is compatible with CR-10 (Regular, Mini, S4 S5), CR-10S (Regular, Mini, S4 S5), Ender 3 (Regular & Pro), and Ender 5 Plus (Regular, Pro, Plus), as well as with Sovol SV01. TH3D support and provide official firmware for all of those.

**Additional context**

Recommend check out all the extra hardware adapters that are available as options for EZBoard Lite by TH3D to make it compatible with more printers than just Ender 3.

https://www.th3dstudio.com/product/ezboard-lite/

Hedda, 2020-07-01:
To compare, EZBoard Lite by TH3D list these printers are supported and have information on what hardware adapters are needed:

- CR-10 (Regular, Mini, S4 S5)
  - Models with 2 Z Motors need the Dual Z Motor Adapter Addon -> https://www.th3dstudio.com/product/dual-z-stepper-motor-adapter-w-cable/

- CR-10S (Regular, Mini, S4 S5)
  - Needs Creality Dual Cable LCD Adapter Board if you do not have the EXP3 header on your LCD -> https://www.th3dstudio.com/product/ezboard-creality-dual-lcd-adapter-board/
    - If you have the EXP3 header on your LCD use 1 cable to connect from the EXP3 to our board and you will then not need the dual adapter board.
  - Models with 2 Z Motors need the Dual Z Motor Adapter Addon -> https://www.th3dstudio.com/product/dual-z-stepper-motor-adapter-w-cable/
  - Does NOT work with Touch Screen Models

- Ender 3
- Ender 3 Pro
- Ender 5
- Ender 5 Pro
- Ender 5 Plus – Needs other addons/upgrades listed below
  - Needs Creality Dual Cable LCD Adapter Board -> https://www.th3dstudio.com/product/ezboard-creality-dual-lcd-adapter-board/
  - Needs the 12864 LCD Conversion Kit -> https://www.th3dstudio.com/product/ender-5-plus-12864-lcd-upgrade-kit/
  - Needs the Dual Z Motor Adapter Addon -> https://www.th3dstudio.com/product/dual-z-stepper-motor-adapter-w-cable/
  - Does NOT work with the stock Touch Screen, only with the 12864 LCD Conversion linked above

- Sovol SV01
  - Needs Dual Z Motor Adapter Addon -> https://www.th3dstudio.com/product/dual-z-stepper-motor-adapter-w-cable/

Hedda, 2020-07-01:
Please also see feature request https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/295

Hedda, 2020-07-01:
Please also see this upstream request for config example files -> https://github.com/MarlinFirmware/Configurations/issues/141

Hedda, 2020-07-01:
PS: We are also missing config examples for Ender 5 series which should also be hardware compatible though not marketed so.

Hedda, 2020-07-02:
Mounting the board in a CR-10S might be an issue? According to TH3D website; CR-10S has a V2.0 mainboard with Atmel 2560 chip but looking at the pictures I can not tell for sure if it looks to have different mounting holes or not from the CR-10? 

https://support.th3dstudio.com/hc/en-us/articles/360043293512-Creality-CR-10-and-CR-10S-Models-What-Printer-do-you-have-

Pictures of V2.1 and V2.2 mainboards for CR-10S looks to have removed one mounting hole in order to put in more fan ports and/or a port for filaments detector, however, otherwise, it at least just looking that the pictures the mounting holes on V2.x board for CR-10S boards look very similar to that of CR-10 boards?

Hedda, 2020-07-02:
For reference, here are some CR-10 and CR-10S configuration example files for SKR Mini E3 V1.2

CR-10S configuration example files for SKR Mini E3 V1.2 by @MickyKom

https://github.com/MickyKom/SKR-Mini-E3-CR-10s

CR-10 configuration example files for SKR Mini E3 V1.2 with direct BMG extruder and BLTouch by @samgozman

https://github.com/samgozman/CR-10-BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/Marlin

Apparently also Kay3D Cheetah 5.0 Marlin Firmware fork also claim supports SKR Mini E3 v2.0 and the whole Creality CR-10 series but they too only refer to upstream Marlin Firmware for configurations files:

https://kay3d.com/pages/kay3d-cheetah-5-firmware

https://github.com/KAY3D/Marlin

Gamester17, 2020-07-03:
For all your information; I sent a similar question about CR-10S support to BIQU customer service at support@bigtree-tech.com

They replied today saying that CR-10S is hardware compatible SKR Mini E3 V2.0 however they also added 

  _"firmware needs to be configured by the customer, because we do not know the specific situation of the customer, 
we do not support the configuration of special firmware"_

Hopefully, as it is in their interest to improve sales BIQU should update all their marketing material for the SKR Mini E3 V2.0  with more clear information on what Creality 3D-printers are compatible or not. 

If it is compatible with CR-10S then it should also be compatible with CR-10, CR-10S, CR-10 S4, CR-10 S5, CR-10 Mini, Ender 3, Ender 3 Pro, Ender 5, Ender 5 Pro, Ender 5 Plus.

For reference I have also posted the same question on CR10 Reddit here:

https://www.reddit.com/r/CR10/comments/hjwire/bigtreetech_skr_mini_e3_20_board_hardware_and/

I would have posted that on the BIGTREETECH Reddit instead but they are still having problems with restricted posting:

https://www.reddit.com/r/BIGTREETECH

webwookie, 2020-07-05:
I haven't found any issues with installing the SKR Mini E3 V2.0 board in my CR-10S; however, i've had no success to date with compiling firmware that will run on the otherwise stock CR-10S.

themodfatherinc, 2020-07-22:
im currently using it on a CR-10 clone (CTC A10S) and its mostly working. the ender 3 firmware which comes on the board WILL WORK however it thinks your printer is smaller than it is. I have successfully configured my firmware and uploaded it with the exception that i am not getting accurate temperatures on my hot end. Im not sure if this is the boards fault or thermistor or something else entirely. I am still trying to work this one issue out. i can print PLA just fine but when i try to print nylon or petG im having to turn the temp up like 15-20 degrees higher than i needed it before changing the board out and still not getting the results i really want. i may swap out my heater cartridge and thermistor again for good measure just in case its one of those. The board does work though and im hoping to iron out this temp issue soon

webwookie, 2020-07-25:
> im currently using it on a CR-10 clone (CTC A10S) and its mostly working. the ender 3 firmware which comes on the board WILL WORK however it thinks your printer is smaller than it is. I have [...]

I assume that you successfully reconfigured the x, y, and z limits for the larger printer envelope?  Could you upload the .bin that you used for the SKR Mini E3 V2.0 to github?

themodfatherinc, 2020-07-26:
> I assume that you successfully reconfigured the x, y, and z limits for the larger printer envelope? Could you upload the .bin that you used for the SKR Mini E3 V2.0 to github?

i got you. took me too long to figure this shit out lol. enjoy. theres a pre-compiled bin in the main folder or you can edit and recompile it yourself. i had to fork from the actual marlin not here bc this version for whatever reason wouldnt play nice with my thermistor. its working great for me now. petG running at 240 again like expected. hope it works for you, if you have any issues let me know

Configured for CR-10 type printer 300*300*400
https://github.com/themodfatherinc/Marlin

Gamester17, 2020-10-01:
Please help test submitted Marlin config files in https://github.com/MarlinFirmware/Configurations/pull/237 and https://github.com/MarlinFirmware/Configurations/pull/238 that should work with stock CR-10 or stock CR-10S upgraded with BigTreeTech SKR Mini E3 2.0 board.

Gamester17, 2020-10-05:
FYI, MarlinFirmware Configurations upstream have now committed BigTreeTech SKR Mini E3 2.0 config examples for both stock Creality CR-10 and stock Creality CR-10S

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/BigTreeTech%20SKR%20Mini%20E3%202.0

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%20Mini%20E3%202.0

Now all that is missing is for @bigtreetech to provide precompiled SKR Mini E3 2.0 binary firmware files for CR10 and CR10S .