bigtreetech/BIGTREETECH-SKR-mini-E3 #311: [BUG] BL Toch won’t work like it should.

00ihsan, 2020-07-03:
### Description

On the printer itself:

When I home the printer via the LCD it home’s correctly. Then I choose level bed, the printer starts levelling but after the first probing it freezes and returns to the homescreen. 

On octoprint:

On octoprint I use a tool called bed visualizer. I give it the following commands: G28
M155 S30
@BEDLEVELVISUALIZER
G29 T
M155 S3

After those commands the printer starts homing and starts to probe in the same left lower corner. After the first probe i get a serial error “probing failed” and on the printer “M112 Shutdown PRINTER HALTED Please Reset”. 

### Steps to reproduce

1. Connect bltouch
2. Install firmware bltoch
3. Start bedleveling in the menu and octoprint 

**Expected behavior**

Bed levveling success. Mesh updated and setting stored. 

**Actual behavior**

Printer freezes without octoprint and with octoprint if gives the errors as described above. 

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

choochus, 2020-07-08:
I have the same issue, with BTT SKR mini E3 V2.0 and genuine BL-Touch 3.1
Auto homes to bottom left
Probes first point correctly, then raises and stops. 

I've manually leveled and set z offset (-1.4 for me)

choochus, 2020-07-08:
Seems to be duplicate of issue [#268](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/268)